esphome/issues #5149: http_request not releasing resources (https POST only)

HLFCode, 2023-11-24:
### The problem

I'm trying to post data to emoncms.org using https and my esphome device runs out of heap.

http_request.post using http**s** permanently depletes free heap.
The same request using http (no s) does not.
The loss is significant - about 14000 (getFreeHeap)

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A, problem in ESPHome only

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

http_request

### Example YAML snippet

```yaml
substitutions:
  remote_url: emoncms.org/input/post
  api_key: !secret emoncms_api_key
  
esphome:
  name: test-http
  friendly_name: Test Http
  on_boot:
    priority: 0
    then:
      - lambda: |-
          id(url).publish_state("${remote_url}");
esp8266:
  board: d1_mini
  restore_from_flash: True

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "BLgqm59yAA53+g7vhQEcg1Z4mWolTgGshs4vQ9PiN4I="
  reboot_timeout: 0s # prevent reboots every 15 minutes when no Home Assistant connected

ota:
  password: "c0b7e2a60cdadb4ca2efe9ff2abb3f4b"

wifi:
  ssid: !secret office_wifi_ssid
  password: !secret office_wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Http Fallback Hotspot"
    password: "OvA4uH8ACuWQ"

captive_portal:

http_request:
  useragent: esphome/device
  timeout: 5s
  id: http_request_data

# enable web server so we can interactively try http and https
web_server:
  port: 80
  ota: True
  local: false # change to true if no internet

sensor:
# show the free heap
  - platform: template
    name: Free Heap
    id: free_heap
    unit_of_measurement: " "
    update_interval: 10sec
    accuracy_decimals: 0
    lambda: "return ESP.getFreeHeap();"

text_sensor:
  - platform: template
    name: Url
    id: url
    update_interval: never

button:
  - platform: restart
    name: Restart
    id: restart_button
# button to try http protocol
  - platform: template
    name: Send http
    on_press:
      then:
        - script.execute:
            id: send_httpx
            prefix: "http"
# button to try https protocol
  - platform: template
    name: Send https
    on_press:
      then:
        - script.execute:
            id: send_httpx
            prefix: "https"

# script to send the same data with either http or https
script:
  - id: send_httpx
    parameters:
      prefix: string
    then:
      - lambda: |-
          ESP_LOGI("send_httpx", "sending via %s, free heap %d", prefix.c_str(), ESP.getFreeHeap());
      - http_request.post:
          url: !lambda |-
            return prefix + "://" + id(url).state;
          verify_ssl: false
          headers:
            Content-Type: application/x-www-form-urlencoded
          body: node=10&data={a:1}&apikey=${api_key}

          on_response:
            then:
              - lambda: |-
                  ESP_LOGI("send_httpx", "on_response: status %d, Free heap %d, response %s", status_code, ESP.getFreeHeap(), id(http_request_data).get_string());

# log free heap routinely
interval:
  - interval: 30s
    then:
      - logger.log:
          format: "Free Heap %d"
          args: [ESP.getFreeHeap()]
          level: INFO
```

### Anything in the logs that might be useful for us?

```txt
The test app allows the user to POST via http or https through its web server
The log shows the fall in free heap the first time https is used.
The heap is never released...

08:32:39	[I]	[main:133]	Free Heap 34864
08:33:09	[I]	[main:133]	Free Heap 35120
08:33:39	[I]	[main:133]	Free Heap 35120
# stable free heap of 35000, trying http
08:33:40	[I]	[send_httpx:098]	sending via http, free heap 32536
08:33:40	[I]	[send_httpx:110]	on_response: status 200, Free heap 34192, response ok
08:33:40	[W]	[component:204]	Component web_server took a long time for an operation (0.08 s).
08:33:41	[W]	[component:205]	Components should block for at most 20-30ms.
08:34:09	[I]	[main:133]	Free Heap 34696
08:34:39	[I]	[main:133]	Free Heap 34696
08:35:09	[I]	[main:133]	Free Heap 34696
# minimal loss of free heap
# trying https
08:35:10	[I]	[send_httpx:098]	sending via https, free heap 34632
08:35:11	[I]	[send_httpx:110]	on_response: status 200, Free heap 22928, response ok
08:35:11	[W]	[component:204]	Component web_server took a long time for an operation (1.06 s).
08:35:11	[W]	[component:205]	Components should block for at most 20-30ms.
08:35:39	[I]	[main:133]	Free Heap 22928
08:36:09	[I]	[main:133]	Free Heap 22928
# loss of about 12000 free heap (1/3)
# try again
08:36:16	[I]	[send_httpx:098]	sending via https, free heap 19968
08:36:17	[I]	[send_httpx:110]	on_response: status 200, Free heap 22536, response ok
08:36:17	[W]	[component:204]	Component web_server took a long time for an operation (1.06 s).
08:36:17	[W]	[component:205]	Components should block for at most 20-30ms.
08:36:39	[I]	[main:133]	Free Heap 22536
08:37:09	[I]	[main:133]	Free Heap 22832
# no significant further loss
08:37:12	[I]	[send_httpx:098]	sending via https, free heap 22464
08:37:13	[I]	[send_httpx:110]	on_response: status 200, Free heap 22832, response ok
08:37:13	[W]	[component:204]	Component web_server took a long time for an operation (1.07 s).
08:37:13	[W]	[component:205]	Components should block for at most 20-30ms.
08:37:39	[I]	[main:133]	Free Heap 22832
# and again
08:37:45	[I]	[send_httpx:098]	sending via https, free heap 20264
08:37:46	[I]	[send_httpx:110]	on_response: status 200, Free heap 22832, response ok
08:37:46	[W]	[component:204]	Component web_server took a long time for an operation (1.09 s).
08:37:46	[W]	[component:205]	Components should block for at most 20-30ms.
08:38:09	[I]	[main:133]	Free Heap 22832
# free heap stable (but still lost 12000)
```

HLFCode, 2023-11-27:
If anybody wants to work on this I can provide a temporary apikey for the secret `emoncms_api_key`
If not it's easy to create a test account with [emoncms ](https://emoncms.org)
With this the issue can be duplicated using the above yaml.
I imagine the heap loss is also an issue with any https usage but I haven't tried anything other than `http_request.post`

Unfortunately I wouldn't know where to start

guillempages, 2024-07-21:
@HLFCode the http_request component has been heavily refactored on the 2024.6 release; could you try if by any chance the refactoring fixed your issue?