esphome/issues #5152: Can't update Shelly Dimmer 2 STM32 firmare

djranm, 2023-11-25:
### The problem

I seem to encountering a similar issue to #3242.

I have several Shelly Dimmer 2s flashed with ESPHome, running STM32 firmware version 51.5. When I try to update ESPHome and increment the firmware version to 51.7, it doesn't update and tells me to include 'update: true' in my yaml, whether it's there or not. I've tried power cycling the circuit and there's no difference on cold boot vs. warm. How can I update the firmware to 51.7, even manually?

### Which version of ESPHome has the issue?

2023.11.4

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

shelly_dimmer

### Component causing the issue

shelly_dimmer

### Example YAML snippet

```yaml
esphome:
  name: lounge-back-lights
  platform: ESP8266
  board: esp01_1m
  comment: Lounge Back Lights

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  manual_ip:
    static_ip: 10.1.1.10
    gateway: 10.1.1.254
    subnet: 255.255.255.0
    dns1: 10.1.1.251
    dns2: 10.1.1.251

api:
  encryption:
    key: !secret api_password

ota:
  password: !secret ota_password

uart:
    tx_pin: 1
    rx_pin: 3
    baud_rate: 115200

sensor:

logger:

web_server:

light:
    - platform: shelly_dimmer
      name: Lounge Back Lights
      id: lounge_back_lights
      power:
        name: Lounge Back Lights Power
      voltage:
        name: Lounge Back Lights Voltage
      current:
        name: Lounge Back Lights Current
      min_brightness: 60
      warmup_brightness: 220
      default_transition_length: 0s
      firmware:
        version: "51.7"
        update: true

binary_sensor:
  - platform: gpio
    name: 'Switch 1'
    id: switchID_1
    pin:
      number: GPIO14
      mode: INPUT
    on_state:
      then:
        - logger.log: "Light switch toggled"
        - light.toggle: lounge_back_lights
```

### Anything in the logs that might be useful for us?

```txt
[13:24:11][I][app:102]: ESPHome version 2023.11.4 compiled on Nov 25 2023, 13:23:37
[13:24:12][C][wifi:559]: WiFi:
[13:24:12][C][wifi:391]:   Local MAC: 3C:61:05:E3:5E:7F
[13:24:12][C][wifi:396]:   SSID: 'wifi'
[13:24:12][C][wifi:397]:   IP Address: 10.1.1.10
[13:24:12][C][wifi:398]:   BSSID: B0:39:56:93:92:E5
[13:24:12][C][wifi:400]:   Hostname: 'lounge-back-lights'
[13:24:12][C][wifi:402]:   Signal strength: -49 dB ▂▄▆█
[13:24:12][C][wifi:406]:   Channel: 6
[13:24:12][C][wifi:407]:   Subnet: 255.255.255.0
[13:24:12][C][wifi:408]:   Gateway: 10.1.1.254
[13:24:12][C][wifi:409]:   DNS1: 10.1.1.251
[13:24:12][C][wifi:410]:   DNS2: 10.1.1.251
[13:24:12][C][logger:416]: Logger:
[13:24:12][C][logger:417]:   Level: DEBUG
[13:24:12][C][logger:418]:   Log Baud Rate: 115200
[13:24:12][C][logger:420]:   Hardware UART: UART0
[13:24:12][C][uart.arduino_esp8266:102]: UART Bus:
[13:24:12][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[13:24:12][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[13:24:12][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[13:24:12][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[13:24:12][C][uart.arduino_esp8266:109]:   Data Bits: 8
[13:24:12][C][uart.arduino_esp8266:110]:   Parity: NONE
[13:24:12][C][uart.arduino_esp8266:111]:   Stop bits: 1
[13:24:12][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[13:24:12][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
[13:24:12][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Switch 1'
[13:24:12][C][gpio.binary_sensor:016]:   Pin: GPIO14
[13:24:12][C][light:103]: Light 'Lounge Back Lights'
[13:24:12][C][light:105]:   Default Transition Length: 0.0s
[13:24:12][C][light:106]:   Gamma Correct: 1.00
[13:24:12][C][web_server:168]: Web Server:
[13:24:12][C][web_server:169]:   Address: 10.1.1.10:80
[13:24:12][C][mdns:115]: mDNS:
[13:24:12][C][mdns:116]:   Hostname: lounge-back-lights
[13:24:12][C][ota:097]: Over-The-Air Updates:
[13:24:12][C][ota:098]:   Address: 10.1.1.10:8266
[13:24:12][C][ota:101]:   Using Password.
[13:24:12][C][api:139]: API Server:
[13:24:12][C][api:140]:   Address: 10.1.1.10:6053
[13:24:12][C][api:142]:   Using noise encryption: YES
[13:24:12][C][shelly_dimmer:118]: ShellyDimmer:
[13:24:12][C][shelly_dimmer:119]:   NRST Pin: GPIO5
[13:24:12][C][shelly_dimmer:120]:   BOOT0 Pin: GPIO4
[13:24:12][C][shelly_dimmer:122]:   Leading Edge: NO
[13:24:12][C][shelly_dimmer:123]:   Warmup Brightness: 220
[13:24:12][C][shelly_dimmer:126]:   Minimum Brightness: 60
[13:24:12][C][shelly_dimmer:127]:   Maximum Brightness: 1000
[13:24:12][C][shelly_dimmer:129]:   Update Interval: 10.0s
[13:24:12][C][shelly_dimmer:131]:   STM32 current firmware version: 51.5 
[13:24:12][C][shelly_dimmer:132]:   STM32 required firmware version: 51.7
[13:24:12][E][shelly_dimmer:137]:   Firmware version mismatch, put 'update: true' in the yaml to flash an update.
```

guyke01, 2023-12-23:
I've got the same issue with my new Shelly Dimmer 2. Tried version 51.5, 51.6 and 51.7 but the same message is showing
`[20:32:31][C][shelly_dimmer:131]:   STM32 current firmware version: 0.0 `
`[20:32:31][C][shelly_dimmer:132]:   STM32 required firmware version: 51.5`
`[20:32:31][E][shelly_dimmer:137]:   Firmware version mismatch, put 'update: true' in the yaml to flash an update.`

The line _update: true_ is present in the config.

DrBlokmeister, 2023-12-24:
I think I fixed it somehow, but I'm not sure how I achieved this. Could you compare your config to mine see if there are differences? https://github.com/DrBlokmeister/HASS_NUC/blob/master/esphome/templates/shelly_dimmer.yaml

guyke01, 2023-12-24:
Thanks @DrBlokmeister, yesterday i've flashed the Shelly Dimmer back to Tasmota with STM 51.7. After that i've flashed ESPHOME on the dimmer and it worked with the 51.7 version.

I will check your config with mine for some tweaking on my side.

bernikr, 2024-01-01:
I have the same issue, the logs tell me to put "update: true" into the config even though it already is

bernikr, 2024-01-01:
I found the issue:

the logger is interfering on the same UART port as the dimmer. by changing the logger config to

```
logger:
  baud_rate: 0
```

you can disable the uart logging and updating the firmeware worked for me.

DrBlokmeister, 2024-01-02:
I created a PR to update the documentation. This issue is horrible and I think I even threw away a Shelly module because of this. See PR here: https://github.com/esphome/esphome-docs/pull/3509