esphome/issues #5154: Firmware won't compile for esp32doit-devkit-v1 / Shelly Plus 1PM

czikocz, 2023-11-25:
### The problem

Hello, I am having compilation errors since version 2023.10.5 (first version I tried with the device, it had error similar to [#4865](https://github.com/esphome/issues/issues/4865) even after fix was implemented. Since version 2023.11.0 I am facing different error messages and with current 2023.11.4 it additionally takes forever to compile before ending in error. I installed esphome on Win to test and it compiled correctly with any of the mentioned versions.

### Which version of ESPHome has the issue?

2023.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP32-IDF

### Board

Shelly Plus 1PM

### Example YAML snippet

```yaml
substitutions:
  device_name: "shelly-p1pm-obz01"
  # Higher value gives lower watt readout
  current_res: "0.00106"
  # Lower value gives lower voltage readout
  voltage_div: "1785"

esphome:
  name: "${device_name}"
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret fallback_wifi_password

captive_portal:
  
logger:

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

time:
  - platform: homeassistant

[... 243 lines cut ...]
  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/shelly-p1pm-obz01/.pioenvs/shelly-p1pm-obz01/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_a8ac8 && [1/2] Building C object CMakeFiles/cmTC_a8ac8.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_a8ac8
    FAILED: cmTC_a8ac8 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_a8ac8.dir/testCCompiler.c.obj  -o cmTC_a8ac8   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)

======================== [FAILED] Took 1138.49 seconds ========================
```

jorisjansen31, 2023-11-29:
I have the exact same problem with a shelly plus 1pm, esphome 2023.11.6, Core 2023.11.3, Supervisor 2023.11.6, Operating System 11.1.

ssieb, 2024-03-08:
Do you have a 32-bit install?  Run `uname -m` in ssh if you don't know.

czikocz, 2024-03-08:
> Do you have a 32-bit install? Run `uname -m` in ssh if you don't know.

It returns armv7l, which is 32-bit I believe

ssieb, 2024-03-08:
Investigating with someone else, it appears that that just doesn't work.  I strongly suggest reinstalling HA with full 64-bit.

jorisjansen31, 2024-03-08:
I run full 64 bit and it also doesn't work.

ssieb, 2024-03-08:
@jorisjansen31 same error or something else?  If you aren't getting the liblto error, then it's not this issue.

czikocz, 2024-07-30:
I migrated to RPi 5, issue does not appear anymore on 64-bit system.

closed 2024-07-30 by czikocz