esphome/issues #5157: OTA MD5 failure on

fredericmorin, 2023-11-26:
### The problem

The second OTA upload fails.

Starting with a fresh mains unplug/plug of the device, I can push a new esphome fw OTA upload.
After that, every other OTA attempt fails with a `MD5 Check Failed` error.

Power cycling the device resolve the issue.

Same behavior is observed from both pip installed esphome and HA addon.

I was able to reproduce this issue without re-compiling the firmware. I ran `esphome upload` twice, without running `esphome compile`.

### Which version of ESPHome has the issue?

2023.11.4

### What type of installation are you using?

pip

### What platform are you using?

RTL87XX

### Board

BEELITE smart switch

### Component causing the issue

OTA

### Example YAML snippet

```yaml
esphome:
  name: plug21
  friendly_name: plug21
rtl87xx:
  board: generic-rtl8710bn-2mb-788k
logger:
ota: {"password": "1234567890"}
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
captive_portal:
web_server:
```

### Anything in the logs that might be useful for us?

```txt
RTL8710BN uart2 tx log output:

[...]
[C][lt.component:013]: LibreTiny:
[C][lt.component:014]:   Version: v1.4.1+sha.9b8e00c on generic-rtl8710bn-2mb-788k, compiled at Nov 26 2023 02:51:55, GCC 10.3.1 (-Os)
[C][lt.component:015]:   Loglevel: 3
[D][sensor:093]: 'WiFi Signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[D][sensor:093]: 'WiFi Signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[D][sensor:093]: 'WiFi Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[D][sensor:093]: 'WiFi Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[D][sensor:093]: 'WiFi Signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[I][ota:117]: Boot seems successful, resetting boot loop counter.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][sensor:093]: 'WiFi Signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[D][ota:151]: Starting OTA Update from 10.32.1.119...
[D][ota:320]: OTA in progress: 0.1%
[... 216 lines cut ...]
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 8.1.2+sha.776ae6c 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-amb1+sha.6297b80 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.4.1+sha.9b8e00c
 - ltchiptool @ 4.8.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.11.4
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
RAM:   [==        ]  20.6% (used 53888 bytes from 262144 bytes)
Flash: [======    ]  64.3% (used 519124 bytes from 806912 bytes)
============================================== [SUCCESS] Took 2.24 seconds ==============================================
INFO Successfully compiled program.
```

Kaldek, 2023-12-20:
I have the same issue.  This is a Tuya WR4 using the generic-rtl8710bn-2mb-468k board type as per the LibreTiny website.

UART log shows the following:
[image]

Jojo-A, 2024-01-04:
Hi,
I've got the same error over here. Tested with four devices and different board configurations, like generic, wr2, T102_V1.1,...
Same behavior after ota update. The devices end up in a bootloop until the safety mode is triggered. Then I can transmit a new firmware ota and everything starts all over again...

Strange thing: if I wait really long time (sometimes several hours) the device manages to boot completely and works as intended.

magtimmermans, 2024-02-19:
Also here same error

hitecSmartHome, 2024-09-11:
Same error ( idf 5.3.0 )

hitecSmartHome, 2024-09-18:
Mine is resolved when I stopped the communication on UART when an OTA update started.

kobaj, 2025-03-04:
> Mine is resolved when I stopped the communication on UART when an OTA update started.

What exactly do you mean by that @hitecSmartHome ?

hitecSmartHome, 2025-03-04:
Well, I have a continous uart communication on uart. If OTA started and I did not stop the communication, the ota failed.

kobaj, 2025-03-04:
> Well, I have a continous uart communication on uart. If OTA started and I did not stop the communication, the ota failed.

How are you stopping the uart communication though? I assume its some combination of `on_begin` within the `ota` block of your yaml. Could you post your entire config?

hitecSmartHome, 2025-03-04:
Sorry, I don't use config files. Iam coding but yes, I use the ota callbacks and just set a flag to not write to uart. But this is all related to uart things not being in iram and the flash and uart are blocking each other. You have to put uart things into iram.