esphome/issues #5161: Nextion display stopped working after upgrading to ESPHome 2023.11.6

edwardtfn, 2023-11-28:
### The problem

Nextion displays aren't stablishing communication to ESPHome since the latest updates. I'm not sure which version broke this, but it was between v2023.11.4 and v2023.11.6.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

nextion

### Example YAML snippet

```yaml
This was reported by several people using 

Basically, the code related to Nextion is the following:

display:
  - id: disp1
    platform: nextion
    uart_id: tf_uart
    start_up_page: 8
    on_page:
      lambda: |-
        static const char *const TAG = "display.disp1.on_page";
        ESP_LOGD(TAG, "Nextion page changed");
        ESP_LOGD(TAG, "New page: %s (%i)" , id(page_names)[x].c_str(), x);
        current_page->publish_state(id(page_names)[x].c_str());
        page_changed->execute(id(page_names)[x].c_str());

    on_setup:
      - script.execute: boot_sequence

The display isn't even detected, so `on_setup` isn't triggered.
```

### Anything in the logs that might be useful for us?

```txt
Before this change, the `dump_config` was shown even with empty info. Now it isn't shown, so probably it is crashing before getting to that point.

This is the logs of what I mean by `dump_config` (and which is not available anymore):

[08:01:23][C][nextion:123]: Nextion:
[08:01:23][C][nextion:124]:   Device Model:     NX4832F035_011C
[08:01:23][C][nextion:125]:   Firmware Version: 72
[08:01:23][C][nextion:126]:   Serial Number:    01234567890123456
[08:01:23][C][nextion:127]:   Flash Size:       16777216
[08:01:23][C][nextion:128]:   Wake On Touch:    True
[08:01:23][C][nextion:139]:   Start Up Page :      8
```

### Additional information

I will try to work in a fix for this, as I've recently made some changes on this area.
Any help would be very welcome. 😉

ugorur, 2023-11-28:
I have same issue too.

edwardtfn, 2023-11-28:
For your info, a super simple code like this works fine:

```yaml
display:
  - id: nextion_display
    platform: nextion

esphome:
  name: office-workstation-panel

esp32:
  board: esp32dev

wifi:
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_password

ota:
  password: !secret wifi_password

uart:
  id: nextion_uart
  baud_rate: 115200
  tx_pin: 16
  rx_pin: 17

logger:
  level: very_verbose

api:

time:
  - platform: homeassistant
    on_time:
      - seconds: /5
        then:
          - lambda: |-
              ESP_LOGD("DEBUG", "nextion_uart->available(): %i", nextion_uart->available());
              ESP_LOGD("DEBUG", "nextion_display->available(): %i", nextion_display->available());
              //ESP_LOGD("DEBUG", "nextion_display->is_detected(): %s", nextion_display->is_detected() ? "True" : "False");
              ESP_LOGD("DEBUG", "nextion_display->is_setup(): %s", nextion_display->is_setup() ? "True" : "False");
```

BigBobbas, 2023-11-28:
i've updated my nspanel to 2023.11.6 and appears to be working ok for me. using blackymas v4.1.1 blueprint

edwardtfn, 2023-11-28:
@BigBobbas, have you applied any change to the code or the same code using before is working now with 2023.11.6?
If you clean your build files and try to install again, does it still working?

edwardtfn, 2023-11-28:
@ugorur, are you also using Blakymas NSPANEL?

I believe the problem is not with ESPHome 2023.11.6, but that nspanel project (where I'm a contributor) was using Nextion from a PR which got updated yesterday, causing the issue.

BigBobbas, 2023-11-28:
> @BigBobbas, have you applied any change to the code or the same code using before is working now with 2023.11.6? If you clean your build files and try to install again, does it still working?

I have just cleaned build and re-flashed and all still ok. I didn't change any code I just flashed after updating the addon. when i reflashed after cleaning I used esphome cli and still all good.

BigBobbas, 2023-11-28:
> @BigBobbas, have you applied any change to the code or the same code using before is working now with 2023.11.6? If you clean your build files and try to install again, does it still working?

this is my config if that helps.
```
substitutions:
  ###### CHANGE ME START ######
  device_name: "kitchenpanel" 
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

  nextion_update_url: "http://192.168.1.8:8123/local/nspanel_eu.tft"

  ##### addon-configuration #####
  ## addon_climate ##
  # addon_climate_heater_relay: "1" # possible values: 1/2

  ##### CHANGE ME END #####

packages:
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files:
      - nspanel_esphome.yaml # Core package
      # - nspanel_esphome_addon_climate_cool.yaml # activate for local climate (cooling) control
      # - nspanel_esphome_addon_climate_heat.yaml # activate for local climate (heater) control
    refresh: 300s

##### My customization - Start #####
esp32:
  framework:
    type: esp-idf

external_components:
  - source: github://pr#5825
    components: [nextion]
    refresh: 1s

time:
  - id: !extend time_provider
    on_time:
      - seconds: /3
        then:
          - lambda: |-
              ESP_LOGD("DEBUG", "disp1->is_detected(): %s", disp1->is_detected() ? "True" : "False");
              ESP_LOGD("DEBUG", "disp1->is_setup(): %s", disp1->is_setup() ? "True" : "False");
##### My customization - End #####
```

ugorur, 2023-11-28:
> this is my config if that helps.
>
> ```
> substitutions:
>   ###### CHANGE ME START ######
>   device_name: "kitchenpanel"
>   wifi_ssid: !secret wifi_ssid
>   wifi_password: !secret wifi_password [...]

this fix my issue. thank you

edwardtfn, 2023-11-29:
This is working properly with ESPHome dev

closed 2023-11-29 by edwardtfn