esphome/issues #5162: after updating NSPanel

skipper79, 2023-11-28:
### The problem

This morning I updated my [NSPanel](https://github.com/Blackymas/NSPanel_HA_Blueprint/tree/main) with ESPHome version from 2023.11.4 to 2023.11.6. Unforfunatly the Nspanel does not start up; still got the status ‘Initializing…’ Also after restart and power off the NSPanel.

TFT version: 4.1
ESPHome :4.1
Blueprint 4.1
[image]

After validating 'nspanel.yaml' and install I get the following error:

[11:49:32][E][on_boot:040]: No response from Nextion display 
[11:49:32][D][on_boot:041]: Turn off Nextion 
[11:49:32][D][switch:016]: 'nspanel Screen Power' Turning OFF. 
[11:49:32][D][switch:055]: 'nspanel Screen Power': Sending state OFF 
[11:49:34][D][on_boot:044]: Turn on Nextion 
[11:49:34][D][switch:012]: 'nspanel Screen Power' Turning ON. 
[11:49:34][D][switch:055]: 'nspanel Screen Power': Sending state ON
 [11:49:34][W][component:214]: Component esphome.coroutine took a long time for an operation (1.53 s). [
11:49:34][W][component:215]: Components should block for at most 20-30ms. 
[11:49:54][E][on_boot:054]: No response from Nextion display

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Board

Sonoff

### Example YAML snippet

```yaml
substitutions:

###### CHANGE ME START ######

 device_name: "nspanel" 
 wifi_ssid: !secret wifi_ssid
 wifi_password: !secret wifi_password

 nextion_update_url: "http://192.168.1.136:8123/local/nspanel_eu.tft" # URL to local tft File

##### CHANGE ME END #####

##### DO NOT CHANGE ANYTHING! #####

packages:
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: nspanel_esphome.yaml
    refresh: 300s
```

skipper79, 2023-11-28:
Also after a second try, where the installation looks well, the same error
[image]

edwardtfn, 2023-11-28:
This is the same as #5161

edwardtfn, 2023-11-29:
This is working fine with ESPHome dev. The error happens only when ESPHome prod is trying to run the Nextion component from some PR.
I think this should be closed as the reference to that PR was removed from NSPANEL project and the project work is fine with ESPHome dev.