esphome/issues #5165: [hidden] is an invalid option for [wifi]. Please check the indentation.

DRXAquosus, 2023-11-29:
### The problem

Trying to configure connecting to a hidden wifi network, config editor shows the error "[hidden] is an invalid option for [wifi]. Please check the indentation." whenever I add the "hidden" parameter to the wifi section.  

I also tried with the multiple network method and while it will accept the hidden parameter it gives an error for the fast_connect parameter. ([fast_connect] is an invalid option for [0]. Please check the indentation.)

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

OS 2023.11.3

### What platform are you using?

ESP32

### Board

Atom Echo

### Component causing the issue

Wifi

### Example YAML snippet

```yaml
Single Network Configuration - Produces the "[hidden] is an invalid option for [wifi]. Please check the indentation." error:
wifi:
  ssid: wifi-name1
  password: Password
  fast_connect: True
  hidden: True
  domain: .custom.domain.us
  enable_btm: true
  enable_rrm: true
  manual_ip:
    static_ip: 10.x.x.xxx
    subnet: 255.255.255.0
    gateway: 10.x.x.x
    dns1: 10.x.xx.xx
    dns2: 10.x.xx.xx

Multiple Network Configuration - Produces the "[fast_connect] is an invalid option for [0]. Please check the indentation." error
wifi:   
  domain: .custom.domain.us
  enable_btm: true
  enable_rrm: true
  networks:
  - ssid: wifi-name1
    fast_connect: True
    password: Password
    hidden: true
    manual_ip:
      static_ip: 10.x.x.xxx
      subnet: 255.255.255.0
      gateway: 10.x.x.x
      dns1: 10.x.xx.xx
      dns2: 10.x.xx.xx
  - ssid: wifi-name2
    password: Password
```

### Additional information

Single Network Configuration:
[image]

Multiple Network Configuration:
[image]

ssieb, 2023-11-29:
Those options are equivalent and the docs are clear about which one can be used in each place.

closed 2023-11-29 by ssieb

DRXAquosus, 2023-11-29:
Where exactly are the "docs are clear" about "hidden" and "fast_connect" being equivalent?

If anything the docs say the exact opposite, that anytime you use specify hidden you must also specify fast_connect.

>**fast_connect** (Optional, boolean): If enabled, directly connects to WiFi network without doing a full scan first. **This >is required for hidden networks** and can significantly improve connection times. Defaults to off. The downside is that >this option connects to the first network the ESP sees, even if that network is very far away and better ones are available.
>https://esphome.io/components/wifi.html#configuration-variables

You have also said this on issue #4713
>"The docs clearly explain that you need that option for hidden SSIDs and also the drawbacks of using it."

ssieb, 2023-11-29:
It's right there...  If you're using the single network config to connect to a hidden network, you have to use `fast_connect`.  And the downside is listed as well.  If you're using the multiple networks option, then you use `hidden` option.

DRXAquosus, 2023-11-29:
Even doing it that way it still will not connect.
When the ssid is broadcast it connects, when it is hidden the connection drops and does not get reestablished.

`
wifi:
  ssid: wifi-name
  password: password
  fast_connect: True
  domain: .custom.domain.us
  enable_btm: true
  enable_rrm: true
  manual_ip:
    static_ip: 10.x.x.110
    subnet: 255.255.255.0
    gateway: 10.x.x.1
    dns1: 10.x.xx.41
    dns2: 10.x.xx.42
`

ssieb, 2023-11-29:
I have tested it before and it works.  If you need more help, then come ask on discord.