esphome/issues #5167: Incorrect addition of the last 3 bytes of the device MAC address to the device name

ReliableBob, 2023-11-29:
### The problem

When using the function of adding the last 3 bytes of the Mac address to the device name, an error occurs.
The last 3 bytes of the MAC address are 71:05:2B, but the following is added to the device name: e1-710528
There is a clear mistake here. Instead of the symbol "B" the number "8" is used.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

ssieb, 2023-11-29:
Can you paste logs?

ReliableBob, 2023-11-30:
`rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:326]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:183]: Performing I2C bus recovery
[C][uart.arduino_esp32:077]: Setting up UART...
[C][pcf8574:010]: Setting up PCF8574...
[C][pcf8574:010]: Setting up PCF8574...
[D][number:012]: 'Long click delay': Sending state 1000.000000
[D][number:012]: 'Server IP 1': Sending state 0.000000
[D][number:012]: 'Server IP 2': Sending state 0.000000
[D][number:012]: 'Server IP 3': Sending state 0.000000
[D][number:012]: 'Server IP 4': Sending state 0.000000
[C][switch.gpio:011]: Setting up GPIO Switch 'LED Red'...
[D][switch:016]: 'LED Red' Turning OFF.
[D][switch:055]: 'LED Red': Sending state OFF
[D][switch:016]: 'LED Red' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'LED Green'...
[D][switch:016]: 'LED Green' Turning OFF.
[D][switch:055]: 'LED Green': Sending state OFF
[D][switch:016]: 'LED Green' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 1'...
[D][switch:016]: 'Relay 1' Turning OFF.
[D][switch:055]: 'Relay 1': Sending state OFF
[D][switch:016]: 'Relay 1' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 2'...
[D][switch:016]: 'Relay 2' Turning OFF.
[D][switch:055]: 'Relay 2': Sending state OFF
[D][switch:016]: 'Relay 2' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 3'...
[D][switch:016]: 'Relay 3' Turning OFF.
[D][switch:055]: 'Relay 3': Sending state OFF
[D][switch:016]: 'Relay 3' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 4'...
[D][switch:016]: 'Relay 4' Turning OFF.
[D][switch:055]: 'Relay 4': Sending state OFF
[D][switch:016]: 'Relay 4' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 5'...
[D][switch:016]: 'Relay 5' Turning OFF.
[D][switch:055]: 'Relay 5': Sending state OFF
[D][switch:016]: 'Relay 5' Turning OFF.
[C][dallas.sensor:032]: Setting up DallasComponent...
[C][ethernet:029]: Setting up Ethernet...
[D][sensor:094]: 'uptm': Sending state 2.03400 s with 0 decimals of accuracy
[D][binary_sensor:034]: 'FN button': Sending initial state OFF
[D][binary_sensor:034]: 'Input 1': Sending initial state OFF
[D][binary_sensor:034]: 'Input 3': Sending initial state OFF
[D][binary_sensor:034]: 'Input 4': Sending initial state OFF
[D][binary_sensor:034]: 'Input 2': Sending initial state OFF
[D][binary_sensor:034]: 'Input 5': Sending initial state OFF
[D][binary_sensor:034]: 'Input 6': Sending initial state OFF
[I][ethernet:138]: Starting ethernet connection
[D][dallas.sensor:143]: 'Dallas Temperature': Got Temperature=24.4°C
[D][sensor:094]: 'Dallas Temperature': Sending state 24.37500 °C with 2 decimals of accuracy
[I][ethernet:149]: Connected via Ethernet!
[C][ethernet:354]:   IP Address: 192.168.90.226
[C][ethernet:355]:   Hostname: 'jethub-e1-710528'
[C][ethernet:356]:   Subnet: 255.255.255.0
[C][ethernet:357]:   Gateway: 192.168.90.1
[C][ethernet:362]:   DNS1: 192.168.90.1
[C][ethernet:363]:   DNS2: 77.88.8.8
[... 177 lines cut ...]
[C][mqtt.text_sensor:024]:   State Topic: 'jethub-e1/sensor/controller_uptime/state'
[C][mqtt.number:033]: MQTT Number 'Long click delay':
[C][mqtt.number:034]:   State Topic: 'jethub-e1/number/long_click_delay/state'
[C][mqtt.number:033]: MQTT Number 'Server IP 1':
[C][mqtt.number:034]:   State Topic: 'jethub-e1/number/server_ip_1/state'
[C][mqtt.number:033]: MQTT Number 'Server IP 2':
[C][mqtt.number:034]:   State Topic: 'jethub-e1/number/server_ip_2/state'
[C][mqtt.number:033]: MQTT Number 'Server IP 3':
[C][mqtt.number:034]:   State Topic: 'jethub-e1/number/server_ip_3/state'
[C][mqtt.number:033]: MQTT Number 'Server IP 4':
[C][mqtt.number:034]:   State Topic: 'jethub-e1/number/server_ip_4/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'FN button':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/fn_button/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 1':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_1/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 2':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_2/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 3':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_3/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 4':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_4/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 5':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_5/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Input 6':
[C][mqtt.binary_sensor:023]:   State Topic: 'jethub-e1/binary_sensor/input_6/state'`

ssieb, 2023-11-30:
Paste your config.

ReliableBob, 2023-11-30:
substitutions:
  devicename: "jethub-e1"
  device_description: "JetHub E1 controller"
  ip: 10.200.1.12
  gateway: 10.200.1.254
esphome:
  name: $devicename
  comment: "${device_description}"
  platform: ESP32
  board: nodemcu-32s
  name_add_mac_suffix: true # This option adds MAC address to the end of the device name

## WARNING: Disable WiFi and captive_portal if Ethernet enabled
#wifi:
#  ssid: "***"
#  password: "***"
#
#  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "JetHub E1 Fallback Hotspot"
#    password: "***"
#
#captive_portal:

# Enable Ethernet
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  # manual_ip:
  #   static_ip: $ip
  #   gateway: $gateway
  #   subnet: 255.255.255.0
# Enable logging
logger:
  logs:
    component: NONE
    text_sensor: NONE
    uptm: NONE
# Enable Home Assistant API
# api:
#   encryption:
#     key: "HiCfFdPxgUv4Lgp3ijVyRPG79NG8ckRudFGqVIYuqvg="

# Enable OTA firmware update
ota:

http_request:
  useragent: esphome/device
  timeout: 10s

mqtt:
  broker: 10.200.1.1
  username: pressen
  password: !secret mqtt_testst_password
  # client_id: $devicename
  discovery: false
  topic_prefix: $devicename
  log_topic: ${devicename}/log
  
web_server:
  port: 80
  version: 2
  include_internal: false
  ota: true
  # log: false
  #local: true

# Enable I2C bus
i2c:
  sda: 5
  scl: 4
  frequency: 400kHz
  scan: true
  id: i2c1

# Enable UART1
uart:
  - tx_pin: 33
    rx_pin: 34
    baud_rate: 115200
    id: uart_1

# Enable I2C GPIO expanders
pcf8574:
  - id: cpu_gpio_exp
    address: 0x20
    pcf8575: true
  - id: mb_gpio_exp
    address: 0x22
    pcf8575: true

ssieb, 2023-11-30:
Ok, I'm pretty sure that the mac address that is used is the wifi one.  It's just an identifier, so that makes sense.

BartOpitz, 2024-02-27:
Yep, looks like it uses always wifi mac, regardless whether it connects via ethernet or wifi. I have this issue too, and it is a silly problem with mikrotik integration which cannot associate presence sensor with this device via MAC association... Would be great id there was a way to use ethernet MAC as identifier...

bartowl, 2024-02-27:
Yep, looks like it uses always wifi mac, regardless whether it connects via ethernet or wifi. I have this issue too, and it is a silly problem with mikrotik integration which cannot associate presence sensor with this device via MAC association... Would be great id there was a way to use ethernet MAC as identifier...

randybb, 2024-09-17:
So by default esp_read_mac() is just base MAC of ESP. Ethernet MAC = base_mac +3 to the last octet

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/misc_system_api.html#mac-address

victorclaessen, 2024-09-19:
So on ESP32, it looks like the Ethernet Info Text Sensor gets the MAC like this:
* There is a [class MACAddressEthernetInfo](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/components/ethernet_info/ethernet_info_text_sensor.h#L62) which [calls](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/components/ethernet_info/ethernet_info_text_sensor.h#L64C79-L64C107) function **get_eth_mac_address_pretty()**
* **get_eth_mac_address_pretty** [calls](https://github.com/esphome/esphome/blob/dev/esphome/components/ethernet/ethernet_component.cpp#L570)  the helper function **get_mac_address_raw**
* **get_mac_address_raw** [calls](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/core/helpers.cpp#L676C3-L676C28) the expressif function [**esp_efuse_mac_get_default**](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/core/helpers.cpp#L662C6-L662C25). 

For the ethernet MAC address, the expressif function [**esp_read_mac**](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/core/helpers.cpp#L662C6-L662C25) can be called, with argument [**ESP_MAC_ETH**](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/misc_system_api.html#_CPPv4N14esp_mac_type_t11ESP_MAC_ETHE).

It's not complicated, but unfortunately, I don't really know how the maintainers of this component would like to handle this. My suggestion is to add a type parameter to **get_mac_address_raw**, and then return the ethernet MAC if that's requested by the parameter value. It could even be extended to return the softAP MAC and bluetooth MAC, depending on the parameter value (and if those MACs are available on the specific platform).

Misiu, 2026-02-18:
I just flashed WT32-ETH01 with BTProxy using https://esphome.io/projects/?type=bluetooth&diy

The device got discovered by Home Assistant and it shows this info:

[image]

it shows: `20:43:A8:65:CA:AC`

but in logs I see: ` 20:43:A8:65:CA:AF`
I see the same MAC in HDCP panel:

[image]

and on my router devices page

```
INFO ESPHome 2026.1.5
INFO Reading configuration /config/esphome/wt32-eth01-65caac.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq/#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from wt32-eth01-65caac.local using esphome API
INFO Successfully resolved wt32-eth01-65caac.local in 0.133s
INFO Successfully connected to wt32-eth01-65caac @ 192.168.0.152 in 0.002s
INFO Successful handshake with wt32-eth01-65caac @ 192.168.0.152 in 0.072s
[17:48:42.166][I][app:210]: ESPHome version 2026.1.5 compiled on 2026-02-18 16:19:34 +0100
[17:48:42.170][I][app:217]: ESP32 Chip: ESP32 r3.1, 2 core(s)
[17:48:42.170][W][app:226]: Set minimum_chip_revision: "3.1" to reduce binary size
[17:48:42.174][C][logger:316]: Logger:
[17:48:42.174][C][logger:316]:   Max Level: DEBUG
[17:48:42.174][C][logger:316]:   Initial Level: DEBUG
[17:48:42.176][C][logger:322]:   Log Baud Rate: 115200
[17:48:42.176][C][logger:322]:   Hardware UART: UART0
[17:48:42.177][C][logger:332]:   Task Log Buffer Size: 768 bytes
[17:48:42.198][C][bluetooth_proxy:124]: Bluetooth Proxy:
[17:48:42.198][C][bluetooth_proxy:124]:   Active: YES
[17:48:42.198][C][bluetooth_proxy:124]:   Connections: 4
[17:48:42.201][C][safe_mode.button:014]: Safe Mode Button 'Safe Mode Boot'
[17:48:42.202][C][safe_mode.button:017]:   Icon: 'mdi:restart-alert'
[17:48:42.219][C][factory_reset.button:014]: Factory Reset Button 'Factory reset'
[17:48:42.225][C][factory_reset.button:017]:   Icon: 'mdi:restart-alert'
[17:48:42.226][C][esp32_ble:652]: BLE:
[17:48:42.226][C][esp32_ble:652]:   MAC address: 20:43:A8:65:CA:AE
[17:48:42.226][C][esp32_ble:652]:   IO Capability: none
[17:48:42.246][C][esp32_ble_tracker:650]: BLE Tracker:
[17:48:42.250][C][esp32_ble_tracker:651]:   Scan Duration: 300 s
[17:48:42.250][C][esp32_ble_tracker:651]:   Scan Interval: 320.0 ms
[17:48:42.250][C][esp32_ble_tracker:651]:   Scan Window: 30.0 ms
[17:48:42.250][C][esp32_ble_tracker:651]:   Scan Type: ACTIVE
[17:48:42.250][C][esp32_ble_tracker:651]:   Continuous Scanning: YES
[17:48:42.251][C][esp32_ble_tracker:659]:   Scanner State: RUNNING
[17:48:42.251][C][esp32_ble_tracker:659]:   Connecting: 0, discovered: 0, disconnecting: 0
[17:48:42.254][C][bluetooth_proxy.connection:091]: BLE Connection:
[17:48:42.272][C][esp32_ble_client:071]:   Address: 
[17:48:42.272][C][esp32_ble_client:071]:   Auto-Connect: FALSE
[17:48:42.272][C][esp32_ble_client:071]:   State: IDLE
[17:48:42.273][C][bluetooth_proxy.connection:091]: BLE Connection:
[17:48:42.275][C][esp32_ble_client:071]:   Address: 
[17:48:42.275][C][esp32_ble_client:071]:   Auto-Connect: FALSE
[17:48:42.275][C][esp32_ble_client:071]:   State: IDLE
[17:48:42.279][C][bluetooth_proxy.connection:091]: BLE Connection:
[17:48:42.280][C][esp32_ble_client:071]:   Address: 
[17:48:42.280][C][esp32_ble_client:071]:   Auto-Connect: FALSE
[17:48:42.280][C][esp32_ble_client:071]:   State: IDLE
[17:48:42.296][C][bluetooth_proxy.connection:091]: BLE Connection:
[17:48:42.305][C][esp32_ble_client:071]:   Address: 
[17:48:42.305][C][esp32_ble_client:071]:   Auto-Connect: FALSE
[17:48:42.305][C][esp32_ble_client:071]:   State: IDLE
[17:48:42.306][C][ethernet:388]: Ethernet:
[17:48:42.306][C][ethernet:388]:   Connected: YES
[17:48:42.308][C][ethernet:680]:   IP Address: 192.168.0.152
[17:48:42.308][C][ethernet:680]:   Hostname: 'wt32-eth01-65caac'
[17:48:42.308][C][ethernet:680]:   Subnet: 255.255.255.0
[17:48:42.308][C][ethernet:680]:   Gateway: 192.168.0.1
[17:48:42.308][C][ethernet:680]:   DNS1: 192.168.0.1
[17:48:42.308][C][ethernet:680]:   DNS2: 0.0.0.0
[17:48:42.323][C][ethernet:702]:   MAC Address: 20:43:A8:65:CA:AF
[17:48:42.323][C][ethernet:702]:   Is Full Duplex: YES
[17:48:42.323][C][ethernet:702]:   Link Speed: 100
[17:48:42.323][C][ethernet:414]:   Power Pin: 16
[17:48:42.326][C][ethernet:416]:   CLK Pin: 0
[17:48:42.326][C][ethernet:416]:   MDC Pin: 23
[17:48:42.326][C][ethernet:416]:   MDIO Pin: 18
[17:48:42.326][C][ethernet:416]:   PHY addr: 1
[17:48:42.346][C][ethernet:423]:   Type: LAN8720
[17:48:42.346][C][esphome.ota:075]: Over-The-Air updates:
[17:48:42.346][C][esphome.ota:075]:   Address: wt32-eth01-65caac.local:3232
[17:48:42.346][C][esphome.ota:075]:   Version: 2
[17:48:42.364][C][safe_mode:021]: Safe Mode:
[17:48:42.364][C][safe_mode:021]:   Successful after: 60s
[17:48:42.364][C][safe_mode:021]:   Invoke after: 10 attempts
[17:48:42.364][C][safe_mode:021]:   Duration: 300s
[17:48:42.371][C][safe_mode:038]:   Bootloader rollback: support unknown
[17:48:42.372][C][api:221]: Server:
[17:48:42.372][C][api:221]:   Address: wt32-eth01-65caac.local:6053
[17:48:42.372][C][api:221]:   Listen backlog: 4
[17:48:42.372][C][api:221]:   Max connections: 8
[17:48:42.375][C][api:228]:   Noise encryption: YES
[17:48:42.418][C][mdns:177]: mDNS:
[17:48:42.418][C][mdns:177]:   Hostname: wt32-eth01-65caac
[17:48:51.281][D][esp32_ble_tracker:789]: Scan complete, set scanner state to IDLE.
[17:48:51.292][D][esp32_ble_tracker:220]: Starting scan, set scanner state to STARTING.
```

my config is very simple:
```
substitutions:
  name: wt32-eth01-65caac
  friendly_name: Bluetooth Proxy Biuro
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/wt32/wt32-eth01.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: XXX

web_server:
  port: 80
  version: 1
```