esphome/issues #5168: Problem accessing a device by host name when adding the last 3 bytes of the Mac address to the device name.

ReliableBob, 2023-11-29:
### The problem

After adding the last 3 bytes of the MAC address to the device name, the host name of the device is formed from the device name and the last 3 bytes of the MAC address.
It is displayed correctly in the router. For example jethub-e1-710528.
However, if you click on the “Visit” button of this device in the dashboard, the device’s web page will be unavailable because the browser will try a page named http://jethub-e1.local/, without the last 3 bytes of the Mac address.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

ssieb, 2023-11-29:
You shouldn't be using that option.  The docs even say that there will be a problem if you do that.
https://esphome.io/components/esphome#adding-the-mac-address-as-a-suffix-to-the-device-name

ssieb, 2023-11-30:
That option is clearly described as being for OEM providers of hardware.  And there's this:
End users will need to create an individual YAML config file if they want to OTA update the devices in the future.

It should have a bigger warning, but it is there.