esphome/issues #5179: Can't compile esp32 with esp-idf

jorisjansen31, 2023-12-01:
### The problem

I can't compile a shelly plus 1pm with esp-idf, with arduino it compiles but doesn't work.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core: 2023.11.3, supervisor: 2023.11.6, OS: 11.1

### What platform are you using?

ESP32-IDF

### Board

shelly plus 1PM

### Component causing the issue

c make?

### Example YAML snippet

```yaml
esphome:
  name: shelly-test
  friendly_name: shelly test

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

# Enable logging
logger:

# Enable Home Assistant API
api:
#  encryption:
#    key: !secret api_encryption_key

ota:
#  password: !secret ota_password
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/shelly-test.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing shelly-test (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/build/shelly-test/.pioenvs/shelly-test/CMakeFiles/CMakeOutput.log".
See also "/data/build/shelly-test/.pioenvs/shelly-test/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/shelly-test/.pioenvs/shelly-test/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_fc13e && [1/2] Building C object CMakeFiles/cmTC_fc13e.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_fc13e
    FAILED: cmTC_fc13e 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_fc13e.dir/testCCompiler.c.obj  -o cmTC_fc13e   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 25.46 seconds =========================
```

czikocz, 2023-12-01:
Same problem for same device I have, posted few days ago. There are more users with this issue: [#5154](https://github.com/esphome/issues/issues/5154)

If you need temporary fix, compilation will work on esphome for Windows, HA will then connect without problem to the flashed shelly.

czikocz, 2023-12-02:
1. Use this manual to install esphome on Windows machine: [Installing ESPHome Manually](https://esphome.io/guides/installing_esphome.html), then in command line start it using command "esphome dashboard config/" (you might need to create the config folder in some location). If you know ways of python, using venv enviroment is always prefferable.
2. Create yaml configuration in the config folder for your device (just copy from HA instance)
3. Open dashboard localy (http://localhost:6052/)
4. Compile and flash shelly using this local esphome instance as you normally would
5. Then make sure you have the same device config in your HA esphome config folder and it will connect normally, because it does not distinguish where was the device's code compiled/installed. I can see the device online in both local windows esphome instance and HA esphome instance at the same time.

czikocz, 2023-12-02:
> > compilation will work on esphome for Windows
> 
> What do you mean exactly? Downloading the install file and then compile it on esphome web thingy, and then let it connect to home assistant esphome?

1. Use this manual to install esphome on Windows machine: [Installing ESPHome Manually](https://esphome.io/guides/installing_esphome.html), then in command line start it using command "esphome dashboard config/" (you might need to create the config folder in some location). If you know ways of python, using venv enviroment is always prefferable.
2. Create yaml configuration in the config folder for your device (just copy from HA instance)
3. Open dashboard localy (http://localhost:6052/)
4. Compile and flash shelly using this local esphome instance as you normally would
5. Then make sure you have the same device config in your HA esphome config folder and it will connect normally, because it does not distinguish where was the device's code compiled/installed. I can see the device online in both local windows esphome instance and HA esphome instance at the same time.

ssieb, 2024-03-08:
Do you have a 32-bit install?  Run `uname -m` in ssh if you don't know.