esphome/issues #5183: Cannot compile m5stack-atom-echo voice assistant example

nmeachen, 2023-12-02:
### The problem

Trying to compile the m5stack-atom voice assistant example using an aarch64-based container on a Raspberry Pi 4B running bookworm fails.

### Which version of ESPHome has the issue?

ESPHome 2023.11.6 

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

m5stack-atom

### Example YAML snippet

```yaml
---
esphome:
  name: family-room-echo
  friendly_name: family-room-echo
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.11.1

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
api:
ota:

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

wifi:
  on_connect:
    - delay: 5s # Gives time for improv results to be transmitted
    - ble.disable:
  on_disconnect:
    - ble.enable:
  ap:

improv_serial:

esp32_improv:
  authorizer: none

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
[... 221 lines cut ...]
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/config/.esphome/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 1232:
    ensure_python_venv_available()
  File "/config/.esphome/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 1208:
    _create_venv(venv_dir)
  File "/config/.esphome/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 1200:
    assert os.path.isfile(
========================= [FAILED] Took 15.95 seconds =========================
```

nmeachen, 2023-12-02:
I stripped it back and it still fails, seems like only the arduino framework compiles on an aarch64 esphome container.

---
esphome:
  name: family-room-echo
  friendly_name: family-room-echo
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.11.1

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
api:
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

nmeachen, 2023-12-05:
This is effectively the same error as #4992. If you don't attempt to override any of the versions it still does not compile. Seems like a container construction related problem as creating a pure Python environment directly in Ubuntu compiles this ESP-IDF project without any issues.

nmeachen, 2023-12-28:
This issue seems to be caused by the value of PROJECT_CORE_DIR which in the container for my project is /config/config/.esphome/build/family-room-echo/"/"/.platformio 

The embedded double-quotes in the path prevent the virtual environment from being created. If I patch /usr/local/lib/python3.11/dist-packages/platformio/builder/main.py to escape the double-quotes when creating the venv it gets past the pip not existing error but fails later on.

When comparing the container run to a local Python installation the .platformio and venv folder gets created directly off the folder where the dashboard is run from. In the container they are created under the device specific build folder

/config/config/.esphome/build/family-room-echo/\"/\"

pmffromspace, 2024-01-21:
Just for the people finding this issue because they wnat to protect the atoms device api with a password:

I did following workaround:

1. install a VM (vor example multipass is easy for that) (I used ubuntu)
2. run following commands:
```
# creating VM
multipass launch -n esphome --disk 10G

multipass shell esphome

# install esphome in the VM
 
sudo apt update && sudo apt upgrade -y 
sudo apt install python3 -y
sudo apt install python3-pip -y
sudo apt install python3.10-venv

pip3 install wheel
pip3 install esphome

# logout and login so you can actually use esphome command
exit
mutlipass shell esphome
```

# create the atom-esphome-yaml:
nano atom.yaml 
# -> paste in the code, add the api stuff you want (easily generate a api key with the web dashboard running in Docker)
```
api:
  encryption:
    key: "YOUR_API_KEY"

ota:
  password: "YOUR_OTA_PASSWORD"
```

# add to the wifi part the ip of your already install atom in you wifi:
# (change static_ip and gateway to your routers config)

```
wifi:
  on_connect:
    - delay: 5s # Gives time for improv results to be transmitted
    - ble.disable:
  on_disconnect:
    - ble.enable:
  ap:
  manual_ip:
    static_ip: 10.0.1.190
    gateway: 10.0.1.1
    subnet: 255.255.255.0

```

## update your atom device via wifi:
```
esphome run atom.yaml
```

# optionally:
you can just save your atom.yaml file as a device in the esphome dashboard, so you can further use it when the docker container gets fixed.. ;)

This is not best practice, but it works.

clydebarrow, 2024-03-27:
This fixed a similar problem for me on Ubuntu 20.04:

```
sudo apt install python3.12-venv
```