esphome/issues #5187: ESP32 S3 keeps on rebooting when compiled with esp-idf framework

jmr61, 2023-12-02:
### The problem

ESP32 S3 keeps on rebooting when compiled with esp-idf framework

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP32-IDF

### Board

Espressif ESP32-S3-DEVKITC-1 (N8R2)

### Component causing the issue

esp-idf framework

### Example YAML snippet

```yaml
esphome:
  name: test28
  friendly_name: test28

esp32:
  board: esp32-s3-devkitc-1
  framework: 
    type: esp-idf
    version: recommended    
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y    

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "tiLw6RJi6W/WwkNIsUvOz9b/6mp89Uw3yiqXfONhxLg="

ota:
  password: "72ddd404236d974bcd5a893fd77a40fa"

wifi:
  id: wifi_id
  ssid: "LaSama"
  password: "LaSamaritaine!!!"

  manual_ip:
    static_ip: 192.168.1.206
    gateway: 192.168.1.254
    subnet: 255.255.0.0   

captive_portal:

web_server:
  port: 80

switch:
- platform: gpio
  name: "sw"
  id: sw
  pin: GPIO40
```

### Anything in the logs that might be useful for us?

```txt
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78
....................
```

### Additional information

it was working up to 2023.11.3 ( not sure, but it was working ) and i noticed the issue after upgrading to 2023.11.4
Same issue with another board with ESP32-S3 N16R2

randybb, 2023-12-02:
Not aware about change on esphome side, so I guess that something has been changed on in idf or in platformio. Anyway, this is needed at least for ESP32-S3R8:
```yaml
esphome:
  platformio_options:
    board_build.flash_mode: dio
```

closed 2023-12-03 by jmr61

echavet, 2024-03-11:
great thanks @randybb and @jmr61 for this solution and exposed pb.