esphome/issues #5195: Cant update ESPHome for waveshare display

1liminal1, 2023-12-06:
### The problem

I understand there was an breaking changes in 2023.6 for fonts ect. But its really not clear how to fix this? 

I have pasted my YAML and the compile errors, but please let me know if anything else is needed :) 

Thank you 

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP32

### Board

Dev Board 

### Component causing the issue

Not sure

### Example YAML snippet

```yaml
# Global variables for detecting if the display needs to be refreshed. (Thanks @paviro!)
globals:
  - id: data_updated
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: initial_data_received
    type: bool
    restore_value: no
    initial_value: 'false'

external_components:
  - source:
      type: git
      url: https://github.com/guardmedia/esphome.git
      ref: waveshare-color-5
    components: [ waveshare_epaper, display ]

# Include custom fonts
font:
  - file: fonts/materialdesignicons-webfont.ttf
    id: icon_font
    size: 50
    glyphs:
      - "\U000F0590" # weather-cloudy
      - "\U000F0F2F" # weather-cloudy-alert
      - "\U000F0E6E" # weather-cloudy-arrow-right
      - "\U000F0591" # weather-fog
      - "\U000F0592" # weather-hail
      - "\U000F0F30" # weather-hazy
      - "\U000F0898" # weather-hurricane
      - "\U000F0593" # weather-lightning
      - "\U000F067E" # weather-lightning-rainy
      - "\U000F0594" # weather-night
      - "\U000F0F31" # weather-night-partly-cloudy
[... 298 lines cut ...]
                                                                                      ^~~~~~~
src/esphome/components/display/display_buffer.h:283:69: error: 'esphome::time::ESPTime' has not been declared
   void strftime(int x, int y, Font *font, const char *format, time::ESPTime time)
                                                                     ^~~~~~~
*** [.pioenvs/to-do/src/esphome/components/font/font.cpp.o] Error 1
src/esphome/components/display/display_buffer.cpp:378:36: error: 'esphome::time::ESPTime' has not been declared
                              time::ESPTime time) {
                                    ^~~~~~~
src/esphome/components/display/display_buffer.cpp: In member function 'void esphome::display::DisplayBuffer::strftime(int, int, esphome::display::Font*, esphome::Color, esphome::display::TextAlign, const char*, int)':
src/esphome/components/display/display_buffer.cpp:380:21: error: request for member 'strftime' in 'time', which is of non-class type 'int'
   size_t ret = time.strftime(buffer, sizeof(buffer), format);
                     ^~~~~~~~
src/esphome/components/display/display_buffer.cpp: At global scope:
src/esphome/components/display/display_buffer.cpp:384:95: error: 'esphome::time::ESPTime' has not been declared
 void DisplayBuffer::strftime(int x, int y, Font *font, Color color, const char *format, time::ESPTime time) {
                                                                                               ^~~~~~~
src/esphome/components/display/display_buffer.cpp:387:99: error: 'esphome::time::ESPTime' has not been declared
 void DisplayBuffer::strftime(int x, int y, Font *font, TextAlign align, const char *format, time::ESPTime time) {
                                                                                                   ^~~~~~~
src/esphome/components/display/display_buffer.cpp:390:82: error: 'esphome::time::ESPTime' has not been declared
 void DisplayBuffer::strftime(int x, int y, Font *font, const char *format, time::ESPTime time) {
                                                                                  ^~~~~~~
*** [.pioenvs/to-do/src/esphome/components/display/display_buffer.cpp.o] Error 1
========================== [FAILED] Took 5.11 seconds ===================
```

ssieb, 2023-12-06:
github issues are not for external components.  Come ask on discord.

closed 2023-12-06 by ssieb

1liminal1, 2023-12-06:
Thats what it was :) :) thank you so much! 

I have no idea if the screen works lol, will have to check when I get home. But remove that (the link is 404, might be gone) and I could update thanks again