esphome/issues #5199: mlx90614 reports invalid temperature and nan humidity

khenderick, 2023-12-06:
### The problem

The build-in component `mlx90614` sometimes reports an invalid temperature and `nan` humidity. Re-installing the device might resolve the issue sometimes. Once my device has invalid readings, I need to keep reinstalling the same code again until it is solved.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

mlx90614

### Example YAML snippet

```yaml
...
i2c:
  sda: 22
  scl: 19
  frequency: 100kHz

sensor:
  - platform: mlx90614
    ambient:
      name: "IR ambient temperature"
      unit_of_measurement: °C
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 4
            send_every: 2
    object:
      name: "IR object temperature"
      unit_of_measurement: °C
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 4
            send_every: 2
...
```
```

### Anything in the logs that might be useful for us?

```txt
[10:08:21][D][mlx90614:112]: Got Temperature=-124.2°C Ambient=nan°C
[10:08:21][D][sensor:094]: 'IR object temperature': Sending state -124.18999 °C with 2 decimals of accuracy
```
```

### Additional information

I'm using an M5Stack Atom and their [mlx90614](https://shop.m5stack.com/products/ncir-sensor-unit) module.

tommy4st, 2023-12-17:
I had got the same problem with version 2023.10.6 after running the sensor for some time. I upgraded to 2023.11.6 and the problem persisted, so I pulled the plug and plugged it in again, now it's working properly once more.

It's actually a "nan" ambient temperature and not humidity.

Details regarding my hardware/software:
```
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
```

khenderick, 2024-04-16:
Unfortunately I still have this issue, so I'd like to keep it open and eventually fixed if possible.

jandechent, 2024-08-18:
Remove the filters, for me that works. Although this is of course a core function.

jandechent, 2024-08-18:
The MLX90393 integration has a much richer sensor.py. In particular it has a property `CONF_FILTER`. Maybe adding is not ultra complicated, but I dont have the experience to do that.

jandechent, 2024-08-20:
Actually, I had previously reported I have the same issue. Then suddenly it worked and then suddenly not. So I went to the hardware and one of the I2C wire connectors was slightly loose. I replaced my wiring all works since then. Or it was because I disconnected everything from power. Not sure, point is, the software seems to work perfectly.

rjt-rockx, 2025-04-10:
I've been running into the same issues as well, ESP32 DOIT DevKit v1 with both the Arduino as well as the ESP-IDF frameworks have a chance to trigger these invalid readings after OTA updates. My guess is that something's throwing the I2C communication with the MLX90614 off, which causes it to latch onto a persistent error state that only goes away with a full power cycle. 

I've been able to replicate this on multiple different wiring setups so hardware/wiring issues likely aren't the cause here. Soft-rebooting the ESP32 doesn't help fix the sensor either. Disconnecting and reconnecting the 3V3 line to the sensor is a quick way of testing/reproducing/temporarily fixing this issue.

closed 2026-09-10 by jesserockz

jandechent, 2026-09-10:
Thank you! Just in time before winter so i get my wood stove temperature right.