esphome/issues #5201: ADC is not working on BK7231n modules

redskinhu, 2023-12-06:
### The problem

I have couple of BK7231n modules in different devices.
I tried to use the ADC but I got incorrect and constants (2.048V) values independently the voltage on the ADC on all devices.

It looks the ADC is not working.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

BK72XX

### Board

BK7231n / CB2S CBU

### Component causing the issue

adc

### Example YAML snippet

```yaml
sensor:
  - platform: adc
    pin: ADC3
    name: "${upper_devicename} VCC"
    unit_of_measurement: "V"
```

### Anything in the logs that might be useful for us?

```txt
[21:06:42][D][debug:076]: ESPHome version 2023.11.6
[21:06:42][D][debug:080]: Free Heap Size: 27336 bytes
[21:06:42][D][debug:356]: LibreTiny Version: 1.4.1
[21:06:42][D][debug:357]: Chip: BK7231N (7b1c) @ 120 MHz
[21:06:42][D][debug:358]: Chip ID: 0x0652DD
[21:06:42][D][debug:359]: Board: generic-bk7231n-qfn32-tuya
[21:06:42][D][debug:360]: Flash: 2048 KiB / RAM: 256 KiB
[21:06:42][D][debug:361]: Reset Reason: SW Reboot
[21:06:42][D][text_sensor:064]: 'Door Sensor BK7231n Reset Reason': Sending state 'SW Reboot'
[21:06:48][D][sensor:093]: 'Door Sensor BK7231n VCC': Sending state 2.04800 V with 2 decimals of accuracy
[21:06:58][D][sensor:093]: 'Door Sensor BK7231n VCC': Sending state 2.04800 V with 2 decimals of accuracy
[21:07:08][D][sensor:093]: 'Door Sensor BK7231n VCC': Sending state 2.04800 V with 2 decimals of accuracy
[21:07:18][D][sensor:093]: 'Door Sensor BK7231n VCC': Sending state 2.04800 V with 2 decimals of accuracy
[21:07:28][D][sensor:093]: 'Door Sensor BK7231n VCC': Sending state 2.04800 V with 2 decimals of accuracy
```

szupi-ipuzs, 2023-12-15:
@redskinhu there's a pending fix libretiny-eu/libretiny#220, would you like to test it?

redskinhu, 2023-12-15:
Hello @szupi-ipuzs 

I installed this: v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Dec 15 2023 19:14:29, GCC 10.3.1 (-O1) through ESPHome.

ADC behavior has not changed, but I think this is not the right way to build the FW with the modified libretiny.

Any advice?

Thx

redskinhu, 2024-05-28:
Latest ESPhome is installed, still have the same issue.

Log:

INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/bedroom_door-bk7231n.yaml...
WARNING 'bedroom_door_bk7231n': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Europe/Dublin'
INFO Starting log output from 192.168.2.187 using esphome API
INFO Successfully connected to bedroom_door_bk7231n @ 192.168.2.187 in 0.011s
INFO Successful handshake with bedroom_door_bk7231n @ 192.168.2.187 in 0.223s
[16:23:15][I][app:100]: ESPHome version 2024.5.4 compiled on May 28 2024, 13:39:14
[16:23:15][C][status_led:019]: Status LED:
[16:23:15][C][status_led:020]:   Pin: 15
[16:23:15][C][wifi:580]: WiFi:
[16:23:15][C][wifi:408]:   Local MAC: 18:DE:50:DD:52:06
[16:23:15][C][wifi:413]:   SSID: [redacted]
[16:23:15][C][wifi:416]:   IP Address: 192.168.2.187
[16:23:15][C][wifi:419]:   BSSID: [redacted]
[16:23:15][C][wifi:421]:   Hostname: 'bedroom_door_bk7231n'
[16:23:15][C][wifi:423]:   Signal strength: -58 dB ▂▄▆█
[16:23:15][C][wifi:427]:   Channel: 1
[16:23:15][C][wifi:428]:   Subnet: 255.255.252.0
[16:23:15][C][wifi:429]:   Gateway: 192.168.0.1
[16:23:15][C][wifi:430]:   DNS1: 192.168.1.101
[16:23:15][C][wifi:431]:   DNS2: 192.168.1.101
[16:23:15][C][logger:185]: Logger:
[16:23:15][C][logger:186]:   Level: DEBUG
[16:23:15][C][logger:188]:   Log Baud Rate: 115200
[16:23:15][C][logger:189]:   Hardware UART: UART2
[16:23:15][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bedroom Door Sensor'
[16:23:15][C][gpio.binary_sensor:015]:   Device Class: 'door'
[16:23:15][C][gpio.binary_sensor:016]:   Pin: 22
[16:23:15][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bedroom Door Button'
[16:23:15][C][gpio.binary_sensor:016]:   Pin: 14
[16:23:15][C][uptime.sensor:031]: Uptime Sensor 'Bedroom Door Uptime'
[16:23:15][C][uptime.sensor:031]:   Device Class: 'duration'
[16:23:15][C][uptime.sensor:031]:   State Class: 'total_increasing'
[16:23:15][C][uptime.sensor:031]:   Unit of Measurement: 's'
[16:23:15][C][uptime.sensor:031]:   Accuracy Decimals: 0
[16:23:15][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[16:23:15][C][homeassistant.time:010]: Home Assistant Time:
[16:23:15][C][homeassistant.time:011]:   Timezone: 'IST-1GMT0,M10.5.0,M3.5.0/1'
[16:23:15][C][restart.button:017]: Restart Button 'Bedroom Door Restart'
[16:23:15][C][status:034]: Status Binary Sensor 'Bedroom Door Status'
[16:23:15][C][status:034]:   Device Class: 'connectivity'
[16:23:15][C][adc:097]: ADC Sensor 'Bedroom Door VCC'
[16:23:15][C][adc:097]:   Device Class: 'voltage'
[16:23:15][C][adc:097]:   State Class: 'measurement'
[16:23:15][C][adc:097]:   Unit of Measurement: 'V'
[16:23:15][C][adc:097]:   Accuracy Decimals: 2
[16:23:15][C][adc:097]:   Icon: 'mdi:current-dc'
[16:23:15][C][adc:102]:   Pin: 23
[16:23:15][C][adc:142]:   Update Interval: 300.0s
[16:23:15][C][captive_portal:088]: Captive Portal:
[16:23:15][C][web_server:173]: Web Server:
[16:23:15][C][web_server:174]:   Address: bedroom_door_bk7231n.local:80
[16:23:15][C][mdns:115]: mDNS:
[16:23:15][C][mdns:116]:   Hostname: bedroom_door_bk7231n
[16:23:15][C][ota:096]: Over-The-Air Updates:
[16:23:15][C][ota:097]:   Address: bedroom_door_bk7231n.local:8892
[16:23:15][C][ota:103]:   OTA version: 2.
[16:23:15][C][api:139]: API Server:
[16:23:15][C][api:140]:   Address: bedroom_door_bk7231n.local:6053
[16:23:15][C][api:142]:   Using noise encryption: YES
[16:23:15][C][wifi_info:009]: WifiInfo IPAddress 'Bedroom Door IP Address'
[16:23:16][C][wifi_info:009]:   Icon: 'mdi:ip-network'
[16:23:16][C][wifi_signal.sensor:009]: WiFi Signal 'Bedroom Door WiFi RSSI'
[16:23:16][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[16:23:16][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[16:23:16][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[16:23:16][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:23:16][C][lt.component:013]: LibreTiny:
[16:23:16][C][lt.component:014]:   Version: v1.5.1 on generic-bk7231n-qfn32-tuya, compiled at May 28 2024 13:06:30, GCC 10.3.1 (-O1)
[16:23:16][C][lt.component:015]:   Loglevel: 3
[16:23:16][D][text_sensor:064]: 'Bedroom Door Version': Sending state 'v1.5.1 on generic-bk7231n-qfn32-tuya, compiled at May 28 2024 13:06:30, GCC 10.3.1 (-O1)'
[16:23:16][C][debug:067]: Debug component:
[16:23:16][D][debug:079]: ESPHome version 2024.5.4
[16:23:16][D][debug:083]: Free Heap Size: 28104 bytes
[16:23:16][D][debug:359]: LibreTiny Version: 1.5.1
[16:23:16][D][debug:360]: Chip: BK7231N (7b1c) @ 120 MHz
[16:23:16][D][debug:361]: Chip ID: 0x0652DD
[16:23:16][D][debug:362]: Board: generic-bk7231n-qfn32-tuya
[16:23:16][D][debug:363]: Flash: 2048 KiB / RAM: 256 KiB
[16:23:16][D][debug:364]: Reset Reason: SW Reboot
[16:23:16][D][text_sensor:064]: 'Bedroom Door Reset Reason': Sending state 'SW Reboot'
[16:24:21][D][sensor:093]: 'Bedroom Door Uptime': Sending state 9867.11426 s with 0 decimals of accuracy
[16:24:44][D][sensor:093]: 'Bedroom Door WiFi RSSI': Sending state -57.00000 dBm with 0 decimals of accuracy
[16:24:46][D][sensor:093]: 'Bedroom Door VCC': Sending state 2.04800 V with 2 decimals of accuracy

ryan-ronnander, 2026-09-24:
I ran into the same flat 2.048 V on a BK7231N door sensor. On mine, the battery divider only turns on when another GPIO is driven high, so without it the ADC just sat at its max. Your board might have something similar, so check for a pin that switches the divider and turn it on briefly before each reading. After that you'll need a `multiply` filter for your divider ratio.

libretiny-eu/libretiny#416 also fixes the unit conversion, so readings come out in real volts.