esphome/issues #5205: UART component Hex sending weird behaviour

Peemouse, 2023-12-08:
### The problem

Hi,
I'm trying to trigger opening and closing on a motorized videoscreen using RS232.
I built an ESP32-ETH01 with UART to RS232 board.

Before connecting it to my screen (LUMENE Majestic HD200), I scoped the RS232 output using a USB to RS2332 converter and serial terminal in Visual Studio Code.

The issue is related to the HEX sending.
When I try to send like [0xFF, 0xEE] (uart.write: [0xFF, 0xCC]) , it always ends with sending FFFD FFFD no matter what.

However, if I send ASCII, it behaves as expected : uart.write: "Test" leads to 54 65 73 74

Thank you.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.0

### What platform are you using?

ESP32

### Board

ESP32-ETH01

### Component causing the issue

UART

### Example YAML snippet

```yaml
globals:
   - id: lastVPstate
     type: bool
     restore_value: no
     initial_value: '0'
     
esphome:
  name: esp-placard
  friendly_name: ESP Placard
  platform: ESP32
  board: esp-wrover-kit

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

uart:
  tx_pin: GPIO17
  rx_pin: GPIO5
  baud_rate: 2400
  data_bits: 8
  stop_bits: 1
  parity: NONE
  debug:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

# Video Screen management
cover:
  - platform: template
    name: "Ecran de projection"
    id: "videoscreen"
    open_action:
      #- uart.write: [0xFF, 0xEE, 0xEE, 0xEE, 0xDD]
      - uart.write: "Test"
    close_action:
      #- uart.write: [0xFF, 0xEE, 0xEE, 0xEE, 0xEE]
      - uart.write: [0xFF, 0xCC]
    stop_action:
      - uart.write: [0xFF, 0xEE, 0xEE, 0xEE, 0xCC]
    optimistic: true
```

### Anything in the logs that might be useful for us?

```txt
[11:51:54][D][cover:076]: 'Ecran de projection' - Setting
[11:51:54][D][cover:086]:   Command: CLOSE
[11:51:54][D][cover:170]: 'Ecran de projection' - Publishing:
[11:51:54][D][cover:178]:   State: CLOSED
[11:51:54][D][cover:186]:   Current Operation: IDLE
[11:51:54][D][uart_debug:114]: >>> FF:CC
[11:51:54][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:51:54][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[11:51:58][D][sensor:094]: 'ESP Placard Uptime': Sending state 1173.91199 s with 0 decimals of accuracy
[11:51:58][D][cover:076]: 'Ecran de projection' - Setting
[11:51:58][D][cover:086]:   Command: OPEN
[11:51:58][D][cover:170]: 'Ecran de projection' - Publishing:
[11:51:58][D][cover:176]:   State: OPEN
[11:51:58][D][cover:186]:   Current Operation: IDLE
[11:51:58][D][uart_debug:114]: >>> 54:65:73:74
```

### Additional information

[image]

ssieb, 2023-12-08:
That's a problem with your terminal or monitor.  0xfffd is the character used when the incoming character is unknown.  You need to use something that won't mangle the incoming data.

closed 2023-12-08 by ssieb

Peemouse, 2023-12-10:
Hi, thanks for the hint.
I tried SerialTool and I can see that the output is correct.
Time to focus elsewhere for debugging.

Isaacbg84, 2024-02-22:
> Hi, thanks for the hint. I tried SerialTool and I can see that the output is correct. Time to focus elsewhere for debugging.

Exactly the same issue. For what i have read is a problem with Visual code monitor. However my issue still remains. Any numer i send over 127 the monitor says : ff fd .   My problem is even if i disconnect the ESP32 from serial monitor the data does not activate a modbus device, but when i send the data using the serial monitor it does. Im using MAX232 connected to 3.3v and its eems working fine.

When i use micropython its the same thing. Could it be a ESP32 issue?

ssieb, 2024-02-22:
As mentioned earlier, characters over 127 are getting converted by your monitor.  Use something else that doesn't mangle the data.  Come ask on discord if you need more help with this device.