esphome/issues #5206: dashboard logs don't work for mqtt logging.

dbaarda, 2023-12-08:
### The problem

Using current dev branch, running `esphome dashboard ./` in my yaml configs directory and trying to view logs for one of my devices using mqtt for logging gives the error;

```
ERROR No remote or local logging method configured (api/mqtt/logger)
```

This used to work, and seems to have been broken since the last time I pulled updates.  I definitely have correctly configured mqtt logging, and I can see the logging in HomeAssistant using "listen to a topic". Interestingly running `esphome logs underhouse.yaml` directly I do see the logs just fine.

 I turned on dashboard verbose mode by running `esphome -v dashboard ./` which included the following line;

```
2023-12-09 00:21:40,506 INFO Running command 'esphome --dashboard logs ./underhouse.yaml --device 192.168.7.144'
```

Interestingly, if I run `esphome --dashboard logs ./underhouse.yaml --device 192.168.7.144` it gives the same error.  I still get the error if I remove the `--dashboard` argument . However, if I remove the `--device 192.168.7.144` it works.

### Which version of ESPHome has the issue?

2023.12.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

ESP32C3

### Component causing the issue

dashboard 

### Example YAML snippet

```yaml
substitutions:
  devicename: underhouse

esphome:
  name: ${devicename}
  platformio_options:
    board_build.flash_mode: dio

esp32:
# At some point the actual board might be supported.
#  board: adafruit_qt_py_esp32-c3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  variant: ESP32C3

mqtt:
  id: mqtt_client
  broker: 192.168.7.6
  username: "####"
  password: "####"
  discovery: true
  discovery_retain: true
  discovery_unique_id_generator: mac
  discovery_object_id_generator: device_name

# Enable logging
logger:
  level: VERBOSE
```

### Anything in the logs that might be useful for us?

```txt
$ esphome --dashboard logs underhouse.yaml --device 192.168.7.144
INFO ESPHome 2023.12.0-dev
INFO Reading configuration underhouse.yaml...
ERROR No remote or local logging method configured (api/mqtt/logger)
(venv)

$ esphome --dashboard logs underhouse.yaml
INFO ESPHome 2023.12.0-dev
INFO Reading configuration underhouse.yaml...
INFO Starting log output from underhouse/debug
INFO Connected to MQTT broker!
[00:47:29]\033[0;36m[D][sensor:094]: 'wifi strength': Sending state -80.00000 dBm with 0 decimals of accuracy\033[0m
^C(venv)
```

### Additional information

I had a bit of a dig around the code and it seems that error happens here;

https://github.com/esphome/esphome/blob/a6f1701902355489bdfcb7dee53671abf063d1aa/esphome/__main__.py#L366

And it gets to that line if `get_port_type(port)` doesn't return `MQTT`, and I'm guessing it doesn't if you have a `--device` argument.  Further digging suggests that maybe https://github.com/esphome/esphome/pull/3887 broke this about 7 months ago.

bdraco, 2023-12-21:
We need to drop `--device` if `api:` is missing from the yaml

bdraco, 2023-12-21:
I'm not sure that will actually help though

https://github.com/esphome/esphome/commit/6c3717a1775ea360397502dcd41be3c6c2eefdc1

bdraco, 2023-12-21:
I thought I broke this with the recent dashboard changes, but it looks like its been broken for a while, and since I don't have an MQTT setup, I'm not going to be in a position to fix it.

digiblur, 2023-12-21:
I can also confirm this is broken on 2023.12.  Rolling back my container to the 2023.11 tag fixes it.

bdraco, 2023-12-21:
Does it work if you run it as 

`esphome --dashboard logs underhouse.yaml --device OTA`?

If so https://github.com/esphome/esphome/commit/6c3717a1775ea360397502dcd41be3c6c2eefdc1 should fix it.  If not I'm not sure whats going on

digiblur, 2023-12-22:
> Does it work if you run it as
> 
> `esphome --dashboard logs underhouse.yaml --device OTA`?
> 
> If so [esphome/esphome@6c3717a](https://github.com/esphome/esphome/commit/6c3717a1775ea360397502dcd41be3c6c2eefdc1) should fix it. If not I'm not sure whats going on

Works...

```
/config# esphome --dashboard logs eh-mbath-ceiling.yaml --device OTA
INFO ESPHome 2023.12.1
INFO Reading configuration eh-mbath-ceiling.yaml...
INFO Starting log output from eh-mbath-ceiling/debug
INFO Connected to MQTT broker!
[21:29:44]\033[0;36m[D][sensor:093]: 'MBath Humidity': Sending state 54.84703 % with 1 decimals of accuracy\033[0m
[21:30:02]\033[0;36m[D][sht3xd:064]: Got temperature=23.08°C humidity=54.87%\033[0m
[21:30:02]\033[0;36m[D][sensor:093]: 'MBath Temp': Sending state 23.07736 °C with 1 decimals of accuracy\033[0m
[21:30:02]\033[0;36m[D][sensor:093]: 'MBath Humidity': Sending state 54.86992 % with 1 decimals of accuracy\033[0m
```

bdraco, 2023-12-22:
Great. Thanks for testing. I'll move that fix forward tomorrow

closed 2023-12-22 by jesserockz

dbaarda, 2023-12-22:
I just caught up with this and want to say thanks heaps bdraco@ for fixing this, and digiblur@ for testing that the fix works. Your work is well appreciated.