esphome/issues #5208: Error reading nfc card

Ayuus, 2023-12-08:
### The problem

I use ESPHome for reading nfc-cards, using Adonno Tag reader (https://github.com/adonno/tagreader).
```
[22:18:17][I][app:102]: ESPHome version 2023.11.6 compiled on Nov 28 2023, 15:08:02
[22:18:18][I][app:104]: Project adonno.tag_reader version 1.4
```
For nfc-cards written witrh the Android companion app (2023.10.2-full), ESPHome throws error. From log:
```
[22:18:44][D][pn532:311]: Mifare ultralight
[22:18:44][E][nfc.ndef_message:049]: Corrupt record encountered; NdefMessage constructor aborting
[22:18:44][D][light:036]: 'TagReader LED' Setting:
[22:18:44][D][light:047]:   State: ON
[22:18:44][D][light:051]:   Brightness: 100%
[22:18:44][D][light:058]:   Red: 0%, Green: 100%, Blue: 0%
[22:18:44][D][light:078]:   Flash length: 0.5s
[22:18:44][D][pn532:190]: Found new tag '04-97-1F-FE-9F-61-80'
[22:18:44][D][pn532:194]:   NDEF formatted records:
[22:18:44][D][pn532:323]: Waiting to read next tag
```
The same nfc-card can be read by HA android an ios app as well as NfcTools for Android without problems.

Config:
```
substitutions:
  name: "jukebox"
packages:
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <key>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
I recall this has been working, probably with an older ESPHome version. However, I am not sure.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.0

### What platform are you using?

ESP8266

### Board

ESP8266 D1 Mini

### Component causing the issue

PN532 NFC Reader

ssieb, 2023-12-08:
What's the problem?  It's reading the tag fine.  It's just complaining that the NDEF section isn't valid.

Ayuus, 2023-12-08:
The issue is that ESPHome cannot read the tag data, where other nfc readers can. So, I would not call that "reading the tag fine".

ssieb, 2023-12-08:
What data?  `[22:18:44][D][pn532:190]: Found new tag '04-97-1F-FE-9F-61-80'`

Ayuus, 2023-12-09:
Hi SSieb, thanks for your reply. The data you mention above is the serial number of the card only. 
The actual data written on the card by the Home Assistant companion app, as seen by HA developer tools when reading the card by phone:
```
event_type: tag_scanned
data:
  tag_id: >-
    https://open.spotify.com/album/652QafvLhoEGkSUFGyIf1K?si=y9oCT_B6RdudAHxACF2J9A&context=spotify%3Aalbum%3A652QafvLhoEGkSUFGyIf1K
  name: null
  device_id: 09b4049bb24z9062478c91de3965b0e3d
origin: LOCAL
time_fired: "2023-12-09T13:25:09.075866+00:00"
context:
  id: 01HH7CA4TKJRKVP1RMV2QR0EJZ
  parent_id: null
  user_id: 5141db875895301f8922bde21c912797
```
This was working in previous ESPHome versions.

Also, refer to attached picture for same card 
[image] read using nfctools.

Ayuus, 2023-12-11:
Here is my config:
```
substitutions:
  name: "jukebox"
packages:
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <redacted>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Ayuus, 2023-12-16:
A suspect ESPHome cannot read ndef records > 80 bytes.

gdst4rp, 2023-12-16:
> A suspect ESPHome cannot read ndef records > 80 bytes.

I see the same issue, though I haven't tested the exact length. Some of my tags read fine, but others don't. I started to write tags with longer and longer lengths, but didn't have the time to test enough and record the lengths. Though I can confirm that some tags read just fine, and others don't.

It also seems like if you scan the tags that can't be read, the device hangs and reboots (loses ping, ESPHome logs stop populating, etc.)

Ayuus, 2023-12-17:
Did some more testing: string of 110 bytes worked ok, 120 bytes did not. 
Android companion app can read those longer strings, so to me, it looks like a buffer allocation limitation in ESPHome.
I have noticed the "restart issue" described above, too, but I am not sure under what conditions that occurs.

gdst4rp, 2024-02-10:
Any workarounds for this issue? I'd rather not have to update all my NFC cards to use shorter byte strings. Looks like this issue is still open and unassigned?