esphome/issues #5209: Mopeka temperature reading incorrect at -1 deg C

doctorkb, 2023-12-08:
### The problem

In their app, Mopeka reports temperature of -1 deg C.  ESPHome plug-in reporting temperature of 255 deg C.

Also, in their app, Mopeka reports temperature of -2 deg C.  ESPHome plug-in reporting temperature of 254 deg C.

### Which version of ESPHome has the issue?

2023.12.0-dev (Dec 8 2023, 15:50:58)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.0

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

mopeka_pro_check

### Example YAML snippet

```yaml
- platform: mopeka_pro_check
    mac_address: C2:05:04:9C:7E:FC
    tank_type: 20LB_V
    temperature:
      name: "Propane temp"
    level:
      name: "Propane level"
    distance:
      name: "Propane distance"
    battery_level:
      name: "Propane battery level"
```

### Anything in the logs that might be useful for us?

```txt
[16:03:29][D][sensor:094]: 'Propane temp': Sending state 255.00000 °C with 0 decimals of accuracy
[16:03:49][D][sensor:094]: 'Propane temp': Sending state 254.00000 °C with 0 decimals of accuracy
```

### Additional information

Using DEV branch as my sensor is one of the "Universal" ones that was only added to support list two weeks ago.

doctorkb, 2023-12-08:
I suspect that this line: https://github.com/esphome/esphome/blob/a6f1701902355489bdfcb7dee53671abf063d1aa/esphome/components/mopeka_pro_check/mopeka_pro_check.cpp#L102

`uint8_t temp_in_c = this->parse_temperature_(manu_data.data);`

should actually be signed:
`int8_t temp_in_c = this->parse_temperature_(manu_data.data);`

ssieb, 2023-12-09:
I agree.  You could make a PR.

doctorkb, 2023-12-09:
> I agree. You could make a PR.

I would, except it looks like there's more to it than that (there's a function that processes the temperature) and I'm not sure how that works.

ssieb, 2023-12-09:
`return (message[2] & 0x7F) - 40`
It should be returning an int.  Probably no one tested it below 0. :-)

doctorkb, 2023-12-09:
Makes sense... I just don't know why that bitwise operator is there, nor what the content of the message looks like.  This isn't my primary language -- I can figure out just enough to be dangerous and probably break something else. 🙂

ssieb, 2023-12-09:
I was going to do it myself, but I see the author was working on it recently.  I've pinged the author on discord.