esphome/issues #5211: ESP32 Voice chrashes on Response

IronMate, 2023-12-10:
### The problem

The ESP chrashes and restarts on every Respones it gets from the Home Assistant. It recognises the Wake word, Made Text-to-speech, and starts building the Sentence for the answer. But mid Sentence it crashes. In my yaml file i deactivated the sepaker, because with the speeker I got the problem, that after a few sentences the "Speaker buffer is full". 

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP32

### Board

ESP32 DevKit

### Component causing the issue

voice_assistant

### Example YAML snippet

```yaml
esphome:
  name: esp32-mic-speaker
  friendly_name: esp32-mic-speaker
  on_boot:
     - priority: -100
       then:
         - wait_until: api.connected
         - delay: 1s
         - if:
             condition:
               switch.is_on: use_wake_word
             then:
               - voice_assistant.start_continuous:

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ANONYMOUS

ota:
  password: ANONYMOUS
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Mic-Speaker"
    password: ANONYMOUS

i2s_audio:
  i2s_lrclk_pin: GPIO27
  i2s_bclk_pin: GPIO26

microphone:
  - platform: i2s_audio
    id: mic
    adc_type: external
    i2s_din_pin: GPIO13
    pdm: false

speaker:
  - platform: i2s_audio
    id: big_speaker
    dac_type: external
    i2s_dout_pin: GPIO25
    mode: mono

voice_assistant:
  microphone: mic
  use_wake_word: false
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
#speaker: big_speaker
  id: assist

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(assist).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(assist).set_use_wake_word(false);
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/espvoice.yaml...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[12:07:55][D][voice_assistant:529]: Event Type: 10
[12:07:55][D][voice_assistant:538]: Wake word detected
[12:07:55][D][voice_assistant:529]: Event Type: 3
[12:07:55][D][voice_assistant:543]: STT started
[12:07:57][D][voice_assistant:529]: Event Type: 11
[12:07:57][D][voice_assistant:680]: Starting STT by VAD
[12:07:58][D][voice_assistant:529]: Event Type: 12
[12:07:58][D][voice_assistant:684]: STT by VAD end
[12:07:58][D][voice_assistant:422]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[12:07:58][D][voice_assistant:428]: Desired state set to AWAITING_RESPONSE
[12:07:58][D][voice_assistant:422]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[12:07:58][D][esp-idf:000]: I (85772) I2S: DMA queue destroyed
[12:07:58]
[12:07:58][D][voice_assistant:422]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[12:08:00][D][voice_assistant:529]: Event Type: 4
[12:08:00][D][voice_assistant:557]: Speech recognised as: " Fernseher an."
[12:08:00][D][voice_assistant:529]: Event Type: 5
[12:08:00][D][voice_assistant:562]: Intent started
[12:08:01][D][voice_assistant:529]: Event Type: 6
[12:08:01][D][voice_assistant:529]: Event Type: 7
[12:08:01][D][voice_assistant:585]: Response: "fernseher eingeschalteGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[12:08:01]
[12:08:01]Core  1 register dump:
[12:08:01]PC      : 0x400da5e3  PS      : 0x00060e30  A0      : 0x800e01ea  A1      : 0x3ffbecf0  
WARNING Decoded 0x400da5e3: std::_Function_handler<void (), esphome::voice_assistant::VoiceAssistant::on_event(esphome::api::VoiceAssistantEventResponse const&)::{lambda()#7}>::_M_invoke(std::_Any_data const&) at /data/build/esp32-mic-speaker/src/esphome/components/voice_assistant/voice_assistant.cpp:589
 (inlined by) _M_invoke at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
[12:08:01]A2      : 0x3ffbf120  A3      : 0x3ffbf26c  A4      : 0x3ffbf4f0  A5      : 0x3ffbf4f4  
[12:08:01]A6      : 0x3ffb2fb0  A7      : 0x3ffb3aa0  A8      : 0x800da5de  A9      : 0x3ffbecd0  
[12:08:01]A10     : 0x00000000  A11     : 0x3ffbecf0  A12     : 0x00000000  A13     : 0x3ffbed00  
[12:08:01]A14     : 0x3ffbed08  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
[12:08:01]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[12:08:01]
[12:08:01]
[12:08:01]Backtrace: 0x400da5e0:0x3ffbecf0 0x400e01e7:0x3ffbed30 0x400de2f5:0x3ffbed80 0x400e0576:0x3ffbedb0 0x400d72ca:0x3ffbedd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400da5e0: std::_Function_handler<void (), esphome::voice_assistant::VoiceAssistant::on_event(esphome::api::VoiceAssistantEventResponse const&)::{lambda()#7}>::_M_invoke(std::_Any_data const&) at /data/build/esp32-mic-speaker/src/esphome/components/voice_assistant/voice_assistant.cpp:589
 (inlined by) _M_invoke at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
WARNING Decoded 0x400e01e7: std::function<void ()>::operator()() const at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) esphome::Scheduler::call() at /data/build/esp32-mic-speaker/src/esphome/core/scheduler.cpp:234
WARNING Decoded 0x400de2f5: esphome::Application::loop() at /data/build/esp32-mic-speaker/src/esphome/core/application.cpp:69
WARNING Decoded 0x400e0576: loop() at /config/esphome/espvoice.yaml:108
WARNING Decoded 0x400d72ca: esphome::loop_task(void*) at /data/build/esp32-mic-speaker/src/esphome/components/esp32/core.cpp:69 (discriminator 1)
```

toggm, 2023-12-19:
I have the same issue on longer responses, even with an x-low speaking profile (german)

frustreermeneer, 2023-12-21:
Same here, when the speaker is commented out, it crashes upon response.

DaWheelz, 2024-02-24:
Same here :( I hope this will be fixed soon