esphome/issues #5214: using platformio_options to specify a larger flash size does not increase allowed OTA binary size

nkinnan, 2023-12-11:
### The problem

I have an 8285 board with 2mb flash.  The default flash size for the 8285 board is 1mb.  Using platformio_options I set flash size to 2mb.

```
esphome:
  ...<snip>...
  # this board has 2mb flash, but esphome assumes 1mb for the 8285 board id
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152

esp8266:
  board: esp8285
```

After flashing a minimal build with these options, I again tried flashing my full build which generates a binary just slightly too large for the default 1mb flash partitioning.

expected:
Partition size modified to allow a larger OTA binary and flash is successful.

actual:
`ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

### Which version of ESPHome has the issue?

2023.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP8266

### Board

8285 / 2mb

### Component causing the issue

flash partitioning

### Example YAML snippet

```yaml
esphome:
  ...<snip>...
  # this board has 2mb flash, but esphome assumes 1mb for the 8285 board id
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152

esp8266:
  board: esp8285
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.11.4
INFO Reading configuration /config/esphome/ir-transceiver-heatpump.yaml...
INFO Detected timezone 'America/Los_Angeles'
INFO Generating C++ source...
INFO Compiling app...
Processing ir-transceiver-heatpump (board: esp8285; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 2MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  47.6% (used 39000 bytes from 81920 bytes)
Flash: [======    ]  62.0% (used 635153 bytes from 1023984 bytes)
========================= [SUCCESS] Took 15.27 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ir-transceiver-heatpump.local
INFO  -> 192.168.10.245
INFO Uploading /data/build/ir-transceiver-heatpump/.pioenvs/ir-transceiver-heatpump/firmware.bin (639312 bytes)
INFO Compressed to 438181 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

### Additional information

Note that parts of the logs show 2mb as specified, and parts still show 1mb - for example "Flash: [======    ]  62.0% (used 635153 bytes from 1023984 bytes)"

I'm not familiar enough with this environment to actually locate the build files, or I'd pull the partitions csv to take a look and verify it hasn't been updated.

nkinnan, 2024-01-12:
I figured out a workaround, I'm documenting it here and leaving the bug open because I'm not sure if this is the intended method.  In another discussion for the esp32 board I saw a custom partitions.csv being used, but here I had to specify a ldscript in [platform-espressif8266](https://github.com/platformio/platform-espressif8266) which is very unintuitive and not documented anywhere.

```
esp8266:
  board: esp8285

esphome:
  # this board has 2mb flash, but esphome assumes 1mb for the 8285 board id
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152
    board_build.ldscript: "eagle.flash.2m256.ld"
```

I've made a pull request to [platform-espressif8266](https://github.com/platformio/platform-espressif8266) to add a board definition esp8285_2mb for the 2mb esp8285 boards in order to make this easier for others, but there has been no activity on it.

https://github.com/platformio/platform-espressif8266/pull/312

nkinnan, 2024-06-30:
> The answer is here [esp8266/Arduino#6681](https://github.com/esp8266/Arduino/issues/6681)

Thanks for the link!  

It's been a while since I looked at this, but even if the max binary size is 1mb, there is also space reserved for OTA storage and other partitions.  It seems we can override the build ldscript to expand these to more fully utilize the full 2mb (with 1mb max for the binary), but I'm still not sure if this is the "desired way" to do it.  From what I can remember last I looked into it, for the ESP32 the partitions are dynamically sized based on the specified flash size, whereas here they are hard coded based on what ldscript you select, and the ldscript used defaults to a 1mb flash size for esp82xx variants.

felipecrs, 2024-11-30:
Another option would be to select a board that has 2MB, right? Like:

```yaml
esp8266:
  board: esp_wroom_02
```