esphome/issues #5219: wifi password with number only dosen't work correctly

pey5531, 2023-12-13:
### The problem

when you configure ssid with number only password only some password that begin with 0 work correctly
example
  - ssid: ssid1
    password: '01823456'
  - ssid: ssid2
    password: '12345678'
 
after validate it become
  - ssid: ssid1
    password: 01823456
  - ssid: ssid2
    password: '12345678'

and only ssid1 would connect and work correctly.

and if the password begin with 0 without quotes, the yaml file would accept and validate without error and connect. any other number would through an error (without quotes) or validated (with quotes) with 2 extra single quotes (ex. '12345678') and make it not connect.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
wifi:
  networks:
  - ssid: example ssid 1
    password: 01823456
    priority: 2.0
  - ssid: example ssid 2
    password: '92873515'
    priority: 1.0
```

### Anything in the logs that might be useful for us?

```txt
Must be string, got <class 'esphome.helpers.EInt'>. did you forget putting quotes around the value?.

some but not all password that begin with 0 doesn't have this error
example:
01823456
04893011
```

### Additional information

yaml snippet without quotes
[image]

yaml snippet with quotes
[image]

after validation
[image]

ssieb, 2023-12-14:
A number starting with `0` will be interpreted as octal.  If it's not a valid octal number, then it will be considered a string.  It needs to be quoted.
It's not clear what you're saying doesn't work.  Do you have a quoted example that doesn't work?

pey5531, 2023-12-15:
[image]
the above example the top one works while the bottom doesn't work.

Unpoor, 2025-03-21:
I got the same error when using password starting with a 9 and only numbers, I simply cant change it cuss it is loads of equipment that uses this pw, soo this isnt working to have numbers for password? 9 in my case..