esphome/issues #5220: Modbus switch on holding register with bitmask causes unexpected side effects

wowtor, 2023-12-13:
### The problem

Not sure if this is a bug or a feature request.

I have a Modbus switch configured with `bitmask` and with `register_type` set to `holding`.

The expected behavior is that the masked bit is read/written, and nothing else.

The observed behavior is that toggling the switch causes the full register to be written. Bits outside the bitmask are always filled with zeros. This may cause unexpected side effects because the other bits may be used for something else.

The fix should be IMHO, in order of desirability:

- Have the write command change the masked bit only. The other bits are set with existing values obtained from the previous read command.

- Throw an error if `bitmask` is used in combination with `holding`. The register type `coil` should be used instead.

- Describe the behavior in the documentation.
 

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.12.0

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

Modbus

### Example YAML snippet

```yaml
switch:
  name: my switch
  platform: modbus_controller
  address: 5
  bitmask: 0x40
  register_type: holding
```

makuser, 2024-05-04:
/unstale

jinjamator, 2024-08-30:
I can see the same behavior, which makes this component unusable with those nice dirt cheep eletchsup 23IOD32 (32 inputs, 32outpus) devices sold on ali.  They are only supporting 0x3, 0x6, 0x16 commands for the outputs. So I have to use holding but I also want to switch more than 1 output per 16 outputs to on at a time.

dominikstupka, 2024-10-02:
Hi guys, do you found any solution for this? I have same issue.

clowrey, 2024-10-25:
I am running up against this same issue - there is probably a way to work around it in YAML with variable for the previous register read - I have not developed this yet. But it would be really nice to have it as a built in option.