esphome/issues #5222: I don't know what you did in the latest release but it KILLED my Sonoff Basics (2 of them!)

Majestic7979, 2023-12-13:
### The problem

I updated my devices using "update all". After that, one of my sonoff basic devices died. Won't connect to wifi anymore. It just blinks green.

Then I took another sonoff basic that was not updated, and it connected. I updated that one (separate config) and it connected.

Then I copied the config from the device that was dead into this one, and updated via serial.

WIFI IS DEAD TOO! Just blinks green.

What have you guys done!?

Ps. I don't need any clowns telling me "if it aint broke". If things were meant to be left the way they are then no updates would ever be needed. Also CVEs are a thing. I update all my devices always.

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Sonoff Basic

ssieb, 2023-12-14:
Basics are known to have random issues with wifi.  You even demonstrated that by saying that it worked with one config, but didn't work with another one.

Majestic7979, 2023-12-14:
> Basics are known to have random issues with wifi. You even demonstrated that by saying that it worked with one config, but didn't work with another one.

And yet they work fine with iTead's own firmware even after an update... I mean I'm not trying to shit on the project but is there any actual proof or is your comment just conjecture? Unless there's some study where it's been verified that it can be replicated again and again under certain circumstances then I'll tend to think it's ESPHome firmware that's causing issues. I used to have this problem between 2017 and 2021, then after that the issue stopped, I could apply ESPHome updates normally. Now the problem has returned, so something has changed in the code!? Even erasing the ESP flash won't make it work again. The green LED just flashes and flashes... plugged into the serial pins to read the logs, it says "scanning wifi... networks found: no networks".

At the end of the day this is an electronic device so it's code that tells it what to do. If the code is crap, then the device will work intermittently.

If it was a hardware problem it would either work or not work. Nothing is BGA soldered on a sonoff basic so it's not even a potential dry joint issue. Having to climb a ceiling to retrieve a non-working device after an update is much more than an inconvenience.

I will be replacing the Basics but they are on the supported list so this is never a good excuse. Either support it or drop it if it's such an unreliable device. A lot of new users (like me a few years ago) will make purchasing decisions based on what device is on an official supported list, causing financial loss. It's really unacceptable.

I know the nature of Open Source software very well, I know what kind of comment is coming and what I've got to say to "if you ain't happy fix it yourself" is that there's really no place for that attitude. If someone's being offered as such a large project it requires a level of QC when it's in the hands of so many users. Imagine if the response at Home Assistant (one of the largest OS projects on GitHub) was "fix it yourself", then that project would not be as big as it is, because face it, the majority of HA users aren't coders.

I hope this gets investigated more deeply and fixed. This isn't the first time. As my Sonoff Basic devices get f&*%$d by bad ESPHome updates I replace them with something else one by one. But I am not happy - it's eWaste and a waste of my money too. It should not fail when the hardware is fine. It's bad coding, period.

ssieb, 2023-12-15:
Where is this "supported list"?
If you don't believe me about the basics, check through the issues here for the ones about basics.  You can probably get it to work by trying random things like changing the logging level, tx power levels, changing components, etc.
I found a similar issue in the Tasmota github as well.
The common idea is that it's a power supply issue, especially since it generally works fine when powered using the serial interface.
And your whole rant about open source is just irrelevant.

Majestic7979, 2023-12-15:
> Where is this "supported list"? If you don't believe me about the basics, check through the issues here for the ones about basics. You can probably get it to work by trying random things like [...]

Irrelevant or not, I am free to "rant" as much as I want.

Go to the ESPHome website and search. https://esphome.io/devices/sonoff_basic

harmptor, 2023-12-15:
I have to agree with @ssieb that Basics are known for being an issue. In my opinion, the basics have weird quirks and esphome fails to work around that. When I first started to encounter issues with my Basics I created this issue #2108 , I recommend you read it, but eventually I could resolve it by downgrading ESPHome. So I also partly agree with you in that it is also an ESPHome code issue. 
About you ranting on open source: You're free to have your opinion, stop using open source software and please voice your opinion where it is appropriate.

ssieb, 2023-12-15:
> Go to the ESPHome website and search. https://esphome.io/devices/sonoff_basic

Those pages aren't linked to from anywhere and should be removed.
Device configs are now at https://devices.esphome.io/, but there's no official "supported" list and no guarantees.

Majestic7979, 2023-12-18:
> > Go to the ESPHome website and search. https://esphome.io/devices/sonoff_basic
> 
> Those pages aren't linked to from anywhere and should be removed. Device configs are now at https://devices.esphome.io/, but there's no official "supported" list and no guarantees.

You are one of those highly illogical and annoying people who always try to win. But you have lost the argument, so stop embarrassing yourself. The page exists and was found, period. I am not engaging in further useless conversation with you. This is proof of my argument - this project is not being effectively maintained because things that need to be done (like QC, documentation) aren't, and it's causing material losses to users. Regardless, that page WAS a source of supported devices AT THE TIME OF PURCHASE. So go waste someone else's time, you are extremely irritating and lack reading comprehension.

closed 2024-02-25 by jesserockz