esphome/issues #5223: SDP3X sensor component trying to connect on wrong I2C address for Sensirion SDP810-125 Sensor

Olaf92, 2023-12-14:
### The problem

Hi I'm having some issues setting up a pressure sensor with esphome. Looking at the log it seems that the sensor is found on I2C address 0x25 (this matches the cut sheet for the Sensirion SDP-810-125PA that I'm using, but then a couple lines down it seems that a connection is attempted on address 0x21 which appears to be the default address for the SDP300 sensors... 

Any ideas on how to get this working for the SDP-800 series which this component library is supposed to work for as well?

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP32

### Board

esp-wroom-32

### Component causing the issue

i2c

### Anything in the logs that might be useful for us?

```txt
[20:49:07][C][logger:416]: Logger:
[20:49:07][C][logger:417]:   Level: DEBUG
[20:49:07][C][logger:418]:   Log Baud Rate: 115200
[20:49:07][C][logger:420]:   Hardware UART: UART0
[20:49:07][C][i2c.arduino:053]: I2C Bus:
[20:49:07][C][i2c.arduino:054]:   SDA Pin: GPIO21
[20:49:07][C][i2c.arduino:055]:   SCL Pin: GPIO22
[20:49:07][C][i2c.arduino:056]:   Frequency: 50000 Hz
[20:49:07][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[20:49:07][I][i2c.arduino:069]: Results from i2c bus scan:
**[20:49:07][I][i2c.arduino:075]: Found i2c device at address 0x25**
[20:49:07][C][sdp3x.sensor:090]:   SDP3X 'HVAC Filter Pressure drop'
[20:49:07][C][sdp3x.sensor:090]:     Device Class: 'pressure'
[20:49:07][C][sdp3x.sensor:090]:     State Class: 'measurement'
[20:49:07][C][sdp3x.sensor:090]:     Unit of Measurement: 'hPa'
[20:49:07][C][sdp3x.sensor:090]:     Accuracy Decimals: 3
[20:49:07][C][sdp3x.sensor:091]:   **Address: 0x21**
**[20:49:07][E][sdp3x.sensor:093]:   Connection with SDP3X failed!**
[20:49:07][C][sdp3x.sensor:095]:   Update Interval: 60.0s
[20:49:07][C][captive_portal:088]: Captive Portal:
[20:49:07][C][mdns:115]: mDNS:
[20:49:07][C][mdns:116]:   Hostname: esphome-web-382364
[20:49:07][C][ota:097]: Over-The-Air Updates:
[20:49:07][C][ota:098]:   Address: esphome-web-382364.local:3232
[20:49:07][C][api:139]: API Server:
[20:49:07][C][api:140]:   Address: esphome-web-382364.local:6053
[20:49:07][C][api:142]:   Using noise encryption: YES
[20:53:52][I][ota:117]: Boot seems successful, resetting boot loop counter.
[20:53:52][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:53:52][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

jsuanet, 2023-12-15:
You didn't add your yaml, so I have to make some assumptions. 
- The I2c scan found a device at address 0x25
- The setup voor the sensor is using it's default (0x21) when no address is added in the sdpx3x section in your yaml. So what happens if you add 
`addres: 0x25`
tot the sdpx3 section?

Olaf92, 2023-12-16:
Thanks Jsuanet!! I added an "s" to "address" in your suggestion, and then I was in business! Sorry for not posting my actual code originally... I'm obviously rather new to GITHUB, and this was my first ESPHOME project too, so I really appreciate you taking a stab at helping me out even though it was a bit of a noob question that I didn't even provide all the required info to properly answer....

In the unlikely event that some fellow HA newbie runs into the same issue, then the yaml code that ended up working for me was:

i2c:
  sda: 21 #data pin used to connect sensor to board
  scl: 22 #second data pin used to sensor to board
  scan: true
  id: bus_a

sensor:
  - platform: sdp3x
    name: "Pressure Sensor"
    id: pressure_1
    address: 0x25

jsuanet, 2023-12-16:
Good to read this is the solution, and sorry for the typo in address ☺️

Please close this issue, because this is not a bug, it works as designed.

closed 2023-12-17 by Olaf92