esphome/issues #5224: 2023.12.0b2 fails to build on ESP32 S3

patoh, 2023-12-15:
### The problem

Compile error since 2023.12.0b2

/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:69:1: error: static assertion failed: "usb_osglue_*_int is not multicore capable"
   69 | _Static_assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_*_int is not multicore capable");
      | ^~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c: In function 'esp_usb_console_read_buf':
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:337:10: error: implicit declaration of function 'esp_usb_console_read_available'; did you mean 'esp_usb_console_write_available'? [-Werror=implicit-function-declaration]
  337 |     if (!esp_usb_console_read_available()) {
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          esp_usb_console_write_available
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c: At top level:
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:365:6: error: conflicting types for 'esp_usb_console_read_available'; have '_Bool(void)'
  365 | bool esp_usb_console_read_available(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:337:10: note: previous implicit declaration of 'esp_usb_console_read_available' with type 'int()'
  337 |     if (!esp_usb_console_read_available()) {
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors

### Which version of ESPHome has the issue?

2023.12.0b2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

lolin_s3_mini

### Component causing the issue

esp_system

### Example YAML snippet

```yaml
substitutions:
  device_name: office-env
  friendly_name: Office Enviroment
esphome:
  name: office-env
  platformio_options:
    platform: espressif32@^6.1.0
    board: lolin_s3_mini
    framework: espidf
    board_build.flash_mode: qio
  on_boot:
  - priority: 200.0
    then:
    - wait_until:
        condition:
          wifi.connected: {}
        timeout: 5s
  build_path: build/office-env
  friendly_name: ''
  area: ''
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2023.12.0
esp32:
  variant: ESP32S3
[... 576 lines cut ...]
      | ^~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c: In function 'esp_usb_console_read_buf':
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:337:10: error: implicit declaration of function 'esp_usb_console_read_available'; did you mean 'esp_usb_console_write_available'? [-Werror=implicit-function-declaration]
  337 |     if (!esp_usb_console_read_available()) {
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          esp_usb_console_write_available
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c: At top level:
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:365:6: error: conflicting types for 'esp_usb_console_read_available'; have '_Bool(void)'
  365 | bool esp_usb_console_read_available(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf@3.50101.230828/components/esp_system/port/soc/esp32s3/usb_console.c:337:10: note: previous implicit declaration of 'esp_usb_console_read_available' with type 'int()'
  337 |     if (!esp_usb_console_read_available()) {
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
*** [.pioenvs/office-env/esp_system/port/soc/esp32s3/usb_console.o] Error 1
========================= [FAILED] Took 73.01 seconds =========================
```

### Additional information

Only failing to build since 2023.12.0b2 - Compile error didn't exist on 2023.12.0b1, and confirmed by downgrading back to 2023.12.0b2 and tested.

May be related to PR #4658 or PR #4658 as the only USB related changes in the the latest beta

ESP-IDF 5.1.1 / platformio/espressif32@6.1.0

petez69, 2023-12-15:
Build on Kincony Board A8 also fails on compile, this is an ethernet connected board.  I rolled back to 2023.12.0b1 which also failed.  Reverted to 2023.11.6, started ok....

patoh, 2023-12-21:
Forcing the build to use the JTAG instead of CDC under esp-idf SDK options has fixed the build issue for on the Lolin S3 Mini boards for me. I haven't had time to test if the console logging output is working, unsure if it ever worked with these boards under esphome (never confirmed)

CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG: y

```
esp32:
  variant: esp32s3
  board: lolin_s3_mini
  framework:
    type: esp-idf  
    version: 5.1.2  
    platform_version: 6.4.0
    sdkconfig_options:
      CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG: y
```

patoh, 2023-12-22:
Resolved in 2023.12.2, assuming it's related to https://github.com/esphome/esphome/pull/5929

closed 2023-12-22 by patoh