esphome/issues #5225: shelly_dimmer doesn't set up without wifi

harmptor, 2023-12-15:
### The problem

Whenever my Shelly Dimmer 2 boots and cannot connect to WiFi, it seems to not set up the shelly_dimmer platform correctly. I have a lightswitch connected to `GPIO14` and when the device boots while the wifi is turned off, hitting the switch doesn't toggle the light. (it does when there is a wifi connection at boot, obviously)
I suspect it is because when setting up the component, it checks for the latest STM firmware and that fails without wifi.
I think this should be fairly easy to reproduce, but if there's anything missing let me know :)

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

shelly dimmer 2

### Component causing the issue

shelly_dimmer

### Example YAML snippet

```yaml
# output from `esphome config jimmy.yaml`:
light:
- platform: shelly_dimmer
  id: dimmer
  name: Jimmy
  internal: false
  restore_mode: RESTORE_DEFAULT_OFF
  default_transition_length: 500ms
  power:
    id: dimmer_power
    internal: true
    name: Jimmy Power
    entity_category: diagnostic
    state_class: ''
    disabled_by_default: true
    force_update: false
    unit_of_measurement: W
    accuracy_decimals: 1
    device_class: power
  voltage:
    id: dimmer_voltage
    internal: true
    name: Jimmy Voltage
    entity_category: diagnostic
    state_class: ''
    disabled_by_default: true
    force_update: false
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: voltage
  current:
    id: dimmer_current
    internal: true
    name: Jimmy Current
    entity_category: diagnostic
    state_class: ''
    disabled_by_default: true
    force_update: false
    unit_of_measurement: A
    accuracy_decimals: 2
    device_class: current
  leading_edge: false
  min_brightness: 160
  max_brightness: 1000
  gamma_correct: 2.8
  warmup_brightness: 0
  firmware:
    version: '51.5'
    update: false
    url: https://github.com/jamesturton/shelly-dimmer-stm32/releases/download/v51.5/shelly-dimmer-stm32_v51.5.bin
    sha256: 553fc1d78ed113227af7683eaa9c26189a961c4ea9a48000fb5aa8f8ac5d7b60
  disabled_by_default: false
  flash_transition_length: 0s
  nrst_pin:
    number: 5
    mode:
      output: true
      analog: false
      input: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false
  boot0_pin:
    number: 4
    mode:
      output: true
      analog: false
      input: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false
  update_interval: 10s
```

ssieb, 2024-04-14:
The component priority is very low, below wifi, because it wants to be able to check for firmware updates.  I think that needs to be reconsidered.

edge90, 2024-09-04:
TL;DR: The component will be marked as failed if `update` is set to `false`.
---

Following the flow of the `light.py`:

https://github.com/esphome/esphome/blob/e882cea47e76c50db674baf8dae0fa8744aabc4c/esphome/components/shelly_dimmer/shelly_dimmer.cpp#L81C8-L81C29

Means: if it doesn't have the firmeware data, the component will be marked as failed.

https://github.com/esphome/esphome/blob/e882cea47e76c50db674baf8dae0fa8744aabc4c/esphome/components/shelly_dimmer/light.py#L190

Means: Only set the firmware data if the hex is not `None.

https://github.com/esphome/esphome/blob/e882cea47e76c50db674baf8dae0fa8744aabc4c/esphome/components/shelly_dimmer/light.py#L78

Means: if `update` is set to `false` we will return `None`.

closed 2024-10-13 by jesserockz