esphome/issues #5226: New validation on v2023.12.0beta prevents sharing pin with deep sleep

edwardtfn, 2023-12-15:
### The problem

On my project, I use the same pin to wake up the system from deep sleep or for connecting a button when the system is awake. This worked fine until v2023.11, however some new validation on 2023.12.0b2 is failing and blocking the building.

### Which version of ESPHome has the issue?

2023.12.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
binary_sensor:

  ###### LEFT BUTTON BELOW DISPLAY TO TOGGLE RELAY#####
  - name: ${device_name} Left Button
    platform: gpio
    id: left_button
    pin:
      number: 14
      inverted: true
...
# Define the wake-up button. Use pin 14 for left button or pin 27 for right button
deep_sleep:
  wakeup_pin: 14
  wakeup_pin_mode: INVERT_WAKEUP
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.0b2
INFO Reading configuration /config/esphome/office-workstation-panel.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5910/head
INFO Updating https://github.com/esphome/esphome.git@pull/5930/head
INFO Detected timezone 'Europe/Stockholm'
Failed config

binary_sensor.gpio: [source <unicode string>:955]
  
  Pin 14 is used in multiple places.
  platform: gpio
  name: office-workstation-panel Left Button
  id: left_button
  pin: 
    number: 14
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  on_multi_click: 
    - timing: 
        - state: True
          min_length: 800ms
      then: 
        - logger.log: 
            format: Left button - Long click
            args: []
            tag: main
            level: DEBUG
        - script.execute: 
            id: ha_button
            page: !lambda |-
              return current_page->state.c_str();
            component: hw_bt_left
            command: long_click
      invalid_cooldown: 1s
    - timing: 
        - state: True
          min_length: 0ms
          max_length: 800ms
      then: 
        - logger.log: 
            format: Left button - Short click
            args: []
            tag: main
            level: DEBUG
        - if: 
            condition: 
              or: 
                - switch.is_on: 
                    id: relay1_local
                - and: 
                    - lambda: !lambda |-
                        return id(relay_1_fallback);
                    - or: 
                        - not: 
                            and: 
                              - api.connected: {}
                        - not: 
                            and: 
                              - wifi.connected: {}
            then: 
              - switch.toggle: 
                  id: relay_1
        - script.execute: 
            id: ha_button
            page: !lambda |-
              return current_page->state.c_str();
            component: hw_bt_left
            command: short_click
      invalid_cooldown: 1s
    - timing: 
        - state: True
          min_length: 15s
      then: 
        - switch.turn_off: 
            id: screen_power
        - delay: 5s
        - switch.turn_on: 
            id: screen_power
        - delay: 2s
        - lambda: !lambda |-
            disp1->soft_reset();
        - delay: 2s
        - script.execute: 
            id: boot_sequence
      invalid_cooldown: 1s
  disabled_by_default: False
deep_sleep: [source <unicode string>:37]
  
  Pin 14 is used in multiple places.
  wakeup_pin: 
    number: 14
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  wakeup_pin_mode: INVERT_WAKEUP
```

kiklhorn, 2023-12-15:
+1
Even in case where input pads is used for multiple input signals (counters, etc.) And this is valid and working before. 
page 50: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

dtto for st7789v display component where same pin is defined as backlight in the component itself and used as LEDC backlight control in the esphome yaml sketch.

edwardtfn, 2023-12-15:
Looks like there is a work around for v2023.12.0 (not working for v2023.11.0), but please onky use this if you know what you are doing, as you are basically disabling the check/protection from the system:

```yaml

binary_sensor:

  ###### LEFT BUTTON BELOW DISPLAY TO TOGGLE RELAY#####
  - name: ${device_name} Left Button
    platform: gpio
    id: left_button
    pin:
      number: 14
      inverted: true
      allow_other_uses: true
...
# Define the wake-up button. Use pin 14 for left button or pin 27 for right button
deep_sleep:
  wakeup_pin:
    number: 14
    allow_other_uses: true
  wakeup_pin_mode: INVERT_WAKEUP

```

edwardtfn, 2023-12-15:
There's a thread on Discord: https://discord.com/channels/429907082951524364/1184783083048812616

edwardtfn, 2023-12-15:
For reference, this was intentional and was introduced by this PR: https://github.com/esphome/esphome/pull/5666

edwardtfn, 2023-12-23:
I'm closing this as this is not a bug, but a breaking change. `allow_other_uses: true` is the way to go.

closed 2023-12-23 by edwardtfn