esphome/issues #5228: Docs: voice_assistant vad_threshold configuration parameter

trickv, 2023-12-16:
### The problem

I've been tinkering with my M5Stack Atom Echo and was curious about the parameter `vad_threshold`, only to find in [the docs](https://esphome.io/components/voice_assistant.html) that it's not documented.  Can you add some documentation for this parameter?

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

voice_assistant

### Additional information

I noticed _some_ discussion about tweaking this parameter here: https://github.com/esphome/issues/issues/5038

jleinenbach, 2024-03-12:
Answer is here: https://github.com/esphome/issues/issues/5038#issuecomment-1872202187
So I suggest:

### Configuration Parameter: vad_threshold

**Definition:**

`vad_threshold` specifies the threshold for a counter that tracks the number of samples detected to contain voice based on the outcome of `vad_process`. This counter increases each time a sample is identified to contain voice and decreases when it does not (with the minimum value clamped at zero). Once the counter surpasses the `vad_threshold`, voice is considered "detected," and the processing pipeline is initiated.

**Mechanism:**

- **Incrementing the Counter:** Occurs when `vad_process` positively identifies voice within a sample.
- **Decrementing the Counter:** Occurs when no voice is detected in a sample, with the counter's minimum limit being 0.
- **Voice Detection Trigger:** Activated when the counter exceeds the `vad_threshold`.

**Adjustment and Sensitivity:**

Adjusting `vad_threshold` to a lower value (e.g., 1) will trigger the processing pipeline the first time `vad_process` detects voice in a sample. A lower `vad_threshold` value theoretically makes the system more sensitive, increasing the likelihood of voice detection potentially even in absence of actual voice inputs.

**Practical Consideration:**

Users configuring this parameter should consider that setting the threshold too low might lead to the activation of the processing pipeline in situations where no actual voice is present, without significantly improving the system's overall performance.

trickv, 2024-03-21:
Thanks for digging that up! I think this would be a good addition to the docs.

It might be worth giving an example of how many samples is typically needed. As someone who only vaguely understands this...is one enough? 100? Is there a maximum setting, or a setting that's obviously overkill (how long in milliseconds is 1 sample?)

Are the collected samples sent to the voice processing pipeline if voice is indeed detected, or would having a large setting here result in the first few bits of speech being discarded / clipped?

Really helpful. I'll have to get back to tinkering with my M5 and see what I can see.