esphome/issues #5229: Voice Assistant "Assist in progress" stuck on after "nevermind"

joshuaboniface, 2023-12-16:
### The problem

I have a voice assistant pipeline as detailed below.

Upon triggering the wake word, the "Assist in progress" starts. If I say a command, everything is fine. If I say an invalid command, everything is fine. However if I say "nevermind", the voice assistant seems to get stuck: it no longer responds to wake words or even "voice_assistant.stop" actions, and the "Assist in progress" indicator remains "on".

### Which version of ESPHome has the issue?

2023.11.3, 2023.11.5, 2023.12.0b2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

voice_assistant

### Example YAML snippet

```yaml
201 api:       
202   encryption:
203     key: !secret api_encryption_key
204   on_client_connected:
205     - switch.turn_on: use_wake_word
...
249 voice_assistant:
250   microphone: mic
251   use_wake_word: false
252   noise_suppression_level: 2
253   auto_gain: 31dBFS
254   volume_multiplier: 4.0
255   id: assist
256   on_client_connected:
257     - delay: 2s
258     - light.turn_off:
259         id: output_led
260   on_client_disconnected:
261     - light.turn_on:
262         id: output_led
263         effect: flash_white
264     - switch.turn_off: use_wake_word
265   on_wake_word_detected:
266     - light.turn_off:
267         id: output_led
268     - light.turn_on:
269         id: output_led
270         red: 0
271         green: 0
272         blue: 1
273   on_listening:
274     - light.turn_on:
275         id: output_led
276         red: 0
277         green: 0
278         blue: 1
279   on_stt_end:
280     - light.turn_off:
281         id: output_led
282         transition_length: 1s
283   on_tts_start:
284     - light.turn_off:
285         id: output_led
286         transition_length: 1s
287     - if:
288         condition:
289           lambda: return x != "Sorry, I couldn't understand that";
290         then:
291           - logger.log: "Command successful!"
292           - light.turn_on:
293               id: output_led
294               effect: hold
295               red: 0
296               green: 1
297               blue: 0
298         else:
299           - logger.log: "Command failed!"
300           - light.turn_on:
301               id: output_led
302               effect: hold
303               red: 1
304               green: 0
305               blue: 0 
...
610 switch:  
611   - platform: template
612     name: "Enable Wake Word"
613     icon: mdi:account-voice
614     id: use_wake_word                                                                                                                                                                                                                      
615     optimistic: true
616     restore_mode: ALWAYS_OFF
617     entity_category: config
618     on_turn_on:
619       - lambda: id(assist).set_use_wake_word(true);
620       - if:
621           condition:
622             not:
623               - voice_assistant.is_running
624           then:
625             - voice_assistant.start_continuous
626     on_turn_off:
627       - voice_assistant.stop
628       - lambda: id(assist).set_use_wake_word(false);
```

### Anything in the logs that might be useful for us?

```txt
[04:47:28][D][voice_assistant:422]: State changed from IDLE to START_PIPELINE
[04:47:28][D][voice_assistant:428]: Desired state set to START_MICROPHONE
[04:47:28][D][voice_assistant:124]: microphone not running
[04:47:28][D][voice_assistant:206]: Requesting start...
[04:47:28][D][voice_assistant:422]: State changed from START_PIPELINE to STARTING_PIPELINE
[04:47:28][D][voice_assistant:443]: Client started, streaming microphone
[04:47:28][D][voice_assistant:422]: State changed from STARTING_PIPELINE to START_MICROPHONE
[04:47:28][D][voice_assistant:428]: Desired state set to STREAMING_MICROPHONE
[04:47:28][D][voice_assistant:159]: Starting Microphone
[04:47:28][D][voice_assistant:422]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[04:47:28][D][voice_assistant:529]: Event Type: 1
[04:47:28][D][voice_assistant:532]: Assist Pipeline running
[04:47:28][D][voice_assistant:422]: State changed from STARTING_MICROPHONE to STREAMING_MICROPHONE
[04:47:28][D][voice_assistant:529]: Event Type: 9
[04:47:31][D][voice_assistant:529]: Event Type: 10
[04:47:31][D][voice_assistant:538]: Wake word detected
[04:47:31][D][voice_assistant:529]: Event Type: 3
[04:47:31][D][voice_assistant:543]: STT started
[04:47:33][D][voice_assistant:529]: Event Type: 11
[04:47:33][D][voice_assistant:680]: Starting STT by VAD
[04:47:33][D][voice_assistant:529]: Event Type: 12
[04:47:33][D][voice_assistant:684]: STT by VAD end
[04:47:33][D][voice_assistant:422]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[04:47:33][D][voice_assistant:428]: Desired state set to AWAITING_RESPONSE
[04:47:33][D][voice_assistant:422]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[04:47:33][D][voice_assistant:422]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[04:47:35][D][voice_assistant:529]: Event Type: 4
[04:47:35][D][voice_assistant:557]: Speech recognised as: "nevermind"
[04:47:35][D][voice_assistant:529]: Event Type: 5
[04:47:35][D][voice_assistant:562]: Intent started
[04:47:35][D][voice_assistant:529]: Event Type: 6
[04:47:35][D][voice_assistant:529]: Event Type: 2
[04:47:35][D][voice_assistant:619]: Assist Pipeline ended
```

### Additional information

This seems to have started randomly after the most recent HomeAssistant update (2023.12.3) but I can't be 100% certain.

The logs above show everything from `voice_assistant` from a clean boot to me triggering a "nevermind" event as indicated. At that point, the voice_assistant component is completely hung as indicated.

joshuaboniface, 2023-12-17:
Reverting back to HomeAssistant 2023.12.1 has stopped the stuck "Assist in progress" part, but the ESPHome `voice_assistant` side still dies. So, definitely a combination of the two causing the issue.

Here's some more logs, stripping out all my numerous extra sensor outputs:

An invalid command:

```
[10:49:11][D][main:091]: voice assistant state: running   
[10:49:11][D][main:102]: voice assistant state: connected
[10:49:12][D][main:091]: voice assistant state: running                                                                
[10:49:12][D][main:102]: voice assistant state: connected
[10:49:13][D][voice_assistant:529]: Event Type: 10        
[10:49:13][D][voice_assistant:538]: Wake word detected    
[10:49:13][D][light:036]: 'output_led' Setting:
[10:49:13][D][light:085]:   Transition length: 0.2s
[10:49:13][D][light:036]: 'output_led' Setting:
[10:49:13][D][light:047]:   State: ON
[10:49:13][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[10:49:13][D][light:085]:   Transition length: 0.2s
[10:49:13][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[10:49:13][D][light:085]:   Transition length: 0.2s
[10:49:13][D][main:091]: voice assistant state: running
[10:49:13][D][main:102]: voice assistant state: connected
[10:49:14][D][voice_assistant:529]: Event Type: 11
[10:49:14][D][voice_assistant:680]: Starting STT by VAD
[10:49:14][D][main:091]: voice assistant state: running
[10:49:14][D][main:102]: voice assistant state: connected
[10:49:15][D][voice_assistant:529]: Event Type: 12
[10:49:15][D][voice_assistant:684]: STT by VAD end
[10:49:15][D][voice_assistant:422]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[10:49:15][D][voice_assistant:428]: Desired state set to AWAITING_RESPONSE
[10:49:15][D][voice_assistant:422]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[10:49:15][D][voice_assistant:422]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[10:49:15][D][main:091]: voice assistant state: running
[10:49:15][D][main:102]: voice assistant state: connected
[10:49:16][D][voice_assistant:529]: Event Type: 4
[10:49:16][D][voice_assistant:557]: Speech recognised as: "invalid command"
[10:49:16][D][main:091]: voice assistant state: running
[10:49:16][D][main:102]: voice assistant state: connected
[10:49:16][D][light:036]: 'output_led' Setting:
[10:49:16][D][light:047]:   State: OFF
[10:49:16][D][light:085]:   Transition length: 1.0s
[10:49:16][D][voice_assistant:529]: Event Type: 5
[10:49:16][D][voice_assistant:562]: Intent started
[10:49:16][D][voice_assistant:529]: Event Type: 6
[10:49:17][D][voice_assistant:529]: Event Type: 7
[10:49:17][D][voice_assistant:585]: Response: "Sorry, I couldn't understand that"
[10:49:17][D][light:036]: 'output_led' Setting:
[10:49:17][D][light:085]:   Transition length: 1.0s
[10:49:17][D][main:341]: Command failed!
[10:49:17][D][light:036]: 'output_led' Setting:
[10:49:17][D][light:047]:   State: ON
[10:49:17][D][light:059]:   Red: 100%, Green: 0%, Blue: 0%
[10:49:17][D][light:109]:   Effect: 'hold'
[10:49:17][D][voice_assistant:529]: Event Type: 8
[10:49:17][D][voice_assistant:605]: Response URL: "http://10.100.0.15:8123/api/tts_proxy/dae2cdcb27a1d1c3b07ba2c7db91480f9d4bfd8f_en-gb_010745e5ef_tts.piper.mp3"
[10:49:17][D][voice_assistant:422]: State changed from AWAITING_RESPONSE to IDLE
[10:49:17][D][voice_assistant:428]: Desired state set to IDLE
[10:49:17][D][voice_assistant:422]: State changed from IDLE to START_PIPELINE
[10:49:17][D][voice_assistant:428]: Desired state set to START_MICROPHONE
[10:49:17][D][voice_assistant:529]: Event Type: 2
[10:49:17][D][voice_assistant:619]: Assist Pipeline ended
[10:49:17][D][voice_assistant:124]: microphone not running
[10:49:17][D][voice_assistant:206]: Requesting start...
[10:49:17][D][voice_assistant:422]: State changed from START_PIPELINE to STARTING_PIPELINE
[10:49:17][D][main:364]: voice ended                   
[10:49:17][D][voice_assistant:124]: microphone not running
[10:49:17][D][voice_assistant:124]: microphone not running
[10:49:17][D][voice_assistant:124]: microphone not running
[10:49:17][D][voice_assistant:124]: microphone not running
[... 99 lines cut ...]
[10:50:52][D][main:091]: voice assistant state: running
[10:50:52][D][main:102]: voice assistant state: connected
[10:50:53][D][main:091]: voice assistant state: running
[10:50:53][D][main:102]: voice assistant state: connected
[10:50:54][D][main:091]: voice assistant state: running
[10:50:54][D][main:102]: voice assistant state: connected
[10:50:55][D][main:091]: voice assistant state: running
[10:50:55][D][main:102]: voice assistant state: connected
[10:50:56][D][main:091]: voice assistant state: running
[10:50:56][D][main:102]: voice assistant state: connected
[10:50:57][D][main:091]: voice assistant state: running
[10:50:57][D][main:102]: voice assistant state: connected
[10:50:58][D][main:091]: voice assistant state: running
[10:50:58][D][main:102]: voice assistant state: connected
[10:50:59][D][main:091]: voice assistant state: running
[10:50:59][D][main:102]: voice assistant state: connected
[10:51:00][D][main:091]: voice assistant state: running
[10:51:00][D][main:102]: voice assistant state: connected
[10:51:01][D][main:091]: voice assistant state: running
[10:51:01][D][main:102]: voice assistant state: connected
[10:51:02][D][main:091]: voice assistant state: running
[10:51:02][D][main:102]: voice assistant state: connected
```

Note here especially that both the `is_running` and `connected` results remain as expected, but there's no more wake word loop being triggered every 5 seconds here; at this point it's completely hung.

joshuaboniface, 2023-12-17:
For now, I've bumped HASS back to 2023.12.3 and have simply reconfigured by TTS engine to replace "never[ ]*mind" with "INVALID" so this state won't be triggered. Happy to resume testing of various combinations or fixes though.

Nerivec, 2023-12-23:
I can confirm this behavior. I will also expand it, because to me, it seems any command that doesn't output TTS/speech triggers this behavior, not just `nevermind`.
This needs a fix asap, any logic that bypasses "direct" TTS (e.g. to output on a different board) is broken because it never calls the "speech" part of the pipeline on the spoken-to board. Not to mention blank response overrides if you don't want to hear "turned on ..." every time you do...

Note: I've usually been able to clear the hung state of "assist_in_progress" by toggling use_wake_word (pipeline restart) and asking something with a speech output directly after.

_HA: 2023.12.3 / ESPHome: 2023.12.3_

WebSpider, 2024-01-08:
This may be a cause of #5038 as well, since the behaviour is quite similar.

botfixer, 2024-01-09:
Have exactly the same problem. I use the Atom Echo and have set it up according to the description without making special. It works fine when I give a known or unknown voice commands.

But if I say "vergiss es" (means "forget it" in german) after the wake word, then the "Assist in use" entity does not go back to off, but remains on and there is no more reaction when saying the wake word. 
I then have to unplug the Atom Echo to restart it, to make it work again.

Updatet from ESPhome 2023.3.2 and Home Assiatant 2023.12.3, but nothing changes.

Installation: HAOS x86-64 on NUC
Home Assistant: 2024.1.2
ESPHome 2923.12.5
ESP Board: Atom Echo