esphome/issues #5231: Error writing to GP8403, code 2

kr3ator, 2023-12-17:
### The problem

I'm not able to use GP8403 0-10V output module as a speed fan.
When turning on or off the fan component in HA I get the error: 
[14:32:32][D][fan:021]: 'Okap' - Setting:
[14:32:32][D][fan:024]:   State: ON
[14:32:32][E][gp8403.output:021]: Error writing to GP8403, code 2
[14:32:32][D][fan:092]: 'Okap' - Sending state:
[14:32:32][D][fan:093]:   State: ON
[14:32:32][D][fan:095]:   Speed: 100

I'm using [this hardware](https://www.dfrobot.com/product-2613.html)

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP8266

### Board

D1mini

### Component causing the issue

GP8403

### Example YAML snippet

```yaml
Relevant device config:

i2c:
  frequency: 400kHz

gp8403:
  id: gp8403_10v
  voltage: 10v
  address: 0x5F

output:
  - platform: gp8403
    id: gp8403_0
    gp8403_id: gp8403_10v
    channel: 0

fan:
  - name: "Okap"
    platform: speed
    output: gp8403_0
```

### Anything in the logs that might be useful for us?

```txt
[14:30:20][C][i2c.arduino:053]: I2C Bus:
[14:30:20][C][i2c.arduino:054]:   SDA Pin: GPIO4
[14:30:20][C][i2c.arduino:055]:   SCL Pin: GPIO5
[14:30:20][C][i2c.arduino:056]:   Frequency: 400000 Hz
[14:30:20][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[14:30:20][I][i2c.arduino:069]: Results from i2c bus scan:
[14:30:20][I][i2c.arduino:075]: Found i2c device at address 0x5F
[14:30:20][C][gp8403:015]: GP8403:
[14:30:20][C][gp8403:016]:   Voltage: 10V
[14:30:20][C][gp8403:017]:   Address: 0x5F
[14:30:20][C][speed.fan:016]: Speed Fan 'Okap'
[14:30:20][C][speed.fan:151]:   Speed: YES
[14:30:20][C][speed.fan:152]:   Speed count: 100
[14:30:20][C][gp8403.output:013]: GP8403 Output:
[14:30:20][C][gp8403.output:014]:   Channel: 0
```

ssieb, 2023-12-17:
That error is the device not answering.  Check the serial logs from boot and see if the setup command works.

kr3ator, 2023-12-18:
I've opened the serial console and enabled verbose logging but I was just ensured that the module is not responding on the I2C bus.

I have solved the issue by powering the module from 5V instead of 3.3V (the documentation states that it also should work on 3.3V).

closed 2023-12-18 by kr3ator