esphome/issues #5232: Animation does not work with remote files

telefonmann, 2023-12-17:
### The problem

I would like to show an animation on a display connected to an esp32. Showing local and remote images does work. Showing local animations does work as well. But animations from remote files do not work. Error message is:

`Could not find file '/config/esphome/https://upload.wikimedia.org/wikipedia/commons/2/2c/Rotating...`

It seems the file is being searched locally. But the documentation says "Animation inherits all options from the image component", so as displaying remote images does work, I expect the remote animations to work as well.

Thanks in advance!

### Which version of ESPHome has the issue?

2023.11.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

D1 mini ESP32

### Example YAML snippet

```yaml
animation:
  - file: https://upload.wikimedia.org/wikipedia/commons/2/2c/Rotating_earth_%28large%29.gif
    type: RGB24
    id: dummy
    resize: 100x100
```

guillempages, 2023-12-21:
It's probably a case of "the documentation is outdated" ;-)
The remote image implementation was only done recently, and was only done for images; the animation component does not support that (yet?)

@landonr you might want to take a look at this.

landonr, 2024-01-10:
nice catch I'll check it out

Releah, 2024-06-07:
This is still an issue :(

landonr, 2024-06-08:
@Releah thanks for the reminder, i'll try to work on this today or tomorrow

landonr, 2024-06-09:
https://github.com/esphome/esphome/pull/6876