esphome/issues #5233: Missing characters

andrewigali, 2023-12-18:
### The problem

I don't know how long this problem has existed (there was no such problem in the past), with ESPHome 2023.11.6 when I create a new FW, it does not display the accented letters (even without an accent).

Example:
ESPHome code:
esphome:
  name: eloszoba-erzekelok
  friendly_name: Előszoba érzékelők

binary_sensor:
  - platform: gpio
    pin: GPIO35
    name: "Bejárati ajtó"
    device_class: door

Normally, I should receive such an entity identifier in HA: binary_sensor.eloszoba_erzekelok_bejarati_ajto
Instead I get: binary_sensor.elszoba_rzkelk_bjati_ajt

So the following accents are not converted to unaccented ones: á, é, í, ó, ő, ü, ű (these are also used in Hungarian words).

Maybe I'm wrong but it seems that UTF-8 encoding doesn't work if friendly_name: Előszoba érzékelők is entered under esphome:.
Neither in friendly_name nor in entity ID.

Is it possible to correct this?

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.6

### What platform are you using?

ESP32

### Board

ESP32 Wemos D1, ESP32 WROOM 38pin

### Example YAML snippet

```yaml
esphome:
  name: eloszoba-erzekelok
  friendly_name: Előszoba érzékelők

binary_sensor:
  - platform: gpio
    pin: GPIO35
    name: "Bejárati ajtó"
    device_class: door
```

dgolda, 2024-02-02:
I have same problem with polish characters, so "Prędkość" in Wiatrak is wiatrak_pr_dko.
Please replace accented characters with latin ones:
ą=a, ć=c, ę=e, ł=l, ń=n, ó=o, ś=s, ź=z, ż=z and Ą=A, Ć=C, Ę=E, Ł=L, Ń=N, Ó=O, Ś=S, Ź=Z, Ż=Z

djereg, 2024-03-30:
Same problem with hungarian characters. ESPHome 2024.3.1
á=a, é=e, í=i, ó=o, ö=o, ő=o, ú=u, ü=u, ű=u (extended)

Kazán Controller Bejövő Víz Hőmérséklet
kazan_controller_bej_v_v_z_h_m_rs_klet

leoshusar, 2025-03-02:
I really don't like the behavior of completely replacing chars with underscores. Ideally I'd just keep the default entity ID, but it sometimes becomes unreadable when half of a word are underscores, so I first need to manually rename everything.

For example Z2M + MQTT works correctly in this case, it just strips accents.

g-kiss, 2025-06-14:
https://github.com/home-assistant/core/pull/146796