esphome/issues #5239: installation returns an error FileNotFoundError: [Errno 2] No such file or directory: 'config/'

frankynux, 2023-12-19:
### The problem

I'm trying to install on a raspberry (raspbian) or a laptop with debian. When I launch esphome configuration dashboard, I get this feedback

`esphome dashboard config/
2023-12-19 20:04:08,600 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir config/...
Traceback (most recent call last):
  File "/home/franky/.local/bin/esphome", line 8, in <module> sys.exit(main())
             ^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/__main__.py", line 1003, in run_esphome
    return PRE_CONFIG_ACTIONS[args.command](args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/__main__.py", line 517, in command_dashboard
    return dashboard.start_web_server(args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 1517, in start_web_server
    status_thread = MDNSStatusThread()
                    ^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 977, in __init__
    self._refresh_hosts()
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 981, in _refresh_hosts
    entries = _list_dashboard_entries()
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 726, in _list_dashboard_entries
    return settings.entries()
           ^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 148, in entries
    for file in self.list_yaml_files():
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/dashboard/dashboard.py", line 129, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/util.py", line 292, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/home/franky/.local/pipx/venvs/esphome/lib/python3.11/site-packages/esphome/util.py", line 292, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
                                                            ^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'config/'`

Any help is welcome

### Which version of ESPHome has the issue?

2023.11.6

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

randybb, 2023-12-19:
```
FileNotFoundError: [Errno 2] No such file or directory:

    'config/'`
```
Does it exist?

frankynux, 2023-12-19:
Non it doesn't

randybb, 2023-12-19:
Then why you are providing it as a parameter?

closed 2023-12-19 by randybb

frankynux, 2023-12-20:
because

(venv) ***@***.***:~$ esphome dashboard
usage: esphome dashboard [-h] [--port PORT] [--address ADDRESS]
                         [--username USERNAME] [--password PASSWORD]
                         [--open-ui] [--socket SOCKET]
                         configuration
esphome dashboard: error: the following arguments are required:
configuration

randybb, 2023-12-20:
The directory path has to exist. So create a directory first, then use it or just use `./` for the same dir.
```bash
(esphome) randy@whitemesh:~/test/esphome> mkdir config
(esphome) randy@whitemesh:~/test/esphome> esphome dashboard config
2023-12-20 10:01:12,896 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir config...
^C2023-12-20 10:01:14,529 INFO Shutting down...
(esphome) randy@whitemesh:~/test/esphome> esphome dashboard ./
2023-12-20 10:01:16,726 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir ./...
^C2023-12-20 10:01:17,736 INFO Shutting down...
```

frankynux, 2023-12-20:
> (esphome) ***@***.***:~/test/esphome> mkdir config
> (esphome) ***@***.***:~/test/esphome> esphome dashboard config
> (esphome) ***@***.***:~/test/esphome> esphome dashboard ./

I try
***@***.***:~$ source venv/bin/activate
(venv) ***@***.***:~$ esphome dashboard ./
2023-12-20 19:25:36,985 INFO Starting dashboard web server on
http://0.0.0.0:6052 and configuration dir ./...
2023-12-20 19:26:48,233 INFO 200 GET / (127.0.0.1) 14.49ms
2023-12-20 19:26:48,382 INFO 200 GET
/static/fonts/material-icons/material-icons.css?hash=ead4c276 (127.0.0.1)
13.16ms
2023-12-20 19:26:48,386 INFO 200 GET
/static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838
(127.0.0.1) 2.43ms
2023-12-20 19:26:48,387 INFO 200 GET
/static/css/esphome-2.css?hash=c222cb3b (127.0.0.1) 2.80ms
2023-12-20 19:26:48,397 INFO 200 GET /static/js/esphome/index-6rBAfUo5.js
(127.0.0.1) 11.94ms
2023-12-20 19:26:48,474 INFO 200 GET /static/images/logo-text.svg
(127.0.0.1) 1.88ms
2023-12-20 19:26:48,479 INFO 200 GET /devices (127.0.0.1) 1.02ms
2023-12-20 19:26:48,500 INFO 200 GET /static/js/esphome/c.AMCPNCIe.js
(127.0.0.1) 18.34ms
2023-12-20 19:26:48,527 INFO 200 GET
/static/fonts/material-icons/MaterialIcons-Regular.woff2 (127.0.0.1) 0.80ms
2023-12-20 19:26:48,555 INFO 200 GET
/static/images/favicon.ico?hash=2655a073 (127.0.0.1) 1.10ms
2023-12-20 19:26:48,939 INFO 200 GET /static/js/esphome/c.nnnnLD37.js
(127.0.0.1) 4.82ms
2023-12-20 19:26:48,941 INFO 200 GET /static/js/esphome/c.S1FZv1in.js
(127.0.0.1) 1.58ms
2023-12-20 19:26:53,480 INFO 304 GET /devices (127.0.0.1) 1.44ms

YESS  I can connect to web server

Thanks for your time to help me