esphome/issues #5240: Wrong warning ? "Lambda for parameter red of light.addressable_set should return values in range 0-1 instead of 0-255"

peterwup, 2023-12-19:
### The problem

I use light.addressable_set to access to an LED stripe. 
The hardware is working as expected, but I get a warning  in the log file: 
"Lambda for parameter blue of light.addressable_set should return values in range 0-1 instead of 0-255."

I set the colour by lambda values  0..100

Is it possible that this warning is just wrong ? 

### Which version of ESPHome has the issue?

2023.11.6 Dec 18 2023, 08:53:37

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
api:
  # home assistant can use this interface to set each single LED
  services:
    - service: change_dl_led
      variables:
        # LED identifier 0..15 
        my_led_id: int
        # RGB value 0..100
        my_red: int
        my_green: int
        my_blue : int
      then:
        - light.addressable_set:
            id: led_stripe
            range_from: !lambda 'return my_led_id;'
            range_to: !lambda 'return my_led_id;'
            red: !lambda 'return my_red;'
            green: !lambda 'return my_green;'
            blue: !lambda 'return my_blue;'

light:
  - platform: esp32_rmt_led_strip
    id: led_stripe
    rgb_order: GRB
    pin: GPIO13
    num_leds: 16
    rmt_channel: 0
    chipset: ws2812
```

### Anything in the logs that might be useful for us?

```txt
[W]	[light.automation:011]	
Lambda for parameter blue of light.addressable_set should return values in range 0-1 instead of 0-255.
```

ssieb, 2023-12-19:
Is it working?  If not, then you should pay attention to what it says.

peterwup, 2023-12-19:
Yes it works.

ssieb, 2023-12-19:
But do the colors work properly?  You will only get full on or off doing it like that.

peterwup, 2023-12-21:
I did some tests. 
It works with 0..100 and it works with 0..1 (floating point).
So I will use floating point instead of integer and there is no warning anymore. 
I wonder wy both variants work.

closed 2023-12-21 by peterwup