esphome/issues #5244: Multiple GPIO expanders results in error "Pin X is used in multiple places".

justlikeef, 2023-12-20:
### The problem

Multiple GPIO expanders results in error "Pin X is used in multiple places".

### Which version of ESPHome has the issue?

2024.1.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

esp32-poe

### Component causing the issue

gpio

### Example YAML snippet

```yaml
i2c:
  sda: 13
  scl: 16
  scan: true
  id: bus_a

mcp23017:
  - id: 'mcp23017_hub_a'
    address: 0x20
    i2c_id: bus_a
  - id: 'mcp23017_hub_b'
    address: 0x21
    i2c_id: bus_a

switch:
  - platform: gpio
    id: v1_open
    name: $pretty_devicename Valve 1 Open
    interlock: [v1_close]
    pin:
      mcp23xxx: mcp23017_hub_a
      number: 0
      mode:
        output: true
      inverted: false
  - platform: gpio
    id: v5_open
    name: $pretty_devicename Valve 5 Open
    interlock: [v5_close]
    interlock_wait_time: .25s
    pin:
      mcp23xxx: mcp23017_hub_b
      number: 0
      mode:
        output: true
      inverted: false
```

randybb, 2023-12-20:
Yes, it is a known issue. A fix is on the way https://github.com/esphome/esphome/pull/5973

justlikeef, 2023-12-20:
Confirmed fixed.

closed 2023-12-20 by justlikeef