esphome/issues #5246: Pin validation starting in v2023.12.0 conflicts with status_led usage in example

xconverge, 2023-12-21:
### The problem

This example no longer will validate correctly with esphome 2023.12.0 due to GPIO5 being used for the status_led in 2 places

https://devices.esphome.io/devices/Treatlife-Wall-Switch#basic-configuration-3-way

Is there an issue with the example or the actual functionality?

```
INFO ESPHome 2023.12.0
INFO Reading configuration garageswitch.yaml...
Failed config

status_led: [source <unicode string>:14]

  Pin 5 is used in multiple places.
  pin:
    number: 5
    inverted: True
    mode:
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
switch.gpio: [source <unicode string>:83]

  Pin 5 is used in multiple places.
  platform: gpio
  id: red_led
  pin:
    number: 5
    inverted: True
    mode:
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
  internal: True
  disabled_by_default: False
  restore_mode: ALWAYS_OFF
  interlock_wait_time: 0ms
  name: red_led
```

Similar to #5226

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

Murada99, 2023-12-21:
Try the following:

`  pin:
    number: 5
    allow_other_uses: true
`

randybb, 2023-12-21:
It was [mentioned](https://next.esphome.io/changelog/2023.12.0.html#pin-reuse-validation) as a breaking change - so you need explicitly allow it under each reused pin.

closed 2023-12-21 by randybb

xconverge, 2023-12-21:
> It was [mentioned](https://next.esphome.io/changelog/2023.12.0.html#pin-reuse-validation) as a breaking change - so you need explicitly allow it under each reused pin.

Yep, I saw that, however it is really hard for someone (me) to know which scenario they are in...

"This is a common source of issues and generally a pin does not need to be reused across multiple components in a single configuration. The error can be bypassed by specifically adding another config item to all of the duplicate pin definitions."

am I:

1. the common source of issues and generally a pin does not need to be reused (the validation is doing a great job if so)
2. ignore (1), what I am doing is fine, so bypass it

I have done (2) for now without really knowing if I should have heeded the warning/error.

My issue was also a placeholder issue for fixing the example https://devices.esphome.io/devices/Treatlife-Wall-Switch#basic-configuration-3-way since that repository does not have an issue tracker

Roving-Ronin, 2023-12-22:
Would anyone know what the solution is for where the snesor is "platform: adc" , as the above solution doesn't work for this.
In the example below, the same PIN is used to read a pressure sensor and from that create sensors for the voltage of the physical sensor and then from that the water pressure in psi and kpi, and from that water tank levels, % an litres are then calculated.

```
# ADC Input Port 3 - 0-5V
  - platform: adc
    pin: 35
    name: 'Water Tank - Voltage'
    id: water_tank_voltage
    accuracy_decimals: 3
    device_class: "voltage"
    state_class: "measurement"
    unit_of_measurement: "V"
    update_interval: 5s
    attenuation: 11db  #auto
    filters:
#     - multiply: 3.3
    - median: 
       window_size: 6
       send_every: 6
       send_first_at: 6
#    - calibrate_polynomial:
#       degree: 2
#       datapoints:
#          # Map from SENSOR -> TRUE value
#         - 4.15867 -> 4.01667
#         - 3.75700 -> 3.61500
#         - 3.35533 -> 3.21333
#         - 2.95367 -> 2.81167
#         - 2.55200 -> 2.41000
#         - 2.15033 -> 2.00833
#         - 1.74867 -> 1.60667
#         - 1.34700 -> 1.20500
#         - 0.94533 -> 0.80333
#         - 0.14200 -> 0.00000
    internal: true

# sensor:
  - platform: adc
    pin: 35
    name: 'Water Tank - Pressure (kPa)'
    id: water_tank_pressure_bar
    accuracy_decimals: 2
    device_class: pressure
    state_class: "measurement"
    unit_of_measurement: kPa
    icon: "mdi:gauge"
    update_interval: 5s
    attenuation: 11db  #auto
    filters:
#      - multiply: 3.3
      - median: 
          window_size: 6
          send_every: 6
          send_first_at: 6
      - calibrate_linear:
          # Measured value of X volt maps to y kPa
          - 4.15867 -> 23.6061428   # 100%
          - 3.75700 -> 21.2455285   #  90%
          - 3.35533 -> 18.8849142   #  80%
          - 2.95367 -> 16.5243      #  70%
          - 2.55200 -> 14.1636857   #  60%
          - 2.15033 -> 11.8030714   #  50%
          - 1.74867 -> 9.44245712   #  40%
          - 1.34700 -> 7.08184284   #  30%
          - 0.94533 -> 4.72122856   #  20%
          - 0.54367 -> 2.36061428   #  10%
          - 0.14200 -> 0.0          #   0%
        # From https://www.sensorsone.com/liquid-level-to-pressure-calculator/?lval=241&lunit=centimetres+%28cm%29&lfctr=cm&sg=1&grav=9.79508&punit=kilopascals+%28kPa%29&pfctr=kPa
        # Gouburn 655m - Local Gravity 9.79508 ms-2
    internal: true

  - platform: adc
    pin: 35
    name: 'Water Tank - Pressure (psi)'
    id: water_tank_pressure_psi
    accuracy_decimals: 2
    device_class: pressure
    state_class: "measurement"
    unit_of_measurement: psi
    icon: "mdi:gauge"
    update_interval: 5s
    attenuation: 11db  #auto
    filters:
#      - multiply: 3.3
      - median: 
          window_size: 6
          send_every: 6
          send_first_at: 6
      - calibrate_linear:
          # Measured value of X volt maps to y PSI

          - 4.15867 -> 3.42378155   # 100%
          - 3.75700 -> 3.08140339   #  90%
          - 3.35533 -> 2.73902524   #  80%
          - 2.95367 -> 2.39664708   #  70%
          - 2.55200 -> 2.05426893   #  60%
          - 2.15033 -> 1.71189077   #  50%
          - 1.74867 -> 1.36951262   #  40%
          - 1.34700 -> 1.02713446   #  30%
          - 0.94533 -> 0.68475631   #  20%
          - 0.54367 -> 0.342378155  #  10%
          - 0.14200 -> 0.0          #   0%

        # https://www.sensorsone.com/liquid-level-to-pressure-calculator/?lval=241&lunit=centimetres+%28cm%29&lfctr=cm&sg=1&grav=9.79508&punit=kilopascals+%28kPa%29&pfctr=kPa
    internal: true    
```

randybb, 2023-12-22:
Normally only one sensor is attached to a pin and then another one can take the value ([copy integration](https://esphome.io/components/copy)) from the native sensor and use it for calculation.

Roving-Ronin, 2023-12-24:
@randybb many thanks, had never seen any mention of the copy integration before. Works a charm !