esphome/issues #5247: Pin validation issue starting in v2023.12.0

andypnz, 2023-12-21:
### The problem

Since 2023.12.0 I get the follwoing error when using one ADC pin to generate two readings (raw and modified):

`INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/esp32-waterpump-mon.yaml...
Failed config

sensor.adc: [source <unicode string>:100]
  
  Pin 32 is used in multiple places.
  platform: adc
  name: esp32-waterpump-mon Water Pressure
  pin: 
    number: 32
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  update_interval: 2s
  unit_of_measurement: PSI
  accuracy_decimals: 2
  attenuation: 0db
  filters: 
    - calibrate_linear: 
        datapoints: 
          - from: 0.076
            to: 0.0
          - from: 0.2
            to: 14.7
          - from: 0.41
            to: 40.0
          - from: 0.46
            to: 60.0
          - from: 0.95
            to: 120.0
        method: least_squares
    - sliding_window_moving_average: 
        window_size: 15
        send_every: 15
        send_first_at: 1
  disabled_by_default: False
  force_update: False
  device_class: voltage
  state_class: measurement
  raw: False
sensor.adc: [source <unicode string>:130]
  
  Pin 32 is used in multiple places.
  platform: adc
  name: esp32-waterpump-mon RAW VOLTAGE Water Pressure
  pin: 
    number: 32
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  update_interval: 1s
  unit_of_measurement: V
  accuracy_decimals: 5
  attenuation: 0db
  filters: 
    - exponential_moving_average: 
        alpha: 0.1
        send_every: 10
        send_first_at: 1
  disabled_by_default: False
  force_update: False
  device_class: voltage
  state_class: measurement
  raw: False`

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ADC

peejayw, 2023-12-21:
I am getting the same error "pin 17 is used in multiple locations." Just started.

makkie2002, 2023-12-21:
Same here. Broke my S0 watersensor: [https://github.com/huizebruin/s0tool](url).

```
INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/s0tool-watermeter-ad280f.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Brussels'
Failed config

sensor.pulse_meter: [source <unicode string>:66]
  
  Pin 4 is used in multiple places.
  platform: pulse_meter
  internal_filter_mode: PULSE
  internal_filter: 50ms
  pin: 
    number: 4
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  id: watermeter_flow
  name: watermeter flow
  state_class: measurement
  unit_of_measurement: l/min
  accuracy_decimals: 1
  icon: mdi:water-pump
  filters: 
    - lambda: !lambda |-
        return x * atof(id(Select_pulse_water).state.c_str()) * 1000;
  total: 
    id: sensor_pulse_meter_total
    name: watermeter Totaal
    state_class: total_increasing
    device_class: water
    icon: mdi:water
    unit_of_measurement: m³
    accuracy_decimals: 3
    filters: 
      - multiply: 0.001
    disabled_by_default: False
    force_update: False
  disabled_by_default: False
  force_update: False
  timeout: 5min
binary_sensor.gpio: [source <unicode string>:50]
  
  Pin 4 is used in multiple places.
  platform: gpio
  id: water_sensor_status
  name: Water sensor status
  internal: False
  pin: 
    number: 4
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(total_water_pulses) += 1;
  disabled_by_default: False
```

Murada99, 2023-12-21:
Try the following
`pin: `  
`  number: 4`
`  allow_other_uses: true  `

Ronnie1967, 2023-12-21:
I have the same error... my platform is ultrasonic and I am not using the gpio pins. 
How would I add in the above for: 

  - platform: ultrasonic 
       trigger_pin: D5
       echo_pin: D6

1st try
  - platform: ultrasonic 
       trigger_pin: D5
       allow_other_uses: true 
       echo_pin: D6

just adding it in gives me: 
sensor.ultrasonic: [source <unicode string>:50]
  platform: ultrasonic
  trigger_pin: D5
  
  [allow_other_uses] is an invalid option for [sensor.ultrasonic]. Please check the indentation.
  allow_other_uses: True
  echo_pin: D6

12nd try
  - platform: ultrasonic 
       trigger_pin: D5
           allow_other_uses: true 
       echo_pin: D6

mapping values are not allowed in this context
  in "<unicode string>", line 52, column 23

Murada99, 2023-12-21:
Can you try :

`trigger_pin:`
`  number: D5`  
`  allow_other_uses: true`

Ronnie1967, 2023-12-21:
Thanks @Murada99   that worked....

randybb, 2023-12-21:
It was [mentioned](https://next.esphome.io/changelog/2023.12.0.html#pin-reuse-validation) as a breaking change. Please read release notes first.

closed 2023-12-21 by randybb

ssieb, 2023-12-21:
There's a reason that validation was added.  You shouldn't be doing things like this.

peejayw, 2023-12-21:
So this is my watering sensor firmware, could sme kind person point me to what I need to get it working again (error is pin 17 is used in multiple places.)
Thanks.

INFO ESPHome 2023.12.1
INFO Reading configuration /config/esphome/192-168-1-28.yaml...
Failed config

sensor.adc: [source <unicode string>:47]
  
  Pin 17 is used in multiple places.
  platform: adc
  pin: 
    number: 17
    mode: 
      analog: True
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
  id: adcMoisture
  accuracy_decimals: 0
  unit_of_measurement: %
  filters: 
    - calibrate_linear: 
        datapoints: 
          - from: 0.375
            to: 100.0
          - from: 0.74023
            to: 0.0
        method: least_squares
  name: Plant - Moisture Level
  update_interval: 30s
  disabled_by_default: False
  force_update: False
  device_class: voltage
  state_class: measurement
  raw: False
sensor.adc: [source <unicode string>:58]
  
  Pin 17 is used in multiple places.
  platform: adc
  pin: 
    number: 17
    mode: 
      analog: True
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
  id: adcRaw_Sensor_Data
  accuracy_decimals: 5
  name: Plant - Sensor Voltage
  update_interval: 30s
  disabled_by_default: False
  force_update: False
  unit_of_measurement: V
  device_class: voltage
  state_class: measurement
  raw: False

peejayw, 2023-12-21:
esphome:
  name: "192-168-1-28"
  friendly_name: Watering Sensor
  platformio_options:
    board_build.f_cpu: 160000000L
esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "stl28kifE3Jt/309AT1xkz4oAbvYCCVUQQ2Gppt/kMc="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-9215B0"
    password: "OT5GbjKPwMas"

captive_portal:
web_server:
    port: 80

light:
  - platform: binary
    name: "Blue Light"
    output: light_output
    id: led_light
    restore_mode: ALWAYS_OFF

output:
  - platform: gpio
    pin: GPIO2
    id: light_output
    inverted: true

sensor:
- platform: adc
  pin: A0
  id: "adcMoisture"
  accuracy_decimals: 0
  unit_of_measurement: '%'
  filters:
    - calibrate_linear:
      - 0.37500  -> 100.0
      - 0.74023 -> 0
  name: "Plant - Moisture Level"
  update_interval: 30s
- platform: adc
  pin: A0
  id: "adcRaw_Sensor_Data"
  accuracy_decimals: 5
  name: "Plant - Sensor Voltage"
  update_interval: 30s
- platform: wifi_signal
  name: "Plant - WiFi Signal Sensor"
  update_interval: 30s

ssieb, 2023-12-22:
```yaml
sensor:
- platform: copy
  source_id: adcRaw_Sensor_Data
  id: "adcMoisture"
  accuracy_decimals: 0
  unit_of_measurement: '%'
  filters:
    - calibrate_linear:
      - 0.37500  -> 100.0
      - 0.74023 -> 0
  name: "Plant - Moisture Level"
- platform: adc
  pin: A0
  id: "adcRaw_Sensor_Data"
  accuracy_decimals: 5
  name: "Plant - Sensor Voltage"
  update_interval: 30s
```

If you have any more questions about it, come ask on the discord server.

peejayw, 2023-12-22:
Than you, that worked, dont know how but copy and paste was my friend and
my plants are grateful.