esphome/issues #5248: Another Pin validation issue starting in v2023.12.0

andypnz, 2023-12-21:
### The problem

With 2023.12.0 now unable to use one pin for two functions....

`INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/pc-power-control.yaml...
Failed config

switch.gpio: [source <unicode string>:31]
  
  Pin 5 is used in multiple places.
  platform: gpio
  name: PC Power Toggle
  icon: mdi:electric-switch
  pin: 
    number: 5
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  id: power_short_press
  inverted: False
  on_turn_on: 
    - then: 
        - delay: 150ms
        - switch.turn_off: 
            id: power_short_press
  disabled_by_default: False
  restore_mode: ALWAYS_OFF
  interlock_wait_time: 0ms
switch.gpio: [source <unicode string>:42]
  
  Pin 5 is used in multiple places.
  platform: gpio
  name: PC HARD POWER OFF
  icon: mdi:electric-switch
  pin: 
    number: 5
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  id: power_long_press
  inverted: False
  on_turn_on: 
    - then: 
        - delay: 3500ms
        - switch.turn_off: 
            id: power_long_press
  disabled_by_default: False
  restore_mode: ALWAYS_OFF
  interlock_wait_time: 0ms`

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP8266

### Board

WeMos D1 mini

Ronnie1967, 2023-12-21:
I am having the same issue. 
Board nodemcuv2

Murada99, 2023-12-21:
Try the following
`pin: `  
`  number: 5`
`   allow_other_uses: true  `

khrisperry, 2023-12-21:
I tried to add this pin: section, but maybe I'm not adding it in the right place.  

INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/watersoftner.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1019, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1062, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 916, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 904, in _load_config
    result = validate_config(config, command_line_substitutions)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 832, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 143, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 318, in run
    if component.multi_conf_no_default and isinstance(self.conf, core.AutoLoad):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'multi_conf_no_default'

randybb, 2023-12-21:
again the same - It was [mentioned](https://next.esphome.io/changelog/2023.12.0.html#pin-reuse-validation) as a breaking change. Please read release notes first.

closed 2023-12-21 by randybb

ssieb, 2023-12-21:
And there's a reason that was added.  You shouldn't be doing this.

cctl01, 2023-12-25:
The documentation on this issue is garbage.
The changelog only refers to this page https://esphome.io/guides/configuration-types#config-pin-schema where no mention of over-ruling is made.