esphome/issues #5249: XPT2046 touchsreen component - 2023.12 - compiler/editor marks some documented options as invalid

Zoogara, 2023-12-21:
### The problem

2023.12, XPT2046 touchscreen component.

Options `report_interval:` and `swap_x_y:` are reported as invalid by the compiler and editor, making installation impossible without removing them.

Options still appear in documentation.

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP32

### Board

DevKitC

### Component causing the issue

xpt2046

### Example YAML snippet

```yaml
touchscreen:
  platform: xpt2046
  id: touchscreen1
  cs_pin: 21
  interrupt_pin: 27
  update_interval: 50ms
  report_interval: 1s
  threshold: 350
  calibration_x_min: 280
  calibration_x_max: 3806
  calibration_y_min: 185
  calibration_y_max: 3884
  swap_x_y: false
```

Wummeke, 2023-12-22:
I used the code from the documentation, and it won't compile anymore. ( I already remove the `report_interval` option):

```
touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: 13
  interrupt_pin: 25
  update_interval: 50ms
  threshold: 400
  calibration_x_min: 372
  calibration_x_max: 3470
  calibration_y_min: 3640
  calibration_y_max: 264
  transform:
    mirror_x: false
    mirror_y: false
    swap_xy: false
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );
```

```
<unicode string>: In lambda function:
<unicode string>:49:27: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'touched'; did you mean 'touches_'?
<unicode string>:50:42: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'x'
<unicode string>:50:61: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'y'
```

nielsnl68, 2023-12-22:
Can you show the whole YAML script?
thanks

HugoTheBoss, 2023-12-22:
For me exact the same problem!

Wummeke, 2023-12-22:
> Can you show the whole YAML script? thanks

I think the documentation should reflect the current component. As the original poster mentiones, the documentation contains options that are not valid anymore with the overhauled component.

nielsnl68, 2023-12-22:
True, that is being done as we speak, I forgot to update the documents with the last changes i made. 
But the error does not make sense with the YAML part you shown above. So i want to be sure i dont miss something

Wummeke, 2023-12-22:
You were right, the errors I posted earlier were not making sense related to the code I pasted.
Here are the relevant parts of the YAML:
```
touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: 13
  interrupt_pin: 25
  update_interval: 50ms
  # report_interval: 1s
  threshold: 400
  calibration_x_min: 372
  calibration_x_max: 3470
  calibration_y_min: 3640
  calibration_y_max: 264
  transform:
    mirror_x: false
    mirror_y: true
    swap_xy: true
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              id(my_touchscreen).x,
              id(my_touchscreen).y,
              id(my_touchscreen).x_raw,
              id(my_touchscreen).y_raw
              );
```

And the correctly related error:
```
Compiling .pioenvs/radiocontrol/src/main.cpp.o
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
<unicode string>: In lambda function:
<unicode string>:125:27: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'x'
src/esphome/core/log.h:108:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
<unicode string>:124:7: note: in expansion of macro 'ESP_LOGI'
<unicode string>:126:27: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'y'
src/esphome/core/log.h:108:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
<unicode string>:124:7: note: in expansion of macro 'ESP_LOGI'
<unicode string>:127:27: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'x_raw'
src/esphome/core/log.h:108:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
<unicode string>:124:7: note: in expansion of macro 'ESP_LOGI'
<unicode string>:128:27: error: 'class esphome::xpt2046::XPT2046Component' has no member named 'y_raw'
src/esphome/core/log.h:108:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
<unicode string>:124:7: note: in expansion of macro 'ESP_LOGI'
*** [.pioenvs/radiocontrol/src/main.cpp.o] Error 1
========================= [FAILED] Took 18.61 seconds =========================
```

Wummeke, 2023-12-22:
Ah,  I see the documentation has been updated. The calibration code works fine now. My compiling issue has been solved.

nielsnl68, 2023-12-22:
```yaml
              id(my_touchscreen).x,
              id(my_touchscreen).y,
              id(my_touchscreen).x_raw,
              id(my_touchscreen).y_raw
```
The `id(my_touchscreen)` will not work anymore. you need to use `touch` here

Zoogara, 2023-12-22:
Thanks for the update. The touchscreen seems more responsive with this release.

nagyrobi, 2024-01-03:
Documentation updated.

closed 2024-01-03 by nagyrobi