esphome/issues #5250: [2023.12.0] Component ID tt21100_tt21100touchscreen was not declared to inherit from Component, or was registered twice.

ChristophCaina, 2023-12-21:
### The problem

updating to 2023.12.0 on the ESP32 S3 Box (the older one) is throwing the following error:

```
INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/esp32-box.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4793/head
INFO Updating https://github.com/esphome/esphome.git@pull/4861/head
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/bd715d55/esphome/components/tt21100/touchscreen/__init__.py", line 37, in to_code
    await touchscreen.register_touchscreen(var, config)
  File "/esphome/esphome/components/touchscreen/__init__.py", line 52, in register_touchscreen
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID tt21100_tt21100touchscreen was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
```

The configuration was just made as it is described in the documentation: https://esphome.io/components/touchscreen/tt21100.html?highlight=tt21100

```
binary_sensor:
  - platform: tt21100
    name: "Home"
    index: 0
    
touchscreen:
  - platform: tt21100
    address: 0x24
    interrupt_pin: GPIO3
```
    

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-Box

### Component causing the issue

tt21100

### Example YAML snippet

```yaml
binary_sensor:
  - platform: tt21100
    name: "Home"
    index: 0
    
touchscreen:
  - platform: tt21100
    address: 0x24
    interrupt_pin: GPIO3
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/esp32-box.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4793/head
INFO Updating https://github.com/esphome/esphome.git@pull/4861/head
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/bd715d55/esphome/components/tt21100/touchscreen/__init__.py", line 37, in to_code
    await touchscreen.register_touchscreen(var, config)
  File "/esphome/esphome/components/touchscreen/__init__.py", line 52, in register_touchscreen
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID tt21100_tt21100touchscreen was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
```

book12345678, 2023-12-22:
When I use ft63x6 , I encountered the same problem:
ValueError: Component ID touch_screen was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.

The code in the yanl file:
`external_components:
  ##- source: github://gpambrozio/esphome@FT6336U-touch
  - source:
      type: local
      path: my_components
    components: [ ft63x6 ]`

clydebarrow, 2023-12-24:
> When I use ft63x6 , I encountered the same problem:

You're using an external component, which will need updating by the author. However you should find that using the `ft5x06` touchscreen platform which is supported in ESPHome will work. Try it and see.

clydebarrow, 2023-12-24:
> The configuration was just made as it is described in the documentation: https://esphome.io/components/touchscreen/tt21100.html?highlight=tt21100

Can't reproduce the error with just that snippet - post your entire yaml here. If you are using `external_components` make sure that `tt21100` is _not_ specified there.

ChristophCaina, 2023-12-24:
> Can't reproduce the error with just that snippet - post your entire yaml here. If you are using `external_components` make sure that `tt21100` is _not_ specified there.

thanks, I guess, this was the reason then :)

closed 2023-12-24 by ChristophCaina

patfelst, 2023-12-29:
> > When I use ft63x6 , I encountered the same problem:
> 
> You're using an external component, which will need updating by the author. However you should find that using the `ft5x06` touchscreen platform which is supported in ESPHome will work. Try it and see.

I'm getting the same error as @book12345678 
`ValueError: Component ID touch_screen was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.`

howerver I'm not using an external component, I'm using the released component as follows:
```yaml
touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    interrupt_pin: 39
    id: touch_screen
    on_touch:
      - lambda: |-
          // ESP_LOGD("debug", "touch (x,y) = %d, %d", touch.x, touch.y);
          id(touch_x) = touch.x;
          id(touch_y) = touch.y;
```

clydebarrow, 2023-12-29:
> howerver I'm not using an external component, I'm using the released component as follows:
> 
> ```yaml
> touchscreen:
>   - platform: ft63x6
> ```

I can't reproduce a problem with the ft6x36. Please post your complete yaml and log output.

patfelst, 2023-12-29:
my apologies, it turns out I was using an external component, I thought I'd commented it out, but still had the source type set to local. Is compiling ok now.