esphome/issues #5251: Recent change to XPT2046 touchscreen driver removed `swap_x_y`

avwuff, 2023-12-21:
### The problem

A recent PR to the XPT 2046 touchscreen driver removed the `CONF_TRANSFORM` include, thus disabling the use of SWAP_X_Y, as can be seen in my old config here.
[image]

The file:
https://github.com/esphome/esphome/blob/74281b93c4491f80bfd02db9e56696a86081083c/esphome/components/xpt2046/touchscreen/__init__.py

Without this, using the touchscreen is basically impossible because the buttons are all in the wrong places.

Thanks!

### Which version of ESPHome has the issue?

 2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2023.12.0

### What platform are you using?

ESP32

### Board

Cheap Yellow Display

### Component causing the issue

touchscreen/xpt2046

### Example YAML snippet

```yaml
# Set up the xpt2046 touch platform
touchscreen:
  platform: xpt2046
  id: touchid
  spi_id: spi_touch
  cs_pin: GPIO33
  interrupt_pin: GPIO36
  update_interval: 50ms
  threshold: 400
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_min: 340
  calibration_y_max: 3860
  swap_x_y: false
```

nielsnl68, 2023-12-21:
Those settings are now moved for all existing touchscreen drivers under **`transform`**  like:

```yaml
touchscreen:
   platform: xpt2046
   transform:
       mirror_x: false
       mirror_y: false
       swap_xy: false
```

Zoogara, 2023-12-21:
Any news on when there will be documentation for this available?

nagyrobi, 2024-01-03:
Doc updated.

closed 2024-01-03 by nagyrobi