esphome/issues #5252: 2023.12.0 breaks ESP32 IR fan from working

billmaterial, 2023-12-21:
### The problem

Updating to 2023.12.0 breaks ESP32 IR fan switches from working ("FAN On/Off" switch entity not found) and the other IR fan controls not working.

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

ESP32-DEV

### Component causing the issue

IR

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-d62e00
  friendly_name: ESPHome Web d62e00
  on_boot:
    priority: -100
    then: 
    - lambda: id(my_display).set_contrast(.01);

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
 #level: VERBOSE
 baud_rate: 115200
 deassert_rts_dtr: true

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxx"

ota:
  password: !secret ota_password

wifi:
#  power_save_mode: none
  fast_connect: off
  networks:
  - ssid: !secret wifi_ssid_Ghz
    password: !secret wifi_password_both
  - ssid: !secret wifi_ssid_GHz
    password: !secret wifi_password_both

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-D62E00"
[... 215 lines cut ...]
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    update_interval: 1s
    contrast: 1%
    id: my_display 
    lambda: |-
      if (id(esp_connected_ssid).has_state()){
        it.printf(64, 0, id(my_font_small), TextAlign::TOP_CENTER, "%s", id(esp_connected_ssid).state.c_str());
      }
      it.printf(0, 28, id(my_font), TextAlign::BASELINE_LEFT, "WiFi Signal");
      if (id(wifi_signal_db).has_state()){
        it.printf(127, 28, id(my_font), TextAlign::BASELINE_RIGHT, "%1.0fdB", id(wifi_signal_db).state);
      }
      
      it.printf(0, 45, id(my_font), TextAlign::BASELINE_LEFT, "LED State");
      if ((id(led_status).state) == 0) {
        it.printf(127, 45, id(my_font), TextAlign::BASELINE_RIGHT, "OFF");
      }
      if ((id(led_status).state) == 1) {
        it.printf(127, 45, id(my_font), TextAlign::BASELINE_RIGHT, "ON");
      }

      it.strftime(64, 50, id(my_font), TextAlign::TOP_CENTER, "%H:%M:%S", id(esptime).now());
```

billmaterial, 2023-12-21:
Fixed the issue, the switch was renamed with the update, original name was "switch.esphome_web_d62e00_fan_on_off_2" now it is "switch.esphome_web_d62e00_fan_on_off".
My bad!

closed 2023-12-21 by billmaterial

MarcoL1975, 2023-12-21:
Have the same problem, all my sensors etc are renamed to x_2! Did a rollback to previous version!