esphome/issues #5256: Component ID eink_display was not declared to inherit from Component, or was registered twice

Nicxe, 2023-12-21:
### The problem

I can´t update Firmware on ESP32 after update to ESPHome 2023.12.0 in HomeAssistant. I get the error "ValueError: Component ID eink_display was not declared to inherit from Component, or was registered twice"

```
INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/f1pitboard.yaml...
INFO Detected timezone 'Europe/Stockholm'
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/fc00875a/esphome/components/waveshare_epaper/display.py", line 148, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID eink_display was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
```

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

ESP32

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source:
      type: git
      url: https://github.com/prapador/esphome_waveshare_fix
      ref: dev
    components: [ waveshare_epaper ]
```

rikroe, 2023-12-21:
I'm seeing the same thing with another waveshare external component (YAML below). I also tried rebasing the external_component to 2023.12 

However as I'm experiencing massive connection issues from 2023.12 (#5257) I'm not sure if the error just occurs because ESPHome cannot download/connect to Github.

```yaml
display:
  - platform: waveshare_epaper
    id: epaper_display
    cs_pin: 19
    dc_pin: 21
    busy_pin: 22
    reset_pin: 23
    model: 2.90in-dke
    #Partial not working yet so full only
    full_update_every: 1
    update_interval: never
    lambda: |-
      ...
    rotation: 270

external_components:
  - source:
      type: git
      url: https://github.com/rikroe/esphome
      ref: dke2.9-epaper
    components: [ waveshare_epaper ]

```

guillempages, 2023-12-21:
There was a change in the display structure that breaks all external display components; the fix needs to be done in the external component. See https://github.com/landonr/lilygo-tdisplays3-esphome/pull/46 for an example of the needed changes to have this compile again

Nicxe, 2023-12-21:
Thanks, I disabled the external components and now I can update.

closed 2023-12-21 by Nicxe