esphome/issues #5257: 2023.12.0 DNS issues / ESPHome causes denial of service on DNS server

rikroe, 2023-12-21:
### The problem

I am using ESPHome with a local PiHole DNS server on the same docker host. PiHole is exposed via a macvlan IP (but this shouldn't be the issue).

After upgrading to 2023.12, i noticed that some services on my docker host cannot do DNS resolution anymore. Even local DNS resolution on the host fails.

ESPHome Dashboard also cannot find the devices anymore, whereas Home Assistant (in a seperate Container) still talks to the devices.

I then realized that the fans on my host are spinning a lot and found the following in htop:
[image]

As soon as I stop the ESPHome container, pihole-FTL CPU usage goes down to 0 as well and local DNS resolution on the host works again.

I have now downgroaded to ESPHome 2023.11.6 and don't have this issue anymore. Devices are also shown in the ESPHome dashboard.

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32-IDF

### Board

n/a

### Component causing the issue

n/a

richli, 2023-12-22:
It looks like once I visit the dashboard, then a bunch of `ping` processes keep getting spawned over and over (I notice the PIDs increasing), one for each device. Pretty quickly it hits the pihole rate limits, but the ping storm only ends once I kill the esphome container.

It must be a recent change in how the dashboard checks for the online status?

rikroe, 2023-12-22:
Do your devices even appear online? In my dashboard with 2023.12.0, all devices were shown as offline.

When exec'ing into the docker image, however I am able to get working name resolution:
```
root@ab50928455a1:/config# curl badge-01.fritz.box
curl: (7) Failed to connect to badge-01.fritz.box port 80 after 6 ms: Couldn't connect to server
```

And can confirm - as soon as the dashboard is opened, DNS queries are spammed. As all of the queries are coming from the host interface via macvlan, all query from my server get blocked.

PiHole log for having the dashboard open for ~30 seconds before stopping the docker container:
```
[2023-12-22 10:50:19.148 229M] Rate-limiting 192.168.91.145 for at least 46 seconds
[2023-12-22 10:51:05.317 229/T254] Still rate-limiting 192.168.91.145 as it made additional 243514 queries
[2023-12-22 10:52:05.370 229/T254] Ending rate-limitation of 192.168.91.145
```

No difference with 2023.12.3.

richli, 2023-12-22:
My devices do show up as online. It's probably worth noting that when I run esphome, I set the environment variable `ESPHOME_DASHBOARD_USE_PING` to `true`.

And when I exec into the container, I can also resolve my device names. The issue doesn't appear to be that DNS is broken but that for some reason it's keeps ping-ing all the hosts without any delay.

richli, 2023-12-22:
Checking in the last month, esphome/esphome#5749 updated the functionality to spawn the `ping` processes. Nothing seems that suspect, although it would be really beneficial here to add a delay when retrying a ping to a host. Even 1 or 5 seconds would help considerably.

It looks like the `PingStatus` object only retries hosts where the `ping` process returns non-zero. I wonder if in my case, it *is* returning non-zero. My `esphome` container has both IPv4 and IPv6 connectivity, whereas my esphome devices only have IPv4 addresses (they all use `esp8266` boards and, as I understand, don't support IPv6).

Inside my `esphome` container, resolving the IPv4 address to one of my devices works, so the `ping` exit code is `0`:

```
# ping -c 1 -4 plug01.home.arpa; echo $?   
PING plug01.home.arpa (172.16.16.142) 56(84) bytes of data.
64 bytes from plug01.home.arpa (172.16.16.142): icmp_seq=1 ttl=254 time=2.65 ms

--- plug01.home.arpa ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 2.647/2.647/2.647/0.000 ms
0
```

But it doesn't have an IPv6 address, so the `ping` exit code is `2`:
```
# ping -c 1 -6 plug01.home.arpa; echo $?
ping: plug01.home.arpa: Address family for hostname not supported
2
```

Note that this is the same exit code, but for a different reason, as when the host doesn't exist:

```
# ping -c 1 -4 void.home.arpa; echo $?     
ping: void.home.arpa: Name or service not known
2
```

closed 2023-12-23 by jesserockz

rikroe, 2023-12-23:
Can confirm that #6001 fixes the issue. Thanks @bdraco!