esphome/issues #5258: AsyncTCP 2.1.0 does not compile

homonto, 2023-12-21:
### The problem

```
Compiling .pio/build/box42/lib962/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
.pio/libdeps/box42/AsyncTCP-esphome/src/AsyncTCP.cpp:753:2: error: #else without #if
 #else
  ^~~~
.pio/libdeps/box42/AsyncTCP-esphome/src/AsyncTCP.cpp:755:2: error: #endif without #if
 #endif
  ^~~~~
.pio/libdeps/box42/AsyncTCP-esphome/src/AsyncTCP.cpp:1020: error: unterminated #if
 #if LWIP_IPV4 && LWIP_IPV6
```
here:

```
bool AsyncClient::connect(const char* host, uint16_t port){
    ip_addr_t addr;

    if(!_start_async_task()){
      log_e("failed to start task");
      return false;
    }

    err_t err = dns_gethostbyname(host, &addr, (dns_found_callback)&_tcp_dns_found, this);
    if(err == ERR_OK) {
        if(addr.type == IPADDR_TYPE_V6) {
            return connect(IPv6Address(addr.u_addr.ip6.addr), port);
        }
        return connect(IPAddress(addr.u_addr.ip4.addr), port);
#else
        return connect(IPAddress(addr.addr), port);
#endif
    } else if(err == ERR_INPROGRESS) {
        _connect_port = port;
        return true;
    }
    log_e("error: %d", err);
    return false;
}
```

### Which version of ESPHome has the issue?

AsyncTCP-esphome @ ^2.1.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP8266

jcwren, 2023-12-21:
I experienced the same issue. It would be useful to mention in the README.md where to post about issues since the AsyncTCP-esphome repository itself does not have issue tracking enabled.

homonto, 2023-12-21:
> I experienced the same issue. It would be useful to mention in the README.md where to post about issues since the AsyncTCP-esphome repository itself does not have issue tracking enabled.

bloody hell - I spent ages to find where to post it! ;-)

mrengineer7777, 2023-12-21:
I have this issue as well.
Using library esphome/ESPAsyncWebServer-esphome@^2.1.0 which has compiled fine for over 6 months.
Previously this library referenced [AsyncTCP](https://github.com/esphome/AsyncTCP) ^2.0.1.  Now it's building with AsyncTCP 2.1.0 and throwing the compile errors noted above.

And yes it took forever to find where to post this.

cristians-12, 2023-12-25:
Hello! If you wanna solve the problem you should uninstall libraries related to AsyncTCP 2.1.0 and install the 2.0.0 versions, that solved the problem for me.

homonto, 2023-12-25:
@cristians-12  this is NOT a solution - this is a work around

hsingor, 2023-12-25:
You can try this, It worked for me:

you add the missing #if like this (at line 749):

```
    err_t err = dns_gethostbyname(host, &addr, (dns_found_callback)&_tcp_dns_found, this);
    if(err == ERR_OK) {
#if LWIP_IPV4 && LWIP_IPV6        /// <-----  this is the missing #if
        if(addr.type == IPADDR_TYPE_V6) {
            return connect(IPv6Address(addr.u_addr.ip6.addr), port);
        }
        return connect(IPAddress(addr.u_addr.ip4.addr), port);
#else
        return connect(IPAddress(addr.addr), port);
#endif
```
and you delete or comment this line  (at line1020) :

```
void AsyncClient::_dns_found(struct ip_addr *ipaddr){
// #if LWIP_IPV4 && LWIP_IPV6    /// <-------   you comment or delete this line
    if(ipaddr && ipaddr->u_addr.ip4.addr){
        connect(IPAddress(ipaddr->u_addr.ip4.addr), _connect_port);
    } else if(ipaddr && ipaddr->u_addr.ip6.addr){
        connect(IPv6Address(ipaddr->u_addr.ip6.addr), _connect_port);
    } else {
        if(_error_cb) {
            _error_cb(_error_cb_arg, this, -55);
        }
        if(_discard_cb) {
            _discard_cb(_discard_cb_arg, this);
        }
    }
} 
```

homonto, 2023-12-25:
this is a WORK AROUND
solution is to be done by the author or so of the library (maintainer) and updated the library

Allex303, 2023-12-25:
@hsingor  Thank you very much! You solved my problem too! It works!

meta-space, 2024-01-02:
I was facing the same problem. After finding this thread, I created an upstream PR https://github.com/esphome/AsyncTCP/pull/7

HeMan, 2024-02-27:
Is it still a problem with 2.1.3?

mrengineer7777, 2024-02-27:
> Is it still a problem with 2.1.3?

Not sure.  I rolled back to AsyncTCP 2.0.1 and branched it so upstream changes don't break our projects again.

meta-space, 2024-02-28:
@HeMan I haven't gotten around to testing it. It appears that the minimal changes I suggested in my PR have already been incorporated to the main branch in the past couple of weeks which would my pull request obsolete.  I haven't gotten any response from the them, though.

HeMan, 2024-03-21:
Could this be closed?

HeMan, 2024-03-23:
AsyncTCP 2.1.3 should work.

closed 2024-03-23 by HeMan