esphome/issues #5260: Duplicated items (but reallly not) after last update

VladNord, 2023-12-21:
### The problem

```
deep_sleep:
  run_duration: 3min
  sleep_duration: 1min
  id: deep_sleep_1
  wakeup_pin: 39
  wakeup_pin_mode: INVERT_WAKEUP

```

Error the following:
 Pin 39 is used in multiple places

```
mi:
  
  id: mi1
  ce_pin: 4
  csn_pin: 5
  reset_pin: 0
```

Same situation with MI custom component, where is reset pin  perceived as duplicated gpio0

### Which version of ESPHome has the issue?

2023.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12

### What platform are you using?

ESP32

sheaur, 2023-12-21:
Add "allow_other_uses: true"

VladNord, 2023-12-21:
> ```
> deep_sleep:
>   run_duration: 3min
>   sleep_duration: 1min
>   id: deep_sleep_1
>   wakeup_pin: 39
>   allow_other_uses: true
>   wakeup_pin_mode: INVERT_WAKEUP
> ```

[image]

not possible :(

sheaur, 2023-12-21:
Where is the other pin 39?

VladNord, 2023-12-21:
> Where is the other pin 39?

```
binary_sensor:

#ac_in
  - platform: gpio
    pin:
      number: GPIO39
      inverted: true
      allow_other_uses: true
    name: "Home AC IN"
    id: home_ac_in
    device_class: power
    filters:
      - delayed_on_off: 1s
    on_state:
      then:
        - if:
            condition:
              - binary_sensor.is_on: home_ac_in
            then:
              - deep_sleep.prevent: deep_sleep_1
              - rtttl.play: ${ac_on_song}
        - if:
            condition:
              - binary_sensor.is_off: home_ac_in
            then:
              - deep_sleep.prevent: deep_sleep_1
              - rtttl.play: ${ac_off_song}
```

sheaur, 2023-12-21:
Delete "allow_other_uses: true" in deep_sleep and try

VladNord, 2023-12-21:
> Delete "allow_other_uses: true" in deep_sleep and try

[image]

Done

taygs, 2023-12-22:
I'm having a similar problem, started with 2023.12.0

I have 3x MCP23017's and I'm getting the same "Pin X is used in multiple places." messages for all GPIOs.

Still works in 2023.11.6.

[image]

ssieb, 2023-12-22:
@taygs that was fixed in the last point release.

taygs, 2023-12-22:
@ssieb I'm now on 2023.12.2 and still have the messages.

fixed77, 2023-12-22:
2023.12.2. problem is not resolved.

ssieb, 2023-12-22:
@sheaur @VladNord you're putting it in the wrong place.  It needs to be in the pin schema as described in the docs.

jesserockz, 2023-12-22:
@taygs @fixed77 can you please paste your entire config files so I can attempt to reproduce?

I cannot reproduce the error with the above screenshot config only as it works just fine which points to something else in your config.

ssieb, 2023-12-22:
I tried the following using 2023.12.0 and there were no errors:
```yaml
mcp23017:
  - id: mcp23017_1
  - id: mcp23017_2

binary_sensor:
  - platform: gpio
    id: mcp23017_pin_1
    pin:
      number: 1
      mcp23xxx: mcp23017_1
  - platform: gpio
    id: mcp23017_pin_2
    pin:
      number: 1
      mcp23xxx: mcp23017_2
```

taygs, 2023-12-22:
@jesserockz & @ssieb I have been able to resolve my issue.  There were some actual duplicates hiding in the ±1400 lines of config.  Config will now validate.  Thank you for all the hard work you guys do!

closed 2023-12-22 by clydebarrow