esphome/issues #5261: "ssd1351_spi" fails to display anything other than numbers

3ative, 2023-12-21:
### The problem

"ssd1351_spi" now can only display numbers after 2023.12.0 / 2023.12.1 update.  

Not working here are:
* Graphs
* %s values: I.E. "id(xxxxxx).state.c_str()"

### Which version of ESPHome has the issue?

2023.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

Generic ESP32 Dev Board

### Component causing the issue

display:   - platform: ssd1351_spi

### Example YAML snippet

```yaml
text_sensor:
  - platform: template
    id: co2_warning

graph:
  - id: graph1
    duration: 60min
    x_grid: 10min
    y_grid: 500
    width: 123
    height: 50
    traces:
      - sensor: pms25
        line_type: SOLID
        line_thickness: 1
        color: oled_yellow

display:
  - platform: ssd1351_spi
    id: oled
    model: "SSD1351 128x128"
    reset_pin: GPIO16
    brightness: 0.5 # Default 1.0
    cs_pin: GPIO04
    dc_pin: GPIO05
    lambda: |-
      it.printf(68, 44, id(verdana), TextAlign::CENTER, "%s", id(co2_warning).state.c_str());
      it.graph(2, 62, id(graph1), oled_grid);
```

### Anything in the logs that might be useful for us?

```txt
[16:29:22][W][component:214]: Component display took a long time for an operation (0.10 s).
[16:29:22][W][component:215]: Components should block for at most 20-30ms.
```

### Additional information

Rebooting and re-flashing (_OTA & USB_) didn't help.

3ative, 2023-12-21:
Additional:
Doing a partial restore didn't help - I had to do a Full Home Assistant Restore _and thus get ESPHome 2023.11.6 back_ to get it working again.

3ative, 2023-12-21:
Additional (additional):
Logs have changed. Now getting
```yaml
[18:08:12][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[18:08:12][W][component:215]: Components should block for at most 20-30ms.
```

3ative, 2024-01-09:
It's working after ESPHome update: 2023.12.5 🥳

3ative, 2024-01-30:
Is anyone working on this? 
I's stuck on 2023.11.6 and I can't update ESPHome (or make any more tutorials) until the latest version has fixes. :(

nielsnl68, 2024-01-30:
Not yet, you are the first reporting a problem with this display.

Can you show the startup log and part of the running log with `level: verbose`
Then i can check it out.

3ative, 2024-01-30:
> Not yet, you are the first reporting a problem with this display.
> 
> Can you show the startup log and part of the running log with `level: verbose` Then i can check it out.

Fantastic. Thank you for the response - I'll have to spin-up another instance of HA/ESPHome, update the code and report back...
- Gimme a few hours.

dison4linux, 2024-01-30:
I stumbled across this thread when searching for why I'm seeing "Component display took a long time for an operation"
I'll upload my logs though I'm not seeing the exact issue mentioned by the OP.
[logs_ssd1306_logs.txt](https://github.com/esphome/issues/files/14103286/logs_ssd1306_logs.txt)

3ative, 2024-01-30:
@nielsnl68 
Ok, I've updated to the latest everything, re-flashed the ESP32, powered cycled, etc.
Here's the Log output:
```yaml
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/1-oled.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 1-oled.local using esphome API
INFO Successfully connected to 1-oled @ 192.168.0.220 in 6.144s
INFO Successful handshake with 1-oled @ 192.168.0.220 in 0.039s
[19:30:57][I][app:102]: ESPHome version 2023.12.9 compiled on Jan 30 2024, 18:56:57
[19:30:57][C][wifi:573]: WiFi:
[19:30:57][C][wifi:405]:   Local MAC: 7C:9E:BD:EE:BC:4C
[19:30:57][C][wifi:410]:   SSID: [redacted]
[19:30:57][C][wifi:411]:   IP Address: 192.168.0.220
[19:30:57][C][wifi:413]:   BSSID: [redacted]
[19:30:57][C][wifi:414]:   Hostname: '1-oled'
[19:30:57][C][wifi:416]:   Signal strength: -63 dB ▂▄▆█
[19:30:57][V][wifi:418]:   Priority: 0.0
[19:30:57][C][wifi:420]:   Channel: 11
[19:30:57][C][wifi:421]:   Subnet: 255.255.255.0
[19:30:57][C][wifi:422]:   Gateway: 192.168.0.1
[19:30:57][C][wifi:423]:   DNS1: 194.168.4.100
[19:30:57][C][wifi:424]:   DNS2: 194.168.8.100
[19:30:57][C][logger:439]: Logger:
[19:30:57][C][logger:440]:   Level: VERBOSE
[19:30:57][C][logger:441]:   Log Baud Rate: 115200
[19:30:57][C][logger:443]:   Hardware UART: UART0
[19:30:57][C][spi:067]: SPI bus:
[19:30:57][C][spi:068]:   CLK Pin: GPIO18
[19:30:57][C][spi:069]:   SDI Pin: 
[19:30:57][C][spi:070]:   SDO Pin: GPIO19
[19:30:57][C][spi:072]:   Using HW SPI: SPI
[19:30:57][C][template.number:050]: Template Number 'Brightness'
[19:30:57][C][template.number:051]:   Optimistic: YES
[19:30:57][C][template.number:052]:   Update Interval: 60.0s
[19:30:57][C][template.text_sensor:020]: Template Sensor 'co2_warning'
[19:30:57][C][graph:364]: Graph for sensor pms25
[19:30:57][C][ssd1351_spi:022]: SPI SSD1351
[19:30:57][C][ssd1351_spi:022]:   Rotations: 0 °
[19:30:57][C][ssd1351_spi:022]:   Dimensions: 128px x 128px
[19:30:57][C][ssd1351_spi:023]:   Model: SSD1351 128x128
[19:30:57][C][ssd1351_spi:024]:   CS Pin: GPIO4
[19:30:57][C][ssd1351_spi:025]:   DC Pin: GPIO5
[19:30:57][C][ssd1351_spi:026]:   Reset Pin: GPIO16
[19:30:57][C][ssd1351_spi:027]:   Initial Brightness: 0.50
[19:30:57][C][ssd1351_spi:028]:   Update Interval: 1.0s
[19:30:57][C][mdns:115]: mDNS:
[19:30:57][C][mdns:116]:   Hostname: 1-oled
[19:30:57][V][mdns:117]:   Services:
[19:30:57][V][mdns:119]:   - _esphomelib, _tcp, 6053
[19:30:57][V][mdns:121]:     TXT: version = 2023.12.9
[19:30:57][V][mdns:121]:     TXT: mac = 7c9ebdeebc4c
[19:30:57][V][mdns:121]:     TXT: platform = ESP32
[19:30:57][V][mdns:121]:     TXT: board = esp32dev
[19:30:57][V][mdns:121]:     TXT: network = wifi
[19:30:57][C][ota:097]: Over-The-Air Updates:
[19:30:57][C][ota:098]:   Address: 1-oled.local:3232
[19:30:57][C][api:139]: API Server:
[19:30:57][C][api:140]:   Address: 1-oled.local:6053
[19:30:57][C][api:144]:   Using noise encryption: NO
[19:30:57][C][homeassistant.sensor:030]: Homeassistant Sensor 'current_temp'
[19:30:57][C][homeassistant.sensor:030]:   State Class: ''
[19:30:57][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[19:30:57][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[19:30:57][C][homeassistant.sensor:031]:   Entity ID: 'sensor.bedroom_temperature'
[19:30:57][C][homeassistant.sensor:030]: Homeassistant Sensor 'ap_current_speed'
[19:30:57][C][homeassistant.sensor:030]:   State Class: ''
[... 216 lines cut ...]
[19:31:48][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:48][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:48][W][component:215]: Components should block for at most 20-30ms.
[19:31:49][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:49][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:49][W][component:215]: Components should block for at most 20-30ms.
[19:31:50][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:50][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:50][W][component:215]: Components should block for at most 20-30ms.
[19:31:51][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:51][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:51][W][component:215]: Components should block for at most 20-30ms.
[19:31:52][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:52][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:52][W][component:215]: Components should block for at most 20-30ms.
[19:31:53][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:53][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:53][W][component:215]: Components should block for at most 20-30ms.
[19:31:54][V][graph:164]: Updating graph. ymin 0.000000, ymax 1000.000000
[19:31:54][W][component:214]: Component display took a long time for an operation (0.10 s).
[19:31:54][W][component:215]: Components should block for at most 20-30ms.
```
The "Text sensor seems to be consistent now, as it's updating when I trigger the sensor. 
However, the 'Graph' seem to stop after displaying a few data points - I get a short trace then it's frozen again.
- Rebooting repeats this.

nielsnl68, 2024-01-30:
So it seems an Graph issue is it not?
I belief i have seen some Graph issues popping around here and possible on the esphome discussion forum's. I have no knowledge/experience's with that component. so i cant help you much but for someone that has used it already i hope they can help you.

3ative, 2024-01-30:
> So it seems an Graph issue is it not? I belief i have seen some Graph issues popping around here and possible on the esphome discussion forum's. I have no knowledge/experience's with that component. so i cant help you much but for someone that has used it already i hope they can help you.

Ah ok.... thanks for trying.

3ative, 2024-01-30:
FYI - Here's the log output from ESPHome V2023.11.6 (I re-flashed it back on my original install)
And even thou the logs still show:
```yaml
[19:58:40][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:58:40][W][component:215]: Components should block for at most 20-30ms.
```
Albeit less large blocks of them, the 'Graph' continues to work.

```yaml
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/1-oled.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 1-oled.local using esphome API
INFO Successfully connected to 1-oled in 1.743s
INFO Successful handshake with 1-oled in 0.039s
[19:57:42][I][app:102]: ESPHome version 2023.11.6 compiled on Jan 30 2024, 18:55:34
[19:57:42][C][wifi:559]: WiFi:
[19:57:42][C][wifi:391]:   Local MAC: 7C:9E:BD:EE:BC:4C
[19:57:42][C][wifi:396]:   SSID: [redacted]
[19:57:42][C][wifi:397]:   IP Address: 192.168.0.220
[19:57:42][C][wifi:399]:   BSSID: [redacted]
[19:57:42][C][wifi:400]:   Hostname: '1-oled'
[19:57:42][C][wifi:402]:   Signal strength: -68 dB ▂▄▆█
[19:57:42][V][wifi:404]:   Priority: 0.0
[19:57:42][C][wifi:406]:   Channel: 11
[19:57:42][C][wifi:407]:   Subnet: 255.255.255.0
[19:57:42][C][wifi:408]:   Gateway: 192.168.0.1
[19:57:42][C][wifi:409]:   DNS1: 194.168.4.100
[19:57:42][C][wifi:410]:   DNS2: 194.168.8.100
[19:57:42][C][logger:416]: Logger:
[19:57:42][C][logger:417]:   Level: VERBOSE
[19:57:42][C][logger:418]:   Log Baud Rate: 115200
[19:57:42][C][logger:420]:   Hardware UART: UART0
[19:57:42][C][spi:067]: SPI bus:
[19:57:42][C][spi:068]:   CLK Pin: GPIO18
[19:57:42][C][spi:069]:   SDI Pin: 
[19:57:42][C][spi:070]:   SDO Pin: GPIO19
[19:57:42][C][spi:072]:   Using HW SPI: SPI
[19:57:42][C][template.number:050]: Template Number 'Brightness'
[19:57:42][C][template.number:051]:   Optimistic: YES
[19:57:42][C][template.number:052]:   Update Interval: 60.0s
[19:57:42][C][template.text_sensor:020]: Template Sensor 'co2_warning'
[19:57:42][C][graph:364]: Graph for sensor pms25
[19:57:42][C][ssd1351_spi:022]: SPI SSD1351
[19:57:42][C][ssd1351_spi:022]:   Rotations: 0 °
[19:57:42][C][ssd1351_spi:022]:   Dimensions: 128px x 128px
[19:57:42][C][ssd1351_spi:023]:   Model: SSD1351 128x128
[19:57:42][C][ssd1351_spi:024]:   CS Pin: GPIO4
[19:57:42][C][ssd1351_spi:025]:   DC Pin: GPIO5
[19:57:42][C][ssd1351_spi:026]:   Reset Pin: GPIO16
[19:57:42][C][ssd1351_spi:027]:   Initial Brightness: 0.50
[19:57:42][C][ssd1351_spi:028]:   Update Interval: 1.0s
[19:57:42][C][mdns:115]: mDNS:
[19:57:42][C][mdns:116]:   Hostname: 1-oled
[19:57:42][V][mdns:117]:   Services:
[19:57:42][V][mdns:119]:   - _esphomelib, _tcp, 6053
[19:57:42][V][mdns:121]:     TXT: version = 2023.11.6
[19:57:42][V][mdns:121]:     TXT: mac = 7c9ebdeebc4c
[19:57:42][V][mdns:121]:     TXT: platform = ESP32
[19:57:42][V][mdns:121]:     TXT: board = esp32dev
[19:57:42][V][mdns:121]:     TXT: network = wifi
[19:57:42][C][ota:097]: Over-The-Air Updates:
[19:57:42][C][ota:098]:   Address: 1-oled.local:3232
[19:57:42][C][api:139]: API Server:
[19:57:42][C][api:140]:   Address: 1-oled.local:6053
[19:57:42][C][api:144]:   Using noise encryption: NO
[19:57:42][C][homeassistant.sensor:030]: Homeassistant Sensor 'current_temp'
[19:57:42][C][homeassistant.sensor:030]:   State Class: ''
[... 508 lines cut ...]
[19:59:39][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:39][W][component:215]: Components should block for at most 20-30ms.
[19:59:40][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:40][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:40][W][component:215]: Components should block for at most 20-30ms.
[19:59:41][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:41][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:41][W][component:215]: Components should block for at most 20-30ms.
[19:59:42][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:42][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:42][W][component:215]: Components should block for at most 20-30ms.
[19:59:43][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:43][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:43][W][component:215]: Components should block for at most 20-30ms.
[19:59:44][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:44][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:44][W][component:215]: Components should block for at most 20-30ms.
[19:59:45][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:45][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:45][W][component:215]: Components should block for at most 20-30ms.
[19:59:46][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
[19:59:46][W][component:214]: Component ssd1351_base took a long time for an operation (0.10 s).
[19:59:46][W][component:215]: Components should block for at most 20-30ms.
[19:59:47][V][graph:164]: Updating graph. ymin 500.000000, ymax 1000.000000
```

3ative, 2024-02-01:
Tonight's testing...
I was also running an animated GIF (Bottom Left):
[image]

I've removed that now to see if that stops the Graph from freezing. I'll leave it running for a few hours and report back.

3ative, 2024-02-01:
Well this is weird! **No changes made to either HA or ESPHome installs...**
* Each stage was run for 1hr+ and each time the Graph updating: 

> - Removing the animated GIF and code
> - Power-Cycling the board
> - Reset via on-board button
> - Replaced GIF and code

So, I'm back where I started. Except this time the Graph has continued updating - _Which makes no sense !_

3ative, 2024-03-04:
Closing as after update 2024.2.0 everything's' working now. 
- Even though there was no mention of changes! Weird 😊

closed 2024-03-04 by 3ative