esphome/issues #5262: compiliation fails for raspberry pi pico after update

ausil, 2023-12-21:
### The problem

A previously working device no longer compiles after updating to 2023.12.1

```
INFO ESPHome 2023.12.1
INFO Reading configuration /config/esphome/picow.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picow (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30600.0 (3.6.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.4
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Linking .pioenvs/picow/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: unrecognized option '--no-warn-rwx-segments'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: use the --help option for usage information
collect2: error: ld returned 1 exit status
*** [.pioenvs/picow/firmware.elf] Error 1
========================= [FAILED] Took 23.62 seconds =========================
```

### Which version of ESPHome has the issue?

2023.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

platform

billmaterial, 2023-12-21:
I had the same problem updating to 2023.12.0, I rolled back to 2023.11.6 recompiled, installed 2023.12.0 again and was able to recompile with no issues. I have just updated to 2023.12.1 and the picow compiles with no issues. Hope this helps.

mava75, 2023-12-21:
Thanks for tip @billmaterial that worked find for me.

ausil, 2023-12-21:
That worked here. I noticed that gcc was updated to a version 12 build when dropping esphome back to 2023.11.6 so seemingly the bug was not triggering an update of the toolchain correctly.

danielkent-net, 2023-12-22:
> I had the same problem updating to 2023.12.0, I rolled back to 2023.11.6 recompiled, installed 2023.12.0 again and was able to recompile with no issues. I have just updated to 2023.12.1 and the picow compiles with no issues. Hope this helps.

This process didn't work for me; what did work was removing the PlatformIO files (on my system it was ~/.platformio) and recompiling a configuration, which triggered a redownload of the toolchain. I'm not sure if this is an esphome bug, a PlatformIO bug, or a bug in maxgerhardt/platform-raspberrypi.

jshep321, 2023-12-23:
Same issue here with 2023.12.3.  I can't find the .platformio directory to save my life (even using sudo find ...)

HarvsG, 2023-12-23:
Same issue here.
```
INFO ESPHome 2023.12.3
INFO Reading configuration /config/esphome/front-door.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing front-door (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30600.0 (3.6.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.4
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Linking .pioenvs/front-door/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: unrecognized option '--no-warn-rwx-segments'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: use the --help option for usage information
collect2: error: ld returned 1 exit status
*** [.pioenvs/front-door/firmware.elf] Error 1
========================= [FAILED] Took 22.97 seconds =========================
```

ssieb, 2023-12-23:
If you're using the addon, then you need to ssh into the host or container to access the files.  The easier method would be to re-install the addon.

HarvsG, 2023-12-23:
```sh
ssh hassio@homeasssistant.local
sudo docker ps
```
Made a note of the container ID of esp home `bb1547142a11`
I entered the shell of the container
```sh
docker exec -it bb1547142a11 /bin/bash
rm ~/.platformio/*
```

Then I re-tried the install from the UI

ssieb, 2023-12-23:
Yes, the `/data` location was the right place.  And you deleted all the cached data, so it has to download again.  Sounds like you need something a bit more powerful to compile on though. :-)

odinb, 2023-12-23:
Have the same issue, but am unable to see the docker containers.
`mycomputer ~ % ssh root@192.168.1.12
root@192.168.1.12's password:

| |  | |                          /\           (_)   | |            | |
| |__| | ___  _ __ ___   ___     /  \   ___ ___ _ ___| |_ __ _ _ __ | |_
|  __  |/ _ \| '_ \ _ \ / _ \   / /\ \ / __/ __| / __| __/ _\ | '_ \| __|
| |  | | (_) | | | | | |  __/  / ____ \\__ \__ \ \__ \ || (_| | | | | |_
|_|  |_|\___/|_| |_| |_|\___| /_/    \_\___/___/_|___/\__\__,_|_| |_|\__|

Welcome to the Home Assistant command line.

System information
  IPv4 addresses for end0:  192.168.1.12/24
  IPv6 addresses for end0:  fe80::300:6cab:c4f8:8124/64

  OS Version:               Home Assistant OS 11.2
  Home Assistant Core:      2023.12.3

  Home Assistant URL:       http://homeassistant.local:8123
  Observer URL:             http://homeassistant.local:4357
[core-ssh ~]$ sudo docker ps
-bash: sudo: command not found
[core-ssh ~]$`

This is the version I run:
Core                        2023.12.3
Supervisor              2023.12.0
Operating System  11.2
Frontend.                20231208.2

HarvsG, 2023-12-23:
Hmm possible that when you ssh you are directly entering the home assistant core container. Rather than the host OS machine running the containers...

jshep321, 2023-12-25:
Prologue: 2023.12.5 did not fix this for me.

Ah... not in the core, it's in the addon's docker container.  Makes sense.

I didn't touch the .platformio file, just deleted the raspberrypi files:
```rm -r /data/cache/platformio/raspberrypi*```

And it worked for me.

Thanks @HarvsG & @danielkent-net 

For anyone else on HA OS, here's how to enable this access: 
[https://developers.home-assistant.io/docs/operating-system/debugging/](url)

si458, 2023-12-25:
@HarvsG you suggestion worked a treat for myself thank you!

odinb, 2023-12-25:
Was unable to get the USB-method to work referenced in the developers URL. Tried formatting both in FAT and in ext4, and even made sure the privileges on the file "authorized_keys" were correct. No go!
The command would look successful:
`[core-ssh .ssh]$ ha os import
Command completed successfully.`
but still no login, just the "Connection refused" for port 22222.

Even tried to connect a monitor and a keyboard to my HA-blue, screen worked fine, but keyboard refused to work!

Eventually found out I could add the repo: https://github.com/adamoutler/Addons
Then installed the plugin: "HassOS SSH port 22222 Configurator"
Configured the "id_rsa.pub": "ssh-rsa xxyywwzzz_etc" in the plugin.
Made sure to turn off protection mode, then hit the start button and observe the logs.
Rebooted HA, and port 22222 should now be accessible form the host.
I uninstalled the Add-On at this point.

After logging in, ran "docker ps |grep esphome" and noted the CONTAINER ID of "ghcr.io/esphome/esphome-hassio".
Then ran "docker exec -it 4ac1b0663a3b /bin/bash".
Then instead of doing "rm -r /data/cache/platformio/raspberrypi*" I only removed "raspberrypi" with "rm -rf /data/cache/platformio/platforms/raspberrypi/".

All working now!

scottbob09, 2023-12-25:
2023.12.5 also did not fix this for me.  But shutting down the
regular install, running the dev version, and installing from there works
fine.
Virtualized Hassio,  tried the remove directory and there was no match.

source-creator, 2024-01-06:
Luckily I had created a backup with my `2023.11.6` version, before I upgraded to `2023.12.5`

So when I encountered the error, I clicked on my backup file -> Partial backup (restore) -> only selected ESPHome under Add-ons -> Restore

After the restore, I noticed logs weren't coming through, but then I rebooted Home Assistant, and now everything is working fine again.

DMartelly, 2024-01-06:
I'm running esphome/esphome in a Docker container

Removing the folder config/.esphome/platformio/ was my solution. `rm -r /config/.esphome/platformio/`. I can now compile on RP2040

derkrasseleo, 2024-01-09:
Can confirm removing `/data/cache/platformio/*` in the esp home docker container fixed the compilation

iotdevicesdev, 2024-01-18:
We encountered the same issue when updating our [GGreg20_V3 Geiger counter example for RPi Pico W](https://github.com/iotdevicesdev/RPi-Pico-W_GGreg20_V3-ESPHome). We have ESPHome deployed as a Home Assistant integration in a virtual machine.

The solution for us was to restore with a partial backup back to the version with which there was no problem (December 2023). And after that, without any other action, we immediately upgraded to the newest version using the update. At the same time, our ESPHome configuration files were not affected - all the latest versions of the files were saved with the changes made in recent days.

BR,
Oleksii
IoT-devices, LLC

scottbob09, 2024-01-18:
I used the beta version today, and it compiled and uploaded.  Beta 2023.12.7

Kiendeleo, 2024-02-09:
After talking with people on the discord, we found that uninstalling esphome and re-installing it fixed the issue.  This has been tested on two instances so far.

source-creator, 2024-02-11:
In my previous message, I got esphome working again by downgrading the version, from `2023.12.5` to `2023.11.6`

Today, I upgraded esphome from the previous non-broken version I was using `2023.11.6`, to `2023.12.9`.  Pi Pico compilation works on both these versions.

iotdevicesdev, 2024-02-12:
Yes, everything went well with us too with the upgrade to the latest `2023.12.9` version of ESPHome for our repository [Raspberry Pico W + GGreg20_V3 Geiger counter]( https://github.com/iotdevicesdev/RPi-Pico-W_GGreg20_V3-ESPHome)
Everything is working fine now.
But at first, as described earlier, we had to do a downgrade (`2023.12.5` -> `2023.11.6`) from backups.

ssieb, 2024-02-21:
Then it will stay broken.  There's nothing esphome can do about this.  It's a platformio issue.  Re-installing the addon is quick and easy, so I don't know what your problem is with that.

scottbob09, 2024-02-21:
Did it, and it didnt work.  Hey... your windshield wipers dont work.
Nothing that can be done about it.   I want a solution where it just
works.  Where I dont need to play with it, uninstall it, all because
someone didnt do their regression testing.

skandragon, 2024-03-08:
I am getting this error using the esphome docker image.  I've wiped everything I could find that appeared to be a cache, and it seems to work now.  This issue isn't just limited to the HA plugin.

oddstr13, 2024-06-22:
I also encountered this error upgrading from 2024.5.x to 2024.6.x, forcefully updating the raspberry pi platform by specifying the latest git commit did the trick for me;

```
platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git#503933d
```

This forced a re-download of the platform.
Keep in mind that you should likely update the git hash if you are stumbling upon this or similar issues in the future.

closed 2026-03-12 by bdraco