esphome/issues #5263: NSPanel fails to build after update to ESPHome 2023.12.1

MadOtis, 2023-12-21:
### The problem

A previously working Sonoff NSPanel device now fails to build/update after upgrading ESPHome.  This build error is the result:
`INFO ESPHome 2023.12.1
INFO Reading configuration /config/esphome/nspanel.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/b5254113/esphome/components/nextion/display.py", line 92, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID disp1 was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.`

### Which version of ESPHome has the issue?

2023.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

Sonoff NSPanel (US version)

### Component causing the issue

Possibly Nextion or Display

### Example YAML snippet

```yaml
# Set some variables for convenience
substitutions:
  node_name: nspanel-main
  device_name: Main NSPanel

# Note: this may not be needed if the pull request has been merged.
# Check https://github.com/esphome/esphome/pull/2956 for current status.
external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h

esphome:
  name: $node_name
  comment: $device_name

esp32:
  board: esp32dev

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_password

# Enable logging
logger:

# Enable wireless updates
ota:

# Enable Home Assistant API
api:
  services:
    # Service to play a song
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'
    - service: upload_tft
      then:
        - lambda: 'id(disp1)->upload_tft();'

# A reboot button is always useful
button:
  - platform: restart
    name: $device_name restart
    device_class: restart
    
  - platform: template
    name: $device_name TFT Upload
    device_class: update
    on_press:
      - lambda: 'id(disp1).upload_tft();' 

# Define some inputs
binary_sensor:
  - platform: gpio
    name: $device_name Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - switch.toggle: relay_1
    on_release:
      then:
        - lambda: 'id(disp1).sleep(0);'
        - delay: 1s
        - lambda: 'id(disp1).send_command_printf("sleep_timer.en=1");'

  - platform: gpio
    name: $device_name Right Button
    pin:
      number: 27
      inverted: true
    on_click:
      - switch.toggle: relay_2
    on_release:
      then:
        - lambda: 'id(disp1).sleep(0);'
        - delay: 1s
        - lambda: 'id(disp1).send_command_printf("sleep_timer.en=1");'
  
  - platform: nextion
    name: $device_name Spotify Button
    page_id: 1
    component_id: 4
    
  - platform: nextion
    name: $device_name Vol Up
    page_id: 1
    component_id: 5

  - platform: nextion
    name: $device_name Vol Down
    page_id: 1
    component_id: 6

  - platform: nextion
    name: $device_name Lighting Button
    page_id: 0
    component_id: 5

sensor:
  - platform: wifi_signal
    name: $device_name WiFi Signal
    update_interval: 60s

  - platform: ntc
    id: temperature
    sensor: resistance_sensor
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 10kOhm
    name: $device_name Temperature

  - platform: resistance
    id: resistance_sensor
    sensor: ntc_source
    configuration: DOWNSTREAM
    resistor: 11.2kOhm

  - platform: adc
    id: ntc_source
    pin: 38
    update_interval: 10s
    attenuation: 11db

  # Grab current temperature from Home Assistant
  - platform: homeassistant
    id: current_temperature
    entity_id: weather.home
    attribute: temperature
    on_value:
      # Push it to the display
      then:
        - lambda: 'id(disp1).set_component_text_printf("temp", "%.1f ""\xb0""F", x);'

# Define some outputs
switch:
  # The two relays
  - platform: gpio
    name: $device_name Relay 1
    id: relay_1
    pin:
      number: 22
  - platform: gpio
    name: $device_name Relay 2
    id: relay_2
    pin:
      number: 19

  # Pin 4 always needs to be on to power up the display
  - platform: gpio
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON

number:
  platform: template
  name: $device_name Brightness
  id: brightness
  entity_category: config
#  unit_of_measurement: '%'
  min_value: 0
  max_value: 100
  step: 1
  initial_value: 30
  set_action:
    then:
      - lambda: 'id(disp1).set_backlight_brightness(x/100);'

# Configure the internal bleeper
output:
  - platform: ledc
    id: buzzer_out
    pin:
      number: 21

# Enable ringtone music support
rtttl:
  id: buzzer
  output: buzzer_out

# Configure UART for communicating with the screen
uart:
  id: tf_uart
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200

# Configure the screen itself
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: http://10.0.1.102:8123/local/nspanel.tft
    on_wake:
      then:
        - lambda: 'id(disp1).send_command_printf("sleep_timer.en=1");'
    # A little fun...
    on_setup:
      then:
        - number.set:
            id: brightness
            value: 30
        - lambda: |-
            id(disp1).set_component_text_printf(
              "temp", "%.1f ""\xb0""F", id(current_temperature).state
            );
        - rtttl.play: "twobits:d=4,o=5,b=220:c6,8g,8g,a,g,p,b,c6"
        - lambda: 'id(disp1).send_command_printf("sleep_timer.en=1");'
```

### Anything in the logs that might be useful for us?

```txt
Error stacktrace from build pasted in the comment section above
```

### Additional information

This device was working without issue for several previous releases of ESPHome and without modification.  Suddenly, after the upgrade to 2023.12.1, it started failing to build.

guillempages, 2023-12-21:
There was a change in the display structure that breaks all external display components; the fix needs to be done in the external component. See https://github.com/landonr/lilygo-tdisplays3-esphome/pull/46 for an example of the needed changes to have this compile again.

Probably the PR you are using has not been updated yet

MadOtis, 2023-12-21:
Awesome!  Removing the `external_components:` object did indeed resolve it!  Thank you!

edwardtfn, 2023-12-23:
#2956 was basically requesting an "Exit reparse" before trying to connect to Nextion. That is exactly the same that was implemented with #5868, you just have to add the proper attribute in your Nextion settings:

```yaml
# Example config.yaml
display:
  - id: disp1
    platform: nextion
    exit_reparse_on_start: true
```

Although you will not face this need (and the need for #2956) until you try to connect again to a display in advanced reparse mode (like a brand new device from Sonoff).